OpenKinect/libfreenect2 #123: Installation in Linux Mint 17.1 (ubuntu 14.04 like) - Issue with libusb.h and turbojpeg.so

rammstan, 2015-01-06:
Hello,

Just to inform you i meet 2 issues with linux Mint 17.1 but i resolved it.

1. I had problem with libusb.h
I had to modify in many files the #include<libusb.h> into #include<libusb-1.0/libusb.h>

2. turbojpeg
I had problems other than explained in readme.md. I solved the problem by editing SHARED into STATIC --> ADD_LIBRARY(freenect2 STATIC ${SOURCES}) in CMakeList.txt

BYE

gnbl, 2015-01-11:
I overlooked your 
``[UsbControl::setIrInterfaceState] failed! libusb error -99: LIBUSB_ERROR_OTHER``

I have the same issue: https://github.com/OpenKinect/libfreenect2/issues/126

rammstan, 2015-01-12:
Good to know i m not alone!

gnbl, 2015-01-12:
Hi,

I did not close issue #126, that is just a reference to a similar error caused by the wrong libusb being linked #83.
In the meantime, I have found the same "libusb error -99" on the list at https://groups.google.com/forum/#!topic/openkinect/-DGLyi5oE1E for Mac OS X

Do you get the same dmesg output for the USB device?

rammstan, 2015-01-12:
I'm at work, i ll say you this tonight! ;)

gnbl, 2015-01-12:
I won't be able to help much, but the more data can be collected the better for the developers to pin down the problem..
Also your hardware info etc..
I am wondering if the message "Not enough bandwidth for new device state." could be caused by my modified "Xbox NUI Sensor" cable or because of different firmwares.

rammstan, 2015-01-12:
Here is the DMSEG=

[ 8065.806707] usb 4-1.1: Not enough bandwidth for new device state.
[ 8065.806718] usb 4-1.1: Not enough bandwidth for altsetting 1

[ 6096.414324] usb 3-1: new high-speed USB device number 2 using xhci_hcd
[ 6096.434073] usb 3-1: New USB device found, idVendor=045e, idProduct=02d9
[ 6096.434079] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 6096.434083] usb 3-1: Product: NuiSensor Adaptor      
[ 6096.434087] usb 3-1: Manufacturer: Microsoft Corporation  
[ 6096.434855] hub 3-1:1.0: USB hub found
[ 6096.435307] hub 3-1:1.0: 1 port detected
[ 6096.880074] usb 4-1: new SuperSpeed USB device number 2 using xhci_hcd
[ 6098.213309] usb 4-1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[ 6099.861749] usb 4-1: New USB device found, idVendor=045e, idProduct=02d9
[ 6099.861757] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 6099.861761] usb 4-1: Product: NuiSensor Adaptor      
[ 6099.861764] usb 4-1: Manufacturer: Microsoft Corporation  
[ 6099.862643] hub 4-1:1.0: USB hub found
[ 6099.863078] hub 4-1:1.0: 1 port detected
[ 6100.333928] usb 4-1.1: new SuperSpeed USB device number 3 using xhci_hcd
[ 6100.349529] usb 4-1.1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[ 6100.352526] usb 4-1.1: New USB device found, idVendor=045e, idProduct=02c4
[ 6100.352532] usb 4-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[ 6100.352536] usb 4-1.1: Product: Xbox NUI Sensor
[ 6100.352539] usb 4-1.1: Manufacturer: Microsoft
[ 6100.352542] usb 4-1.1: SerialNumber: 228977733847
[ 6100.587169] usbcore: registered new interface driver snd-usb-audio
[ 6363.385749] perf samples too long (5003 > 5000), lowering kernel.perf_event_max_sample_rate to 25000
[ 8065.435378] usb 4-1.1: reset SuperSpeed USB device number 3 using xhci_hcd
[ 8065.450595] usb 4-1.1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[ 8065.451548] xhci_hcd 0000:04:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff880034b81800
[ 8065.451551] xhci_hcd 0000:04:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff880034b81840
[ 8065.451553] xhci_hcd 0000:04:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff880034b818c0
[ 8065.451556] xhci_hcd 0000:04:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff880034b81880
[ 8065.806707] usb 4-1.1: Not enough bandwidth for new device state.
[ 8065.806718] usb 4-1.1: Not enough bandwidth for altsetting 1

rammstan, 2015-01-20:
For adding informations, i m not using Kinect V2 for windows but the official microsoft adapter for xbox one kinect. Not sure it's different. Someone under 14.04 is running well Kinect for Windows or with adapter?

gnbl, 2015-01-20:
Is it possible for you to test the SDK with Windows 8.1? It can be installed without activation so I think you could test 30 days or so. I get an USB interface warning so maybe it's the adapter (Renesas) (I thought it might be my modified cable but you may be having the same problems with the adapter).

rammstan, 2015-01-21:
Yes, it s wirking well with the SDK with Windows 8.1

gnbl, 2015-01-31:
Must be a different cause from #126, my motherboard 's PCIe 1.1 is not fast enough for the PCIe 2 USB 3 card.

rammstan, 2015-02-07:
It's the same problem than #126 . PCIe 1.1 not fast enought

closed 2015-02-07 by rammstan