OpenKinect/libfreenect2 #126: Console output only (turbojpeg frame rate), no windows

gnbl, 2015-01-11:
Following the instructions from https://github.com/OpenKinect/libfreenect2#ubuntu-1404-perhaps-earlier for a fresh Ubuntu 14.4.1 amd64 install I found the following dependecies neccessary:
```
~$ sudo apt-get update
~$ sudo apt-get install git build-essential libtool cmake libx11-dev checkinstall mesa-utils mesa-common-dev automake libudev-dev libgl-dev libxrandr-dev libxi-dev libglu1-mesa-dev libturbojpeg libopencv-dev 
```
The XBOX Kinect apparently has the same USB IDs as the preview Kinect for Windows, libfreenect2 uses both when looking for devices.

Running requires priviliges `sudo ./bin/Protonect`or modifying the device rights as describet at https://groups.google.com/forum/#!topic/openkinect/qdCWdXGxIog , otherwise it will not find the device:
```
[Freenect2Impl] failed to open Kinect v2 @10:2!
[Freenect2Impl] found 0 devices
[Freenect2Impl] requested device 0 is not connected!
```

Once running, I get the following output in the console, but no graphical output or window (data removed):

```
:~/libfreenect2/examples/protonect$ sudo ./bin/Protonect
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 16 usb devices connected
[Freenect2Impl] found valid Kinect v2 @10:2 with serial xxxxxxxx
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  
0x0010:  
0x0020:  
0x0030:  
0x0040:  
0x0050:  

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  

[UsbControl::setIrInterfaceState] failed! libusb error -99: LIBUSB_ERROR_OTHER
[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[Freenect2DeviceImpl] started
device serial: xxxxxxxxxxx
device firmware: 1.1.3327.0.7
[TurboJpegRgbPacketProcessor::doProcess] Failed to decompress rgb image! TurboJPEG error: 'Not a JPEG file: starts with 0xf7 0xc6'
[TurboJpegRgbPacketProcessor] avg. time: 39.5483ms -> ~25.2856Hz
^Z
[2]+  Stopped                 sudo ./bin/Protonect
```

gnbl, 2015-01-11:
Could be related to https://github.com/OpenKinect/libfreenect2/issues/83 - I will look into it and provide more data soon..

gnbl, 2015-01-11:
It is different in that I get an earlier libusb error.

With ```libusb_set_debug(usb_context_, 4);``` in libfreenect2.cpp line 145 I get the following output

```
$ ./bin/Protonect
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 16 usb devices connected
[Freenect2Impl] found valid Kinect v2 @10:6 with serial xxx
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    ............C..A
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    .!3U... ........
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    ................
0x0040:  31 33 00 00 00 0a 08 11 47 4b 50 32 34 39 2e 31    13......GKP249.1
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                X...........

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        .&..

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.845745] [00000d21] libusb: error [op_set_interface] setintf failed error -1 errno 28
[UsbControl::setIrInterfaceState] failed! libusb error -99: LIBUSB_ERROR_OTHER
[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        .&..

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[ 1.119474] [00000d21] libusb: error [submit_iso_transfer] submiturb failed error -1 errno=2
[TransferPool::submit] failed to submit transfer
...
```

I'm now trying to find out which libusb is being linked and also installing nvidia binary drivers to enable opencl (cmake could not find it).

gnbl, 2015-01-11:
I found the  info on the USB hardware with `sudo lshw`:
```
           *-usb
                description: USB controller
                product: uPD720202 USB 3.0 Host Controller
                vendor: Renesas Technology Corp.
                physical id: 0
                bus info: pci@0000:03:00.0
                version: 02
                width: 64 bits
                clock: 33MHz
                capabilities: pm msi msix pciexpress xhci bus_master cap_list
                configuration: driver=xhci_hcd latency=0
                resources: irq:18 memory:f9100000-f9101fff
```

Also, my setup is a an A plug soldered to most of the Kinect for Xbox 1 cable and a lab power supply.

gnbl, 2015-01-12:
Searching the forums with https://groups.google.com/forum/#!searchin/openkinect/%22-99%22|sort:date brougth the same problem for OSX: https://groups.google.com/forum/#!topic/openkinect/-DGLyi5oE1E

The error -99 also occurs in the very early version of libfreenect2 https://groups.google.com/d/msg/openkinect/7WbjaoZjssk/yoYRURVzC4sJ

gnbl, 2015-01-16:
Will check the libusb being linked later, here's where the error occurs:

https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/src/libfreenect2.cpp#L489
https://github.com/OpenKinect/libfreenect2/blob/bb19703ff4ec605d080ecf95933d32c23d91f479/examples/protonect/src/usb_control.cpp#L301

`int r = libusb_set_interface_alt_setting(handle_, IrInterfaceId, alternate_setting);`

with `IrInterfaceId = 1;` from https://github.com/OpenKinect/libfreenect2/blob/bb19703ff4ec605d080ecf95933d32c23d91f479/examples/protonect/include/libfreenect2/protocol/usb_control.h#L100

docs: http://libusb.sourceforge.net/api-1.0/group__dev.html#ga3047fea29830a56524388fd423068b53

Explanation on SO: http://stackoverflow.com/questions/17546709/what-does-libusb-set-interface-alt-setting-do

Libusb error codes ("other error"): http://libusb.sourceforge.net/api-1.0/group__misc.html#gab2323aa0f04bc22038e7e1740b2f29ef

gnbl, 2015-01-16:
`libfreenect2/examples/protonect/CMakeFiles/freenect2.dir/link.txt`
and
`/home/none/libfreenect2/examples/protonect/CMakeFiles/Protonect.dir/link.txt`

contain arguments like
```
-L/home/none/libfreenect2/examples/protonect/../../depends/libusb/lib 
-lusb-1.0
-Wl,-rpath,/home/none/libfreenect2/examples/protonect/../../depends/libusb/lib:
```

Testing the installed version:
``` 
:~$ sudo apt-get install libusb-dev
:~$ libusb-config --version
0.1.12
``` 

Using 

```c++
	std::cout << "[Freenect2Impl] libusb debuglevel 4" << std::endl;
	libusb_set_debug(usb_context_, 4);

	const libusb_version* vp; 
	vp = libusb_get_version();

	char buffer[64];
	sprintf(buffer, 
		"[Freenect2Impl] libusb: %u %u %u %u %s %s",
		vp->major, 
		vp->minor, 
		vp->micro, 
		vp->nano, 
		vp->rc, 
		vp->describe
		);

	std::cout << buffer << std::endl;
``` 

from 

http://libusb.sourceforge.net/api-1.0/group__misc.html#ga61d07b01404fdea080cd16c1ed8be93d
http://libusb.sourceforge.net/api-1.0/structlibusb__version.html

results in 

``` 
:~/libfreenect2/examples/protonect$ ./bin/Protonect
[Freenect2Impl] libusb debuglevel 4
[Freenect2Impl] libusb: 1 0 19 10905  http://libusb.info
``` 

which should be the one checked out with install_deps.sh

What is causing this error?

christiankerl, 2015-01-17:
you can also check the linked version using the ldd command

if you enable libusb debugging and logging it should also print the system error number (see https://github.com/libusb/libusb/blob/master/libusb/os/linux_usbfs.c#L1417 ). Would be interesting to have the Linux kernel error code to trace this further into the kernel itself

gnbl, 2015-01-17:
>libusb debugging and logging
Do you mean this?
```
[ 0.845745] [00000d21] libusb: error [op_set_interface] setintf failed error -1 errno 28
[UsbControl::setIrInterfaceState] failed! libusb error -99: LIBUSB_ERROR_OTHER
```
If yes, I posted this before.. if no, please clarify. Thanks for your interest.

gnbl, 2015-01-17:
Oh, I forgot: I've run the current SDK on Windows 8.1 with the same hardware and got "Not enough USB controller resources - the Controller does not have enough resources for this device" plus regular device losses and "blinking" of the IR and logo LED (retries maybe every 10 seconds).

I might try an unmodified USB 3 A-B cable made to fit the "KINECT for XBOX ONE" as mentioned on the forum.

gnbl, 2015-01-31:
> I've run the current SDK on Windows 8.1 with the same hardware and got "Not enough USB controller resources - the Controller does not have enough resources for this device"

To use a regular USB cable and solder the power supply wires to the PCB, the top cover needs to be removed by releasing 4 clips/hooks. Unfortunately, this did not help. Neither did messing with Renesas driver Updates for Windows 7 which caused no output or error at all, rather device problems.

The likely solution was found from these resources discussing power supply requirements and PCIe standards required for the USB 3 cards:
 * http://vvvv.org/forum/kinect-2-not-enough-usb-controller-resources
 * https://social.msdn.microsoft.com/Forums/en-US/bb379e8b-4258-40d6-92e4-56dd95d7b0bb/confirmed-list-of-usb-30-pcie-cardslaptopsconfigurations-which-work-for-kinect-v2-during?forum=kinectv2sdk

It turns out my Motherboard's Intel ICH9R "Southbridge"..
    Supports PCI Express 1.1
    - Ports 1-4 can be statically configured as 4x1, or 1x4
    - Support for full 2.5 Gb/s bandwidth in each direction per x1 lane

and the Delock PCI Express card > 2x USB 3.0 No. 89242 (http://www.delock.de/produkte/G_89242/merkmale.html?setLanguage=en) only says "PCI Express x1 standard" but it's likely based on Renesas/NEC µPD720202 (http://am.renesas.com/products/soc/usb_assp/product/upd720202/index.jsp) which states:

    System I/F: PCIe Gen2 x 1 Lane
    USB ports: 2 ports (SS/HS/FS/LS)
    Comply spec to: PCIe 2.0 Base Spec
                              USB3.0 rev1.0 (USB-IF certified: Test ID = 380000043)
                              Intel xHCI rev1.0 

So unless there are PCIe 1.1 USB 3 Cards or some port/lane combination is possible I am at a dead end with my hardware.

closed 2015-01-31 by gnbl