OpenKinect/libfreenect2 #132: libusb error -6: LIBUSB_ERROR_BUSY

DrFuzzy, 2015-01-15:
Succesfully installed libfreenect2 in Ubuntu 14.04 x64. Unfortunatelly when I run ./bin/Protonect, after the Kinect has been detected (Kinect for Windows), I get some libusb busy error:

delk@Darwin:~/libfreenect2/examples/protonect$ ./bin/Protonect
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 11 usb devices connected
[Freenect2Impl] found valid Kinect v2 @2:4 with serial 503XXXXXX542
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[UsbControl::claimInterfaces(IrInterfaceId)] failed! libusb error -6: LIBUSB_ERROR_BUSY
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] deallocating usb transfer pools...
[Freenect2DeviceImpl] closing usb device...
[Freenect2DeviceImpl] closed
[Freenect2DeviceImpl] failed to open Kinect v2 @2:4!
no device connected or failure opening the default one!

Tried running it with sudo, but I get the same libusb busy error.

My USB controller is: Intel Corporation 82801JI (ICH10 Family) USB UHCI Controller
and my Kernel version:  3.13.0-43-generic

Any ideas how to resolve this?

DrFuzzy, 2015-01-19:
Am the only one out there having this problem?

ais-hilgert, 2015-01-20:
I also had this problem due to the fact that XHCI was disabled in BIOS and so the USB Controller was working on USB 2.0 speed instead of the needed USB 3.0!
Could this be the source of your problem? By the way, isn't UHCI for USB 1.0?

DrFuzzy, 2015-01-20:
My controller is USB 2.0 not 3.0. Do I need a USB 3.0 controller?

wiedemeyer, 2015-01-20:
Yes.

DrFuzzy, 2015-01-20:
Oops!

floe, 2015-02-10:
If you use a recent kernel (3.16 or later), both Intel and NEC chipsets should work.

closed 2015-02-10 by floe