OpenKinect/libfreenect2 #133: No color problems with 2 kinects

dfamil, 2015-01-18:
hi

got the shader to work but there is no color image. is the ir image working? how do i call it.

when i did the test of Protonect it work fine.

hooked up 2 kinects
got this

[Freenect2Impl] found 2 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] start preparing...
[Freenect2DeviceImpl] ReadData0x14 response

one side looks fine but the right side is just static.
it appears to be seeing the prototype kinect fine but the commercial one the infrared lights are not on.
when I first connected the commercial version worked fine. when i hooked the two the prototype worked fine but not the commercial one. How when i hook up the commercial one it dosen't work. It still works with windows sdk.

so at some point someone will have to figrue out how to reset the kinnect without un plugging it see
OpenKinect/libfreenect2#78

so when i do this it can than find the commercial one but still no imge

Freenect2Impl] enumerating devices...
[Freenect2Impl] 14 usb devices connected
[Freenect2Impl] found valid Kinect v2 @33:1 with serial 506979642542
[Freenect2Impl] found valid Kinect v2 @33:2 with serial 296149434347
[Freenect2Impl] [sorted] found valid Kinect v2 @33:2 with serial 296149434347
[Freenect2Impl] [sorted] found valid Kinect v2 @33:1 with serial 506979642542
[Freenect2Impl] found 2 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] start preparing...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data

thanks

d

christiankerl, 2015-01-22:
which operating system are you using? can you post the full Protonect output? are you running 2 instances of Protonect or your own code, which uses libfreenect2?

floe, 2015-02-10:
Still an issue? Otherwise, please close.

closed 2015-05-21 by floe