OpenKinect/libfreenect2 #139: OpenCL compilation fail on Ubuntu 14.04

ricardborras, 2015-01-26:
Protonect with OpenCL enabled (as detected by cmake CMakeLists) fails to compile due to SSE instructions not enabled. To solve it, I added this lines

set(CMAKE_C_FLAGS ${CMAKE_C_FLAGS} "-msse3") 
set(CMAKE_CXX_FLAGS ${CMAKE_CXX_FLAGS} "-msse3")

into CMakeLists.txt (inside IF(ENABLE_OPENCL AND OPENCL_FOUND))

floe, 2015-02-10:
For Intel GPUs, this works for me even without the additional compiler flags. Are you using an NVidia GPU?

ricardborras, 2015-02-22:
No, I am using a Intel Express embedded graphic card

xlz, 2015-05-19:
Please provide specific error messages during failure.

xlz, 2015-10-16:
Fixed by #429. This was caused by direct usage of _mm_mfence() in old cl.hpp.

floe, 2015-11-06:
Should be fixed by #429, please re-test with latest master and re-open if necessary.

closed 2015-11-06 by floe