OpenKinect/libfreenect2 #140: "not all subsequences received xy" and "image data too short" leads to no depth data

arne48, 2015-01-26:
I am able to compile everything as described in the readme file.
But when I know execute the Protonect executable in see the rgb image but the depth and the ir data window stays black. Additionally to this behavior I get the following console output where the subsequence and image too short error appears over and over again.

...
[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 176133734547
device firmware: 4.1.3912.0.7
[DepthPacketStreamParser::handleNewData] not all subsequences received 0
[DepthPacketStreamParser::handleNewData] not all subsequences received 960
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 895
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[OpenGLDepthPacketProcessor] avg. time: 6.96966ms -> ~143.479Hz
[TurboJpegRgbPacketProcessor] avg. time: 25.6359ms -> ~39.0078Hz
...

I am using Ubuntu 14.04 without any fancy PPAs or other customizations to the system.
Updating the stock 3.13.44 kernel to 3.16.7 mainline reduced the frequency of the error but I am still not able to see ir or depth data.

larshg, 2015-01-30:
Hi @arne48 

The reason why you get a lot of "not all subsequences received" and "Image data too short", might be due to little processing power.
Atleast when I run protonect on my laptop I get alot more of these messages than when running on my stationary. It might also depend on the hardware in use... Chipset, USB controller etc.

There seems to be problems with the OpenGL pipeline on both Ubuntu and Mac - can you try the OpenCL or cpu implementation?

arne48, 2015-01-30:
Hi @larshg 
I was quite busy the last days but yesterday I figured out that indeed as you say its a problem with OpenGL. After changing the typedef  of the DefaultPacketPipeline to CpuPacketPipeline it worked but it was quite slow. I wasn't able to check the OpenCLPacketPipeline yet. The problem with OpenGL seems to be that my installed mesa version supports only 3.1 and libfreenect2 needs at least. 3.3 The "not all subsequences received" and "Image data too short" reduced now to just one or two in the beginning of the usage of the Kinect. But this doesn't look to severe. I will try to find out more. But I think my issue is closed with this result.

closed 2015-01-30 by arne48

manolovaza, 2015-04-09:
I have the same problem, no depth image and no ir image. I have the problem with Protonect. I have already installed the opencl lib package for kubuntu 64 bit, and still cannot see the depth images.

wiedemeyer, 2015-04-09:
You need to enable opencl in cmake, modify the code of `Protonet.cpp` to use the opencl depth processor and build it again.

manolovaza, 2015-04-09:
hi Thiemo,
**it seems that the Opencl is enabled in the cmake file:**

OPTION(ENABLE_CXX11 "Enable C++11 support" OFF)
OPTION(ENABLE_OPENCL "Enable OpenCL support" ON)
OPTION(ENABLE_OPENGL "Enable OpenGL support" ON)

**When I run it with ros I cannot even visualize the RGB image, I get this result  :**

`[ INFO] [1428583497.070346162]: Connected to master at [localhost:11311]
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/research/ROS_workspace/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4`

`[OpenCLDepthPacketProcessor::init] ERROR: clGetPlatformIDs(-1001)
terminate called after throwing an instance of 'cl::Error'
  what():  clGetPlatformIDs
Aborted (core dumped)`

wiedemeyer, 2015-04-09:
Do you use nvidia? I don't know why, but the first time you start it, you may need to run it with sudo. Try `sudo -i`, then source your workspace and run kinect2_bridge.

manolovaza, 2015-04-09:
no, I am using this GPU : Intel® Ivybridge Mobile 

I found [this](http://answers.ros.org/question/194247/how-to-launch-sudo-rosrun/) about sourcing that is why I didn't do that before. Should I try it once? Actually I was able to visualize the rgb by running `rosrun` or `roslaunch`, but after I installed the opencl lib package I could just visualize it through  `./Protonect`.

manolovaza, 2015-04-15:
hi Thiemo,
I already sourced it and run it with `sudo -i` but I have the same output:
`[OpenCLDepthPacketProcessor::init] ERROR: clGetPlatformIDs(-1001)
terminate called after throwing an instance of 'cl::Error'
  what():  clGetPlatformIDs
Aborted (core dumped)`

manolovaza, 2015-04-16:
this solution worked for me from [issue 74](https://github.com/OpenKinect/libfreenect2/issues/153):

rosrun kinect2_bridge kinect2_bridge _reg_method:=cpu _depth_method:=cpu

ash22194, 2015-05-11:
Even I am getting the same error!

vedanshu95, 2015-05-27:
@arne48 

I am facing the same error. I am unable to see ir and depth image as well. Can you help me with that?

thanks