OpenKinect/libfreenect2 #141: Inconsistent frame rates

jeppewalther, 2015-01-27:
When using libfreenect2 with ROS i discovered that the frame rate from libfreenect2 sometimes drops to 15 FPS.  The issue seems to happen randomly and is not reproducible. 

Further info and output is available here
https://github.com/code-iai/iai_kinect2/issues/44#issuecomment-71645887

jeppewalther, 2015-01-27:
Here is some output where I get 30 Hz then suddenly experience a drop to 15HZ

```
[kinect2_bridge] avg. time: 0.033324 -> ~30.0084 Hz
[DepthPacketStreamParser::handleNewData] not all subsequences received 895
[OpenCLDepthPacketProcessor] avg. time: 2.47218ms -> ~404.501Hz
[TurboJpegRgbPacketProcessor] avg. time: 22.7136ms -> ~44.0265Hz
[kinect2_bridge] avg. time: 0.0337148 -> ~29.6605 Hz
[OpenCLDepthPacketProcessor] avg. time: 2.36659ms -> ~422.548Hz
[TurboJpegRgbPacketProcessor] avg. time: 23.0458ms -> ~43.3919Hz
[kinect2_bridge] avg. time: 0.0333362 -> ~29.9974 Hz
[OpenCLDepthPacketProcessor] avg. time: 2.24803ms -> ~444.834Hz
[TurboJpegRgbPacketProcessor] avg. time: 22.6746ms -> ~44.1022Hz
[kinect2_bridge] avg. time: 0.0357271 -> ~27.99 Hz
[OpenCLDepthPacketProcessor] avg. time: 2.26594ms -> ~441.317Hz
[kinect2_bridge] avg. time: 0.0666868 -> ~14.9955 Hz
[OpenCLDepthPacketProcessor] avg. time: 2.43598ms -> ~410.512Hz
[TurboJpegRgbPacketProcessor] avg. time: 22.681ms -> ~44.0898Hz
[kinect2_bridge] avg. time: 0.0667897 -> ~14.9724 Hz
[OpenCLDepthPacketProcessor] avg. time: 2.4848ms -> ~402.446Hz
[kinect2_bridge] avg. time: 0.0665413 -> ~15.0283 Hz
[kinect2_bridge] avg. time: 0.0666682 -> ~14.9997 Hz
[OpenCLDepthPacketProcessor] avg. time: 2.5746ms -> ~388.41Hz
[TurboJpegRgbPacketProcessor] avg. time: 22.6081ms -> ~44.2319Hz
[DepthPacketStreamParser::handleNewData] not all subsequences received 1022
[kinect2_bridge] avg. time: 0.0666802 -> ~14.997 Hz
[OpenCLDepthPacketProcessor] avg. time: 2.60829ms -> ~383.393Hz
[kinect2_bridge] avg. time: 0.0667112 -> ~14.99 Hz
[OpenCLDepthPacketProcessor] avg. time: 2.61432ms -> ~382.509Hz
[TurboJpegRgbPacketProcessor] avg. time: 22.5379ms -> ~44.3697Hz
[kinect2_bridge] avg. time: 0.066684 -> ~14.9961 Hz
[OpenCLDepthPacketProcessor] avg. time: 2.54729ms -> ~392.574Hz
[kinect2_bridge] avg. time: 0.0665752 -> ~15.0206 Hz
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 959
[OpenCLDepthPacketProcessor] avg. time: 2.58595ms -> ~386.706Hz
[TurboJpegRgbPacketProcessor] avg. time: 22.9331ms -> ~43.605Hz
[kinect2_bridge] avg. time: 0.0666975 -> ~14.9931 Hz
[OpenCLDepthPacketProcessor] avg. time: 2.56566ms -> ~389.764Hz
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 1015
[kinect2_bridge] avg. time: 0.0556653 -> ~17.9645 Hz
[TurboJpegRgbPacketProcessor] avg. time: 22.6196ms -> ~44.2094Hz
[OpenCLDepthPacketProcessor] avg. time: 2.57645ms -> ~388.13Hz
[kinect2_bridge] avg. time: 0.0333336 -> ~29.9998 Hz

```

christiankerl, 2015-01-27:
Which framerate drops, RGB or IR? Does the kinect_bridge use the sync multi frame listener? The RGB rate drops automatically to 15Hz in bad lighting conditions. Could this be the reason?

jeppewalther, 2015-01-27:
@christiankerl it seems like it is the bad lighting conditions with bad lighting conditions meaning a normal room in the evening. The FPS seems to rise to 30 again when i point at my bright monitors. 

@wiedemeyer what is your take on this. If it is only RGB i guess it could be fixed in kinect2_bridge?

wiedemeyer, 2015-01-29:
Ok, so it is related to the lighting conditions. Yes, the kinect2_bridge uses the sync multi frame listener.

It needs some rewrites to fetch and send depth and color with different frame rates. I don't know if it is actually a problem that needs to be fixed. I guess most of the time people would use depth and color together and in that case you are limited by the color frame rate anyway. A simple fix would be to turn on the lights.

ricardborras, 2015-01-29:
Hi,

i think that is a good idea to decouple depth, ir and color. For instance,
in my use case, I only need depth image,

Thanks,

Ricard

gaborpapp, 2015-01-29:
It would be nice to have a steady 30fps depth image regardless of the lighting conditions, I second Ricard's opinion. There are many use-cases when only the depth image is used.

larshg, 2015-01-30:
When you instantiate the SyncMultiFrameListener you can just pass in the Frame::Depth argument:

From this:
```
libfreenect2::SyncMultiFrameListener listener(
libfreenect2::Frame::Color | libfreenect2::Frame::Ir | libfreenect2::Frame::Depth);
  
```

To this:
```
libfreenect2::SyncMultiFrameListener listener(libfreenect2::Frame::Depth);
```

And you should only get the depth frame :) - I haven't tested it directly, but it should work!

wiedemeyer, 2015-01-30:
At the moment kinect2_bridge waits for all frames and processes them all. So if some are missing it would end up in a segfault. But I am currently looking into decoupling depth and color.

larshg, 2015-01-30:
Ah - okay :)

wiedemeyer, 2015-01-30:
I decoupled depth and color. It is inside the devel branch.

jeppewalther, 2015-01-30:
Great work .. thanks for the effort.

closed 2015-01-30 by jeppewalther

gaborpapp, 2015-02-04:
@wiedemeyer where can I find your modifications? I cannot find a devel branch either in https://github.com/wiedemeyer/libfreenect2/ or in https://github.com/OpenKinect/libfreenect2/

wiedemeyer, 2015-02-04:
you can find them here: https://github.com/code-iai/iai_kinect2

gaborpapp, 2015-02-04:
thanks. will you submit a pull request to upstream?

wiedemeyer, 2015-02-04:
There are no changes for libfreenect2 because there is no issue in libfreenect2. If someone uses the `libfreenect2::SyncMultiFrameListener` with color and ir/depth, then it will wait until all frames are present, meaning that it will always be limited to the lower frame rate. If this behavior is not desired one should create two listener, one for ir/depth and one for color.

gaborpapp, 2015-02-04:
thanks for the explanation