OpenKinect/libfreenect2 #143: Depth limited to 4.5 meters?

mmunaro, 2015-01-28:
Hi,

with commit 50868724c02a91e281973cb2e2a1db1a29779454, I could get depth up to 10 meters, while now (I tried with commit 3c0cda4ffb6bea47a883439c1a1fa631eb6ab205) depth seems to be limited to about 4.5 meters.

Any idea why this happens or how to remove this limit as it was before?

Thanks!

Best,
Matteo

christiankerl, 2015-01-28:
You can change the limits by instantiating one of the packet pipeline implementations, call getDepthPacketProcessor and then call its setConfiguration method. Pass your packet pipeline object to the openDevice method

mmunaro, 2015-01-28:
Thanks for your quick reply.
I did this, but nothing changes:

    libfreenect2::DepthPacketProcessor::Config config;
    config.MaxDepth = 10000;
    packetPipeline->getDepthPacketProcessor()->setConfiguration(config);
    device = freenect2.openDevice(camSerial, packetPipeline);

Are these parameters really used?
It looks like these parameters are used instead:
https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/src/depth_packet_processor.cpp#L42

christiankerl, 2015-01-28:
Which pipeline are you using? I'm not sure whether the opencl one respects the parameters @wiedemeyer  ?

mmunaro, 2015-01-28:
Yes, I am using libfreenect2::OpenCLPacketPipeline

wiedemeyer, 2015-01-29:
Yes, it just uses the `DepthPacketProcessor::Parameters` and does not update the min and max distance based on the config. I will fix this and create a pull request.

christiankerl, 2015-01-29:
@mmunaro please close this issue if the fix works for you

mmunaro, 2015-01-29:
Yes, this fix works (if both MinDepth and MaxDepth are specified and if they are expressed in meters).

    libfreenect2::DepthPacketProcessor::Config config;
    config.MinDepth = 0.3f;
    config.MaxDepth = 10.0f;
    packetPipeline->getDepthPacketProcessor()->setConfiguration(config);
    device = freenect2.openDevice(camSerial, packetPipeline);

closed 2015-01-29 by mmunaro

zardchim, 2015-05-31:
Sorry to raise this problem again. I pull the newest code, however when I set the config.MaxDepth to more than 4.5f, it shows "skipping depth packet", and no depth image shown. My code is as follow:

	pipeline_1 = new libfreenect2::OpenCLPacketPipeline();

	libfreenect2::DepthPacketProcessor::Config config;
	config.MinDepth = 0.5f;
	config.MaxDepth = 10.0f;
	pipeline_1->getDepthPacketProcessor()->setConfiguration(config);

	if (freenect2_1.enumerateDevices() == 0)
	{
		std::cout << "Device 1 not connected" << std::endl;
		return -1;
	}
	std::string serial_1 = freenect2_1.getDeviceSerialNumber(0);
	dev_1 = freenect2_1.openDevice(serial_1, pipeline_1);

Any idea why this happen?? Thank you!

spium, 2015-07-15:
Hello,
it seems like the problem is not fully fixed yet. In fact, when building the OpenCL program in OpenCLDepthPacketProcessorImpl::buildProgram(), printing the build options shows that MinDepth and MaxDepth values do not change.
I found out that the issue is in OpenCLDepthPacketProcessor::setConfiguration(). When changing min and/or max depth, the OpenCL program is built BEFORE setting the config to the new values (impl_->config = config). Therefore, the build process still reads the old values.
The fix is to simply update the config before building the OpenCL program.

floe, 2015-07-15:
Ah, this was probably introduced by #293 (/cc @HenningJ). Is it maybe sufficient if you just comment out line 614 in `opencl_depth_packet_processor.cpp`?

spium, 2015-07-15:
I don't think so, otherwise the OpenCL program wouldn't be rebuilt at all and would still be running the old version with the old config. What I did was copy the last line in the method (impl_->config = config) inside the if body just before line 614. If you want I can submit a pull request with the fix, but it's so simple I don't think it's worth the effort.

floe, 2015-07-15:
Proposed fix in #324.

HenningJ, 2015-07-15:
oups, my bad. 
Just removing the line would have worked as well, because then it would have rebuilt the program just before starting it. But it's better to rebuilt immediately, like @floe did in #324