OpenKinect/libfreenect2 #144: Depth measurement offset

wiedemeyer, 2015-01-28:
While calibrating the Kinect2 to our robots, I saw that the depth values do not match the robots model.

I investigated this and wrote a program that compared the measured distances to the calculated distances from a chess board I used for the calibration. From each image I used the region of the board and compared each pixels (valid) depth measurement to the calculated distance by ray intersection to the board plane. Before doing this I undistorted the images.

It seems like the Kinect2 has a static offset of around 24 mm. The depth measurement is always 24 mm further away than it should be. Here are some plots of the results. This offset seems to be not related to image coordinates or distance. The error grows towards the image corners, but there the distortion is much stronger and the values get more noisy.

[image]
[image]
[image]
[image]

I only used the OpenCL depth packet processor. So I can't tell right now, if it is just related to that specific implementation, all implementation or the device itself. Tomorrow I will try out the OpenGL and CPU depth packet processor as well. Did someone experienced a similar behavior until now?

@christiankerl: Do you think it could be a problem of the packet processor implementation?

I added the code for the depth calibration to the [calibration tool](https://github.com/code-iai/iai_kinect2/tree/devel/kinect2_calibration) of my [ROS Kinect2 bridge](https://github.com/code-iai/iai_kinect2/tree/devel/). If the depth values are decreased by the calculated offset the resulting depth cloud perfectly matches the model of our robots.

christiankerl, 2015-01-28:
I guess the problem is that the xtable and ztable are used in the depth calculations, but they are only valid for the device I used during the memory dump. Actually they should be computed from some calibration values stored on each device. However, the formula is unknown

sh0, 2015-01-29:
I heard that Kinect 2 has slight distance offset depending on what material you are observing. I'm not a specialist on ToF cameras, but I guess that when the material is highly reflective then the reflected light intensity is stronger and will activate the photosensitive element faster. Reportedly there can be 1-2 cm difference when comparing specular metal to a material with very diffuse reflection. I haven't tested for this effect myself so I really can't say how true all of this actually is.

wiedemeyer, 2015-01-29:
@christiankerl: And the memory dump came from using the sensor on Windows, with the official SDK, right? And we don't know the formula nor the sensor data from which it is calculated. If it is related to the xtable and ztable, then there should be no offset on your sensor, but if your sensor has an offset too, then it's something else.

@sh0: Yes, I have seen this on shiny metal, opaque and translucent surfaces. But in this case the offset appeared on all surfaces. For example the PR2 has plastic and textile covers on its arms, and KUKA arms have shiny and non shiny pained parts. And the distance measurements are from a chess board printed on paper and glued to a thick and flat acrylic glass.

wiedemeyer, 2015-01-29:
I tested it with the OpenGL depth packet processor and it is the same offset.

djgaspa, 2015-01-29:
How could we dump the proper xtable/ztable of our devices? Is there a procedure?
I've seen that somebody reckon those tables might depend on a couple of tables retrieved together with color intrinsec params. Is there any preliminary analysis of those data?

floe, 2015-02-03:
The blob with the intrinsics seems to contain two "subtables" of floats, one of size 56*46 and one of size 28*23. This is almost the same aspect ratio as the xTable/zTable, which are 512x424, and I'm suspecting that they may be just upsampled versions. Some discussion related to this is also going on in #41.

sh0, 2015-03-02:
I have been working on the xTable/zTable problem.

The packets with suspected tables are deserialized in ``KinectService.exe``. There seem to be actually three not two tables. The reason is that the tables are copied as three separate blocks. Interestingly there seem to be different packet formats, possibly from older development firmware. The structure with tables is then copied around and I can't really pinpoint where it is going due to the massive amount of vcalls going on. The tables are arranged as follows:

    0x0068: float table1[28 * 23 * 2]; // size 0x1420 // from 0x0054
    0x1488: float table2[28 * 23 * 2]; // size 0x1420 // from 0x1474
    0x28a8: float table3[28 * 23 * 1]; // size 0x0a10 // from 0x2894

I've found at least 20 functions with 512x424 loops in them and I'm sure there are more. This means that I don't really know what loop calculates what. Anyhow, eventually the computed tables end up in ``vvtechs.dll`` where they are fed into the shaders. Depth map calculation related shaders take various textures as input, including ones named ``zTable``, ``xTable`` and ``irTable``, particulariliy in that order. One guess would be that the ``table1`` and ``table2`` are related to ``z/xTable`` and ``table3`` is related to ``irTable``. The ``irTable`` is used to look up IR image intensity coefficients. Or at least it looked like it from one shader bytecode.

I have also found couple IR related hardcoded 512x424 tables in ``KinectService.exe``. Basically only one of the tables is used based on Kinect version info from device. The values range from around 40.0 to 4000.0. It has some banding artifacts just like ``z/xTable`` but does not match the value range. In this sense I'm not sure what those tables are.

Doc-Ok, 2015-03-26:
The subtables in the color camera intrinsics reply appear to have the following structure:

* One 28x23 table with 2 floats per entry, defining a 2D position in depth camera space.

[image]

The black outline is the 512x424 depth camera space.

* One 28x23 table with 2 floats per entry, defining a 2D position in color camera space.

[image]

The black outline is the 1920x1080 color camera space. I do not know why the grid is offset and oversized. This may be a way to express the color camera pixel grid in depth camera space, which would lead to some horizontal offset, but that's just a guess. The aspect ratio of the grid sort-of matches the depth camera; this might be a depth-to-color mapping.

* One 28x23 table with 1 float per entry, which in my camera all hover around 870, and the last entry in each row is 0. No idea what these mean

floe, 2015-03-26:
@sh0 about those hardcoded tables - are they actually used anywhere? The current libfreenect2 code only uses xTable & zTable from the original memory dump.
@Doc-Ok Is it possible that this is just the original raw calibration grid data, included for reference, and has already been used to calculate the coefficients sent in the same packet?

sh0, 2015-06-29:
@floe Sorry for not being around, I've been busy with graduation and so on.

So far I have not seen the xTable and zTable in use. They get copied around quite a bit so I probably just can't correctly trace it. It is however possible that these are factory calibration data points embedded just in case some later SDK version needs them.

I do have a couple mystery tables that I have found.

## unktable

There are two hardcoded tables within ``KinectService.exe`` that seem to be some sort of correction tables. Only one of the tables is used at a time and the selection is based on the device version / firmware version as reported by the device.

### unktable1

    0x002b4f70: uint8_t unktable1_unk0 = 1;
    0x002b4f71: uint8_t unktable1_unk1 = 0;
    0x002b4f72: float unktable1_a = 675.61841;
    0x002b4f76: float unktable1_b = 1.0816;
    0x002b4f7a: uint8_t unktable1_data[512 * 424];

### unktable2

    0x002e9f80: uint8_t unktable2_unk0 = 1;
    0x002e9f81: uint8_t unktable2_unk1 = 0;
    0x002e9f82: float unktable2_a = 678.51184;
    0x002e9f86: float unktable2_b = 0.77815998;
    0x002e9f8a: uint8_t unktable2_data[512 * 424];

### Conversion to float

Every ``uint8_t`` element in the tables is later converted in ``vvtechs.dll`` to a float using following method:

    float unkcalc(float r, float a, float b)
    {
        r = (r / 255.0f) * a + b;
        if (1.0f >= r)
            r = 1.0f;
        return 4096.0f / r;
    }
    
### Files

The corrected tables after conversion can be found [here](https://gist.github.com/sh0/22e1ece83e02c65afce9) for convenience.

floe, 2015-07-01:
@sh0 no worries :-) Interesting to see that there are apparently more correction factors... I think we should try to capture xTable/zTable from GPU memory with a different Kinect than the one used originally and compare them. @christiankerl can you maybe describe how you extracted those tables again?

christiankerl, 2015-07-01:
I used api monitor to intercept the directx api calls. The textures are populated using map/unmap. With api monitor you can set a breakpoint on unmap. At this time the texture is already filled, but the buffer is still accessible. Then I used a small program to read this buffer content from the kinect driver process memory. The buffer address can be obtained from the map call, if I remember correctly. So its a little bit of manual labor.

floe, 2015-07-01:
Could you maybe simply repeat this process with a different Kinect, assuming you have access to one?

christiankerl, 2015-07-01:
Currently I don't have access to a windows 8 pc

sh0, 2015-07-01:
I actually extracted the HLSL shaders from ``vvtechs.dll``, SDK version ``v2.0_1409`` some time ago and generated the disassemblies for them. There are 16 files in total, but most are duplicates or not related to depth maps. The main shader is 1500 lines or so of assembly.

The input locations to the shaders may be interesting for capturing x/zTable so I worked out the offsets:
https://gist.github.com/sh0/ccda1b562e9cf1bb4ab3

floe, 2015-07-14:
@christiankerl could you maybe share the program you used to extract the tables?

christiankerl, 2015-07-14:
I don't have it anymore...

floe, 2015-07-14:
OK, I'll try to summarize:

1. get http://www.rohitab.com/apimonitor
1. set a breakpoint for `ID3D11DeviceContext::(Un)Map` (?)
1. start some Kinect SDK sample application
1. when breakpoint gets hit, write down the buffer address
1. use a small wrapper program around `ReadProcessMemory` to extract the raw buffer content

Anything I missed?

christiankerl, 2015-07-14:
I don't know if this is really required: make sure that KinectService.exe is not running, but start it through apimonitor so the api hooking works... maybe attaching to a running KinectService.exe also works, but I don't remember if something interesting already happens on startup. CreateTexture2D or CreateBuffer might also be interesting, they should be used to create the resources passed to Map and contain dimensions, data type and so on...

xlz, 2015-08-17:
The builtin ztable is generated with these parameters:
```matlab
cx 255.16569542
cy 211.82460040
fx 367.28674900
fy 367.28675187
k1 0.0914235179439778
k2 -0.269355436246
k3 0.092570290246685
p1 ?
p2 ?
```

I found a really good curve fit with the ztable.
```
z = fread(fopen('zTable.bin'), [512 424], 'single');
unambigious_dist = 6250.0 / 3;
[y_pix, x_pix] = meshgrid(0.5:423.5, 0.5:511.5);
x = (x_pix-cx)/fx;
y = (y_pix-cy)/fy;
rd2 = x.^2 + y.^2;
fx = (unambigious_dist./z).^2 - 1;
fy = rd2;
ft=fittype('x*(1+k1*x+k2*x^2+k3*x^3)^2');
[fitobject, gof] = fit(fx(:), fy(:), ft, ...);
gof.sse

ans =

      2.41179064892214e-09
```

The above calculation obtains the distortion function `r_d^2 = f(r^2)` used in this procedure, given the value of `zmultiplier(x_d, y_d)` (ztable) to convert distance of focal point to depth to XY plane for each actual (distorted) pixel `x_d,y_d`, and with ideal pixel `x,y`:
```
Define r_d^2 = x_d^2 + y_d^2.
Define r^2 = x^2 + y^2.
To satisfy d_linear = zmultiplier(x_d, y_d) d_focal:
Given zmultiplier(x_d, y_d) = zmultiplier(r_d^2) = unambigious_dist/sqrt(r^2 + 1),
Define g(r_d^2) = r^2= (unambigious_dist/zmultiplier(r_d^2))^2 - 1.
Therefore g(r_d^2) = f^{-1}(r_d^2) = r^2.
Given f(r^2) = r^2(1 + k_1 r^2 + k_2 r^4 + k_3 r^6)^2,
we have the full mapping of f^{-1}(r_d^2), and f(r^2).
Solve k_{1,2,3} with curve fitting.
```

ztable can be generated with `zmultiplier(x_d, y_d)`. Though I haven't figured out what xtable does. It does have a clean form:

```
xmultiplier(r_d^2) = 3 unambigious_dist x h(r_d^2)
```
The last part of `processPixelStage2()` is quite weird. I don't know what exactly `h(r_d^2)` is. But here is a curve fit:
```
     h(x) = p1*x^3 + p2*x^2 + p3*x + p4
Coefficients (with 95% confidence bounds):
       p1 =     0.07799  (0.07779, 0.0782)
       p2 =      0.2151  (0.2149, 0.2153)
       p3 =    -0.09492  (-0.09498, -0.09486)
       p4 =        1.31  (1.31, 1.31)
```
I suspect it does tangential distortion.

floe, 2015-08-18:
Very interesting; so we _should_ be able to generate at least the ztable from the content of DepthCameraParameters? (IIRC p1/p2 are always zero).

One thing which I find confusing: if xtable/ztable are indeed for radial/tangential distortion, then why are the resulting IR/depth images still distorted and need an additional step to "rectify" them?

xlz, 2015-08-18:
ztable is not for radial distortion though. It is for deriving the correct depth value under distortion. The image is not warped. Kinect 1 measures disparity which is proportional to depth to XY plane, but Kinect 2 measures distance to the focal point.

xlz, 2015-08-18:
In the meantime, the 11to16.bin table can be generated exactly by a pure piecewise linear function with parameters all being power of 2.

```cpp
void gen11to16(int16_t *lut)
{
  int x = 0;
  int y = 0;
  for (x = 0 ; x < 1024; x++) {
    int inc = x/128-1;
    if (inc < 0)
      inc = 0;
    lut[x] = y;
    lut[1024+x] = -y;
    y += 1<<inc;
  }
  lut[1024] = 32767;
}
```

The result is exact. This generation does not seem to rely on external parameters, therefore it is likely the table is the same across devices.

xlz, 2015-08-19:
Following @sh0's unktable1:

It has 512x424 numbers.
```
>> u=load('unktable1.csv');
>> u=(4096./u-1.0816)/675.61841;
>> imagesc(u);
```

[image]

This is a single "curve" wrapped modulo 512.

```matlab
i=0:512*424-1;
p=225.28/3; % empirically determined
x=mod(mod(i,512)-p*i/512, 512-p)/(512-p)*512; % unwrap
y=floor(i/512);
z=u';
[xi,yi]=meshgrid(0:511,0:423);
zi=griddata(x,y,z(:),xi,yi);
surf(xi,yi,zi,'EdgeColor','none');
figure; imagesc(zi);
```

[image]
[image]

This looks like something from a Fourier series of a square wave. But frankly I have no idea what this is correcting.

christiankerl, 2015-08-19:
this looks like multipliers to correct the non-uniform illumination in the IR image. I calibrated such an image once myself and it looked pretty similar to this one...

sh0, 2015-08-19:
I now also think that the unktables are used in IR illumination value correction because of some clues in DLLs. Originally I even named them irtables, but I wasn't sure so I renamed them unktables.

@xlz Thanks for the contribution. I got bunch of ideas where to look next.

xlz, 2015-09-02:
It turns out the xtable is literally the `x` table, i.e. `xtable(x_distorted, y_distorted) == x_undistorted`. The plot with `h(r_d^2)` was not really meaningful because it's `h(r^2)` that matters. Anyway...

xtable and ztable can be generated like this, using `kinect_undistort` from https://github.com/OpenKinect/libfreenect2/issues/41#issuecomment-72022111:
```cpp
for (int xi = 0; xi < 512; xi++) { // distorted, unnormalized coordinates, off by 0.5
  for (int yi = 0; yi < 424; yi++) {
    double xd = (xi + 0.5 - cx)/fx; // distorted, normalized coordinates
    double yd = (yi + 0.5 - cy)/fy;
    double xu; // undistorted, normalized coordinates
    double yu;
    kinect_undistort(xd, yd, xu, yu);
    xtable[yi][xi] = 8192*xu;
    ztable[yi][xi] = 6250.0/3/sqrt(1+xu*xu+yu*yu);
  }
}
```

I used a different implementation of `kinect_undistort` with Newton's method and double precision to derive the inverse of distortion.

The original intrinsics of xTable.bin and zTable.bin (elements 119,11 of the two tables are corrupted to zero) are unknown. The best estimates obtained are:
```
fx=367.286994337726;
fy=367.286855347968;
cx=255.165695200749;
cy=211.824600345805;
k1=0.0914203770220268;
k2=-0.269349746097515;
k3=0.0925671408453617;
p1=p2=0;
```

The Sum of Absolute Difference errors between derived x/ztable and original x/zTable.bin are
```
diffx = 34.6827115756027;
diffz = 16.6134954007271;
```
The average is about 1.6e-4 per pixels. I think this is mostly floating point error.

----

I think I have figured out the meaning of the last part of `stage2.fs`:

The following discussion is with respect to each pixel `xi,yi` in the texture, ranging from 0-511, 0-423, the distorted, normalized coordinates `xd,yd == (xi+0.5-cx)/fx, (yi+0.5-cy)/fy`, and the undistorted, normalized coordinates `xu,yu` obtained from `xd,yd` with `kinect_undistort()`.

`Params.unambigious_dist` is 6250.0/3.

`zmultiplier = unambigious_dist/sqrt(xu*xu+yu*yu+1)`, as defined above. Then:

`phase_final`: the travel distance of the reflected ray at `xi,yi` divided by `unambigious_dist*2`
```
  float depth_linear = zmultiplier * phase_final;
```
`depth_linear`: the z distance from the object to the XY plane, assuming the travel distance is equal from the light source to the object, and from the object to the focus (these two distances are not equal actually).
```
  float max_depth = phase_final * Params.unambigious_dist * 2.0;
```
`max_depth`: the maximum possible z distance from the object to the XY plane (if this pixel is the principal point). Or equivalently the travel distance of the reflected ray.

```
  xmultiplier = (xmultiplier * 90.0) / (max_depth * max_depth * 8192.0);
  float depth_fit = depth_linear / (-depth_linear * xmultiplier + 1);
```
[image]

z value can be derived with `sqrt(xu^2+yu^2+1)`, the ratio of distance to the unit depth. The problem is that the measurement is the whole travel distance of the reflected ray from the light source to the focal point, instead of the distance from the focal point to the object.

Here is soms derivation for the last two equations:

[image]

`depth_fit` and `z` are the samething, so it seems `xmultiplier/8192 == xu`, and `L == 45`. But I'm not sure why `-L^2` is missing from the numerator in `depth_fit`. Anyway the estimate of `xmultiplier` derived in this way matches the actual values in `xTable.bin`.

sh0, 2015-09-02:
@xlz Great work!

Some days ago I came across the following hardcoded parameters in one of the Kinect internal functions:

    fx = 369.15128
    fy = 369.15128
    cx = 255.5
    cy = 211.5
    k1 = 0.0
    k2 = 0.0
    k3 = 0.0

They look really similar to the intrinsics you gave. However, these parameters seemed to be defaults that could be overridden depending on function arguments.

xlz, 2015-09-29:
I think @christianker meant in his paper http://vision.in.tum.de/_media/spezial/bib/kerl143dv.pdf figure 2(c).

I guess this is a diffraction pattern for IR lasers.

xlz, 2015-10-26:
xtable/ztable are no longer hardcoded and this part of systematic error is removed. Though I don't know if there is still depth measurement offset. This may be moved out of 0.1 milestone for further investigation.

floe, 2015-11-06:
Agreed. At some point, it would be nice to run a proper test series, but this only makes sense after we have a stable release.

xlz, 2016-02-14:
It seems the ztable also contains certain information about bad pixels, where the value would be zero. Right now the ztable generation does not include the information, but I don't know yet where to get the information about bad pixels.

floe, 2016-02-17:
For the record, when I plotted the original xtable & ztable, one outlier with a value of zero in both tables was clearly visible:

[image]
[image]

xlz, 2016-02-18:
Do the bad pixels matter though? A significant number of lines of code in the depth processing routines are dedicated to dealing with the bad pixels therefore it seems to matter. But the depth images are full of noise, and one more noisy pixel doesn't make it much worse?

floe, 2016-02-18:
Hm, it might have an effect in this scenario (measuring distances to a larger object), because the bad pixels are constant outliers AFAICT. The other noise is there, sure, but it follows a normal distribution and should average out over more measurements, as opposed to the outliers.

wangsen1312, 2017-07-26:
@xlz   from the source code, I can see const double unambigious_dist = 6250.0/3; 
However, the three frequency for Kinect 2 is 120 MHz, 80 MHz, and 16MHz. so the unambigous distance should nearly 1.25, 1.875, 9.375.  
Did I misunderstand something?

xlz, 2017-08-06:
It's just a name for some multiplier. We have not mapped it to any physical quantity.