OpenKinect/libfreenect2 #146: Not showing images on Ubuntu 14.04 with USB controller ASMedia Tehnology Inc. Device 1142

abonnin84, 2015-01-29:
OS: Ubuntu 14.04 64 bits. Kernel 3.17
Computer: Shuttle DS437
Ubuntu USB controller: ASMedia Tehnology Inc. Device 1142

Running Protonect results in no images at all.
Images can be shown on Windows 8.1.
Same configuration works on other machines with other USB controllers.
It seems like a Linux USB driver issue.

floe, 2015-02-02:
The Kinect is notoriously picky regarding USB controllers; I can only suggest trying a newer kernel or adding a simple PCIe card with an extra USB3 controller from NEC (known to work reliably).

xlz, 2015-06-09:
Documented as unsupported hardware in #279.

closed 2015-06-09 by floe

ybott, 2015-07-29:
OS: OSX 10.10.4
Computer: MacPro 3,1
USB Controller: ASMedia Tehnology Inc. Device 1142

I have Protonect utility running on the setup above.

floe, 2015-07-29:
Very interesting, does it run reliably?

ybott, 2015-07-29:
Unfortunately no. 

After reboot Protonect utility will successfully connect to the Kinect, open all opencv windows and perform correctly. The app is exited with issues, the transfer_pool pipes are not shutdown correctly (??).

After this initial successful connection and running Kinect the Protonect utility will no connect, restart is always necessary.