OpenKinect/libfreenect2 #153: OpenCL Fails to Initialize Correctly

mitchellwills, 2015-02-03:
I'm having some issues getting the OpenCL depth packet processor to work. I have been testing this on two different machines with similar results. They are
Lenovo T420: Linux walrus 3.13.0-45-generic # 74-Ubuntu SMP Tue Jan 13 19:36:28 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
Lenovo W540: Linux mitchell-W540 3.13.0-45-generic # 74-Ubuntu SMP Tue Jan 13 19:36:28 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

TL;DR I get a -1001 error when enumerating devices with OpenCL

While both devices have NVIDIA Graphics cards in them I am not trying to use OpenCL on them and only one system has the NVIDIA icd installed. 
I installed the Intel SDK for OpenCL (https://software.intel.com/en-us/intel-opencl) and it appears to have installed correctly. This also installs the OpenCL library and headers, but it doesn't seem to make a difference if I install them through apt.
I also tried to install Beignet using the instructions in #109, but it failed to initialize with some permission errors so I removed it.
When running clinfo I get the following output on the W540 (and similar on the T420):
```
Number of platforms:				 1
  Platform Profile:				 FULL_PROFILE
  Platform Version:				 OpenCL 1.2 LINUX
  Platform Name:				 Intel(R) OpenCL
  Platform Vendor:				 Intel(R) Corporation
  Platform Extensions:				 cl_khr_icd cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_spir cl_intel_exec_by_local_thread cl_khr_depth_images cl_khr_3d_image_writes cl_khr_fp64 

  Platform Name:				 Intel(R) OpenCL
Number of devices:				 1
  Device Type:					 CL_DEVICE_TYPE_CPU
  Device ID:					 32902
  Max compute units:				 8
  Max work items dimensions:			 3
    Max work items[0]:				 8192
    Max work items[1]:				 8192
    Max work items[2]:				 8192
  Max work group size:				 8192
  Preferred vector width char:			 1
  Preferred vector width short:			 1
  Preferred vector width int:			 1
  Preferred vector width long:			 1
  Preferred vector width float:			 1
  Preferred vector width double:		 1
  Native vector width char:			 32
  Native vector width short:			 16
  Native vector width int:			 8
  Native vector width long:			 4
  Native vector width float:			 8
  Native vector width double:			 4
  Max clock frequency:				 2700Mhz
  Address bits:					 64
  Max memory allocation:			 6289463296
  Image support:				 Yes
  Max number of images read arguments:		 480
  Max number of images write arguments:		 480
  Max image 2D width:				 16384
  Max image 2D height:				 16384
  Max image 3D width:				 2048
  Max image 3D height:				 2048
  Max image 3D depth:				 2048
  Max samplers within kernel:			 480
  Max size of kernel argument:			 3840
  Alignment (bits) of base address:		 1024
  Minimum alignment (bytes) for any datatype:	 128
  Single precision floating point capability
    Denorms:					 Yes
    Quiet NaNs:					 Yes
    Round to nearest even:			 Yes
    Round to zero:				 No
    Round to +ve and infinity:			 No
    IEEE754-2008 fused multiply-add:		 No
  Cache type:					 Read/Write
  Cache line size:				 64
  Cache size:					 262144
  Global memory size:				 25157853184
  Constant buffer size:				 131072
  Max number of constant args:			 480
  Local memory type:				 Global
  Local memory size:				 32768
  Error correction support:			 0
  Unified memory for Host and Device:		 1
  Profiling timer resolution:			 1
  Device endianess:				 Little
  Available:					 Yes
  Compiler available:				 Yes
  Execution capabilities:				 
    Execute OpenCL kernels:			 Yes
    Execute native function:			 Yes
  Queue properties:				 
    Out-of-Order:				 Yes
    Profiling :					 Yes
  Platform ID:					 0x1c40f00
  Name:						 Intel(R) Core(TM) i7-4800MQ CPU @ 2.70GHz
  Vendor:					 Intel(R) Corporation
  Device OpenCL C version:			 OpenCL C 1.2 
  Driver version:				 1.2.0.92
  Profile:					 FULL_PROFILE
  Version:					 OpenCL 1.2 (Build 92)
  Extensions:					 cl_khr_icd cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_spir cl_intel_exec_by_local_thread cl_khr_depth_images cl_khr_3d_image_writes cl_khr_fp64 
```

When I compile and run this OpenCL demo (http://www.thebigblob.com/getting-started-with-opencl-and-gpu-computing/) everything functions as expected and the program works.

Note that I am using https://github.com/wiedemeyer/libfreenect2 51e13dba92c08ce88d342555f1a746d75a9c613d and not the primary repo.
However, when I try to compile and run Protonect or kinect2_bridge (https://github.com/code-iai/iai_kinect2/tree/master/kinect2_bridge, which links against the library) both fail to initialize OpenCL correctly. By this I mean that the following code returns zero devices and clGetPlatformIDs returns -1001 as an error. It does not appear to make a difference if the code appears as the first statement in main or if it is called later in the application.
```
  int error = clGetPlatformIDs(0, NULL, &num_platforms);
  std::cerr << "!!!: " << num_platforms << " - " << error << ": " << CL_SUCCESS << std::endl;

  std::cerr << "BEFORE!!!" << std::endl;
  std::vector<cl::Platform> platforms;
  cl::Platform::get(&platforms);
  std::cerr << "AFTER!!!: " << platforms.size() << std::endl;
```

I also did some digging and confirmed that both binaries are linking against the same OpenCL library.
Not sure if anyone can shed some light on this. Also not sure if this is related to #114
Thanks.

wiedemeyer, 2015-02-04:
Is there a file `*.icd` in `/etc/OpenCL/vendors/` that names the opencl lib?
I don't really know what the problem could be. With the nvidia opencl there is sometimes the need to call it the first time with sudo and afterwards it works without sudo.
But anyway, using the CPU for the depth processing will probably result in a low frame rate. I would recommend to use a GPU device. Have you tried to use the nvidia opencl?

mitchellwills, 2015-02-04:
Yes I have an intel64.icd file with the corresponding library. It appears to load correctly in the example referenced in my previous comment, which is why i'm so confused. I have tried to use nvidia card, but it doesn't seem to detect/load correctly (doesn't show up in clinfo). I am also doing development for a system that does not have an nvidia graphics. I may try to use Beignet; however, at this point I have been unsuccessful with using it as described in #109.

floe, 2015-02-10:
Did you try just installing beignet from the Ubuntu repository? The version is somewhat outdated, but maybe it might work around the permission issue?

mitchellwills, 2015-03-04:
Hmm it looks like for some weird reason the Intel CPU OpenCL library just isn't loaded for some reason, but I did get Beignet and the NVIDIA OpenCL libraries working. It must just not want to use CPU implementations for some reason

closed 2015-03-04 by mitchellwills

ipekbocegi, 2015-04-14:
Hi,
Same problem in #114 which leads me here. How did you solve the issue?
When I check /etc/OpenCL/vendors/ I see "nvidia.icd" which includes "libnvidia-opencl.so.1" line in it.
I use Nvidia-346 now (verified in additional drivers window) but I have originally Intel.
* $ lspci -k
00:02.0 VGA compatible controller: Intel Corporation Broadwell-U Integrated Graphics (rev 09)
	Subsystem: ASUSTeK Computer Inc. Device 237a
	Kernel driver in use: i915

* $ rosrun kinect2_bridge kinect2_bridge
modprobe: FATAL: Module nvidia not found.
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/ipek/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[OpenCLDepthPacketProcessor::init] ERROR: clGetPlatformIDs(-1001)
terminate called after throwing an instance of 'cl::Error'
  what():  clGetPlatformIDs
Aborted (core dumped)

* When I 
$ sudo -i
 /home/milvus-ipek/libfreenect2/examples/protonect# ./bin/Protonect
modprobe: FATAL: Module nvidia not found.
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 9 usb devices connected
[Freenect2Impl] found valid Kinect v2 @2:3 with serial 508765442542
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 01 12 0f 47 4d 41 39 38 33 2e 31    31332e2e2e2e2e2e474d413938332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 508765442542
device firmware: 4.3.3912.0.7
[DepthPacketStreamParser::handleNewData] not all subsequences received 0
init done 
opengl support available 
[OpenGLDepthPacketProcessor] avg. time: 5.85551ms -> ~170.779Hz
[TurboJpegRgbPacketProcessor] avg. time: 25.3486ms -> ~39.4499Hz
[OpenGLDepthPacketProcessor] avg. time: 6.02383ms -> ~166.007Hz
[TurboJpegRgbPacketProcessor] avg. time: 25.1904ms -> ~39.6976Hz
[RgbPacketStreamParser::handleNewData] skipping rgb packet!

I see RGB output but not IR or Depth.

What am I still missing? Please help me

wiedemeyer, 2015-04-14:
Do you have a nvidia graphics card in your machine or just the Intel HD?

ipekbocegi, 2015-04-14:
An hour ago I installed Nvidia-346 driver and tried running kinect2_bridge and ./bin/Protonect and got the outputs above. Then I restart my Asus X555ln laptop( a green sticker on it says Nvidia geforce 840M). I got black screen on startup. I uninstall all the nvidia-* and get my screen back. Now It seems I have Xorg thing. So am I using Intel right now? (Sorry for confusion I am newbie for these issues, just trying to do the suggestions)

wiedemeyer, 2015-04-14:
Instead of using nvidia you could try to install beignet to use the integrated GPU. Or you could read on how to setup the nvidia card on ubuntu for your laptop, or maybe use bumblebee for multi GPU configuration.

michaelwolfram, 2015-07-08:
I just want to leave my comment and how I made it work. Maybe it helps somebody else.
- Dell Latitude E37450, hybrid graphics (integrated Intel GPU + NVidia GeForce 840M), nvidia-331, nvidia-prime, nvidia-opencl-dev (I hope I didn't forget anything important here)
- Protonect with "cpu" and "gl" did work after installing nvidia-prime. However, "cl" option gave the error mentioned ("[OpenCLDepthPacketProcessor::init] ERROR: clGetPlatformIDs(-1001)"). As wiedemeyer said, running it once with sudo made it work.

wiedemeyer, 2015-07-08:
I searched for the error and found out that it is related to `/dev/nvidia-uvm`, which is not created on startup. As soon as it is created, OpenCL runs without root rights.