OpenKinect/libfreenect2 #154: Leap + Kinect2

giacomodabisias, 2015-02-05:
Hello,
I know this is a very specific question, but maybe you have some idea where my problem could come from. 
I can use correctly the kinect2, but when I attach a Leap motion sensor (usb2) to the computer (even on a different usb card), the Kinect2 stops working until I remove it. I get the following message over and over again:

[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[DepthPacketStreamParser::handleNewData] not all subsequences received 2
[DepthPacketStreamParser::handleNewData] not all subsequences received 64
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 8
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 260
[DepthPacketStreamParser::handleNewData] not all subsequences received 73
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 8
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 256
[DepthPacketStreamParser::handleNewData] not all subsequences received 128
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 256
[DepthPacketStreamParser::handleNewData] not all subsequences received 128
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] working buffer full, resetting it!
[DepthPacketStreamParser::handleNewData] remaining data should be 8 bytes, but is 11264
[DepthPacketStreamParser::handleNewData] image data too short!

I think its a problem with the linux usb driver; the two devices should run on different lanes on the pci-express so they shouldn't have bandwidth problems.
I just wanted to be sure that it is not a possible bug in libfreenect2.

floe, 2015-02-10:
Which kernel are you using? For me, a lot of USB-related problems went away once I switched to 3.16 (or later).

giacomodabisias, 2015-02-11:
I am using 3.13; I just tried out 3.19 but still same problem; as soon as I connect the leap sensor the kinect2 stops working

floe, 2015-02-11:
Very weird... unfortunately, I think this is outside of the scope of libfreenect2, it sounds more like a Linux issue. There's a parameter to increase USB kernel memory referenced in #97 - perhaps that's worth a try?

giacomodabisias, 2015-02-11:
I'll have a look! Thanks

closed 2015-05-21 by floe