OpenKinect/libfreenect2 #155: Turning down IR intensity

jeppewalther, 2015-02-05:
Have anyone found / investigated a way to turn down the intensity of the IR light source. In order to achieve a near field mode. 

I am currently experimenting with adding physical filters in front of the light source witch seems to work. But it would be much cleaner to simply do this in software.

christiankerl, 2015-02-07:
There was a email by Microsoft announcing an experimental near range firmware to be testable at some hackathon. So I guess this requires a different firmware.

xlz, 2016-01-27:
Also see alternative discussion on turning it off #338.

closed 2016-01-27 by xlz

floe, 2017-04-13:
Coming back to this old thread, did anyone ever actually have access to this rumoured near-mode FW?

fred-dev, 2019-03-25:
> Have anyone found / investigated a way to turn down the intensity of the IR light source. In order to achieve a near field mode.
> 
> I am currently experimenting with adding physical filters in front of the light source witch seems to work. But it would be much cleaner to simply do this in software.

I also looking for some kind of solution for this, can you share what kind of physical filters you are using to achieve this? Primarily I am using the kinect with an HTC vive and getting a lot of interference..

floe, 2019-03-25:
What I used at one point was simply a stack of 1-3 post-it notes in front of the IR emitter. Worked surprisingly well.