OpenKinect/libfreenect2 #156: A question about Auto Gain Control of Kinect 2 driver.

enddl22, 2015-02-06:
Hello, forks.
I would like to ask about Auto Gain Control of Kinect 2 driver. Are there any ways of controlling AGC or AEC of Kinect2? I want to turn off these functions since they change illumination that makes life more challenge. It would be great if anyone has tried this before or has any suggestions for it.

floe, 2015-02-10:
As far as I know, we don't have any background info on the internals of the RGB camera (i.e. commands, registers etc.). Does the official SDK export any related functions? If it doesn't, then it will be very difficult to determine how to access these functions via libfreenect2.

JoshBlake, 2015-02-10:
The official SDK does not support any type of configuration of the sensor,
although this is more a design decision than a hardware capability. The SDK
contains a service that allows multiple apps to connect, and each apps
shares the sensor without owning it. Thus, the SDK does not allow one app
to configure anything that would change the hardware or the expectations
from another app.

stfuchs, 2015-09-08:
Apparently the official SDK seems to offer some functions to retrieve the RGB camera settings:
https://msdn.microsoft.com/en-us/library/microsoft.kinect.kinect.icolorcamerasettings.aspx
Is there a way to get these for libfreenect2 as well?

floe, 2015-09-09:
Not at the moment - we'd need to know what USB command is sent by the official SDK to retrieve these settings.

stfuchs, 2015-09-10:
Do you guys have any documentation how you figured out the other USB commands? Or do you have any general reference? Maybe I'll find some time to play around. Btw, great work so far :)

floe, 2015-09-10:
There are two possible ways:
* if you're running Windows 8/10 natively, you can use MMA to capture USB traffic: https://msdn.microsoft.com/en-us/library/windows/hardware/dn741264%28v=vs.85%29.aspx
* if you're running Windows in a VM (e.g. VirtualBox), route the Kinect USB connection to the VM and then use Wireshark on the host to capture USB traffic: https://wiki.wireshark.org/CaptureSetup/USB

christiankerl, 2015-09-10:
I think the information is in the rgbpacketfooter seehttps://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/src/rgb_packet_stream_parser.cpp you just need to extend the api to pass this information to the frame listened...

floe, 2015-09-10:
Ah, so maybe `unknown3` is the gamma value?

stfuchs, 2015-09-11:
Cool. It would be great if we could pass on `exposure` and `gain` of `RgbPacketFooter` to the listener. Are you sure `unknown3` is always `1.0f`? I would have expected the gamma value to be around `1./2.2` (so `0.4545...`).

I'd be happy to change the API but I don't want to break anything. One way would be to add them to [frame struct](https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/include/libfreenect2/frame_listener.hpp#L37). But this would require to set some default values in case of depth and ir.
We would also need to pass them to the [processor](https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/src/rgb_packet_stream_parser.cpp#L146-L154) and extend the [RgbPacket](https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/include/libfreenect2/rgb_packet_processor.h#L40).

floe, 2015-09-11:
No worries about breaking anything. Create a fork of libfreenect2 and a new branch (perhaps "rgb_image_info" or something), code things up as they work for you, and submit a pull request. We'll discuss things in the PR afterwards.

Adding them to the frame struct is probably the best way to go for now. 1/30 for exposure and 1.0 for gain are probably sane default values that fit for all frames, regardless of type.

stfuchs, 2015-11-10:
Sorry for the delay. I was finally able to do some testing on my own.
The values for `unknown1`, `unknown2` and `unknown4` indeed do not change and seem to be zero all the time. However `unknown3` ranges from 1.0f to 6.40000009537f when the lens is fully covered. This seems very likely for a gamma value. I'll quickly prepare a PR and we can discuss things there.

closed 2016-01-27 by xlz