OpenKinect/libfreenect2 #157: Fails to build from source on ARM introduced in #103

xlz, 2015-02-07:
```
Linking CXX shared library ../lib/libfreenect2.so
CMakeFiles/freenect2.dir/src/rgb_packet_stream_parser.cpp.o:(.rodata+0x0): multiple definition of `typeinfo name for libfreenect2::PacketProcessor<libfreenect2::RgbPacket>'
CMakeFiles/freenect2.dir/src/packet_pipeline.cpp.o:(.rodata+0x0): first defined here
CMakeFiles/freenect2.dir/src/rgb_packet_stream_parser.cpp.o:(.data.rel.ro+0x0): multiple definition of `typeinfo for libfreenect2::PacketProcessor<libfreenect2::RgbPacket>'
CMakeFiles/freenect2.dir/src/packet_pipeline.cpp.o:(.data.rel.ro+0x0): first defined here
[...]
collect2: error: ld returned 1 exit status
```

This bug is introduced in #103. [Here](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=608029) is a better background for this bug. Basically, these two explicit instantiations
```cpp
// explicit instantiation and export to make vsc++ happy
template class LIBFREENECT2_API PacketProcessor<DepthPacket>; 

// explicit instantiation and export to make vsc++ happy
template class LIBFREENECT2_API PacketProcessor<RgbPacket>;
```
in headers being included multiple times violates C++ standard 14.7 clause 5:

 > an explicit instantiation definition shall appear at most once in a program

and GCC on ARM fails because of optimizations assuming the standard.

christiankerl, 2015-02-07:
so what do you suggest? Is wrapping the explicit instantiations in an ifdef for msvc an option?

xlz, 2015-02-07:
@christiankerl Yes we may need a msvc specific flag but I don't have msvc to test it.
@larshg Can you look into this?

emrainey, 2015-02-17:
I would recommend
 #if defined(_MSC_VER)  
 #endif
 From https://msdn.microsoft.com/en-us/library/b0084kay.aspx

closed 2015-04-22 by xlz