OpenKinect/libfreenect2 #159: waitForNewFrame() deadlock

DjLysy, 2015-02-09:
Hi guys,
I'm having this wired deadlock in Protonect and my own code. I'm using a build with enabled C++11 stdlib threading library and OpenCLpacket processor.

Currently, I have 11 threads running in Protonect, of whitch 6 are used by OpenCL library, 2 for PacketProcessors, 2 for USB communication and last main thread.
Deadlocked threads have stopped at:

WAIT_CONDITION(impl_->condition_, impl_->mutex_, l); at frame_listener_impl.cpp line 102, called from waitForNewFrame(). (main thread)
WAIT_CONDITION(packet_condition_, packet_mutex_, l); at async_packet_processor.h, line 100, called from libfreenect2::AsyncPacketProcessor<libfreenect2::RgbPacket>::static_execute
WAIT_CONDITION(packet_condition_, packet_mutex_, l); at async_packet_processor.h, line 100, called from libfreenect2::AsyncPacketProcessor<libfreenect2::DepthPacket>::static_execute

It looks like nothing is waking them up :( But with a little bit of luck, it is possible to run code without deadlock occured. 
Console output:
[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 506833342542
device firmware: 2.3.3912.0.7
and freeezee...

Furthermore, I was able to find out that each thread is actually entering this condition, but only once per thread at the beginning. Sometimes, random packet processor is starting correctly, but the other one is stucked, resulting in no image of any packet processor at the screen.
My hardware is Alienware M14x with NEC USB 3.0 chip so I think it shouldn't be a problem since it sometimes works.
I'm running Ubuntu 14.04 with kernel version: 3.13.0-43-generic and problem is present since i've started working with libfreenect2 in october 2014, and still present in commit c38a321e96

floe, 2015-02-10:
A lot of USB-related problems are fixed with kernel 3.16 or later, maybe you can try upgrading? (apt-get install linux-image-generic-lts-utopic)

arne48, 2015-02-11:
I was playing around a little today. Because we have on Notebook which is just working as a charm but all the others of us show exactly the behavior you mentioned. Till now I didn't figured out what the difference of the the notebooks is nor what the specific reason for the deadlock.
But I found a very dirty quickfix which enables me at least use Protonect for testing purposes.
I just added a 2 seconds sleep after the openDefaultDevice and another 5 seconds sleep after the dev->start(). I know this is really dirty but it keeps libfreenect2 from deadlocking.

DjLysy, 2015-02-16:
Sorry for long feedback, but I finally managed to test your proposed solutions.

First, after installing new kernel, my graphics driver stopped to cooperate with OpenCL and OpenGL processors, throwing -1001 error in OpenCL and not creating OpenGL context. In CPUDepthProcessor, still getting deadlocks, so I suppose this isn't an USB related problem.

In the other hand, sleep() workaround is somehow looking to work for now.. It's ugly but it's working ;)

DjLysy, 2015-02-16:
I'm taking back my words, sleep isn't working in my application :( 
I did some code research and I have some doubts in async_packet_processor.h in method execute() line 100.

There is WAIT_CONDITION(packet_condition_, packet_mutex_, l);
but there is no corresponding NOTIFY_CONDITION if anything else was trying to access to protected data. Is this correct?

floe, 2015-02-17:
I think you can disable the C++11 stdlib threading and use a minimal internal thread implementation, does that change anything?

christiankerl, 2015-02-17:
the process method notifies the condition variable, see https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/include/libfreenect2/async_packet_processor.h#L77

DjLysy, 2015-04-12:
Yaay!! :-) I've found the solution!
The problem is that libusb from depends folder isn't installing to  LIB directory when "make install" is executed, resulting in using old version of libusb-1.0.0. 
I have copied it to my lib directory, which is added to $PATH env variable, and worked smoothly from the first time.

Solution? -add libusb installation to cmake script

closed 2015-05-21 by floe