OpenKinect/libfreenect2 #163: Depth problem with OpenCL processor

xlz, 2015-02-12:
The depth result is blank with this config for OpenCL processor:
```
EnableBilateralFilter = false;
EnableEdgeAwareFilter = true;
```

It seems the edge aware filter `filterPixelStage2` uses `max_edge_test` which is generated in the bilateral filter `filterPixelStage1`.

Is this a bug or a limitation in feature?

wiedemeyer, 2015-02-13:
I think it is a feature. The edge aware filter is using results of the the bilateral filter. It is also a problem of the opengl processor. Either we should disable edge aware filter if bilateral filter is disabled or add code that calculates this max edge data if the bilateral filter is disable. But how likely is it that someone disabled bilateral filter but still wants to use the edge aware filter?

MichaelKorn, 2015-08-31:
For me this is a bug. It should be easy to generate a warning.