OpenKinect/libfreenect2 #164: New libusb version not working on AMD

wiedemeyer, 2015-02-12:
I just tried out the updated libusb from #149. On the Intel board it is working fine, but on an AMD board it justs says `working buffer full, resetting it!` and is not receiving depth frames anymore.

xlz, 2015-02-12:
#149 is not really changing anything. `working buffer full, resetting it` means the magic footer is not found through `512*424*11/8*2` bytes, but it should within `512*424*11/8` bytes. So I guess you are getting packet loss. Maybe you can add some logging code in `DepthPacketStreamParser::onDataReceived` and see how the transfers look like.

wiedemeyer, 2015-02-13:
I know, I don't think that it is a problem of libfreenect2, but maybe some regressions in libusb. With the previous version of libusb receiving depth images works, but as soon as I check out the more recent master version where MAX_ISO_BUFFER_LENGHT is changed, no frames are received.

xlz, 2015-02-13:
What about a bisection of libusb?

wiedemeyer, 2015-02-13:
what do you mean with bisection of libusb?

xlz, 2015-02-13:
git bisect
Start with a good commit, Go to a bad commit, and git bisect will tell you which commit to test, then you identify a particular commit...

floe, 2015-02-17:
So since we know that v1.0.19 was OK, and 51b10191033ca3a3819dcf46e1da2465b99497c2 is not, you would do something like the following in the depends/libusb folder:

    git bisect start
    git bisect bad
    git bisect good v1.0.19

git will checkout a version in between the two commits, then you build & test that version and run `git bisect {good|bad}` afterwards, depending on whether it worked or not. After a few iterations, git will (hopefully) tell you the exact commit which causes the issue.

wiedemeyer, 2015-02-17:
Thanks for the explanation. I didn't know this functionality before. I will try it out.

wiedemeyer, 2015-02-18:
I did the git bisect and it turns out that exactly this commit introduced the problem.

The `MAX_ISO_BUFFER_LENGTH` was changed to `49152 * 128`, but as soon as I change it to just `49152` it works again. Any value higher than `67583` will not work.

christiankerl, 2015-02-18:
OK so I guess we have to revert this change to libfreenect2, submit a bug report to libusb and re-apply it after they've fixed it

floe, 2015-02-19:
Does libusb HEAD work by any chance, i.e. does a later commit maybe fix this anyway?

wiedemeyer, 2015-02-19:
I also tried HEAD, but that does not help. I also tried out a 3.18 kernel instead of the 3.16, but this does not solve the problem either.

larshg, 2015-02-19:
@wiedemeyer  what is the actual transfers you get in each "packet"? 49152 * 128 seems like a awful lot of bytes in one transfer? - On windows its 33792 bytes.

I guess it shouldn't matter how many bytes is transfers per "packet" as along as the whole packet is received?

Maybe you could try my go (#75) at the depth packet parsing or @xlz go (I have a branch with his changes also)?

christiankerl, 2015-02-19:
49152 * 128 is just an upper bound used by libusb for some internal checks

xlz, 2015-02-26:
#174 seems to be another confirmation of this regression.

floe, 2015-05-21:
Still an issue with the new parser merged?

wiedemeyer, 2015-05-21:
I will check next week.

wiedemeyer, 2015-05-27:
So checked it today.
Depth images are only received if I set `MAX_ISO_BUFFER_LENGTH` to `49152`. Then it gives allot messages like `[DepthPacketStreamParser::onDataReceived] not all subsequences received ...`, but at least it receives depth images with ~20Hz.
If `MAX_ISO_BUFFER_LENGTH` is set to `49152 * 128` nothing is received at all and no messages are printed.
With libusb master, not even color images are received.

xlz, 2015-05-27:
How does the old stream parser work? `not all subsequences received` is not ideal.

xlz, 2015-06-09:
It seems we need to reinstate the old patch of 49152 for AMDs.

wiedemeyer, 2015-06-09:
At least on the one AMD machine we use. I don't know if it is true for all.

For example, we had another AMD machine that was getting 30 fps without any warnings or lost packages. But the new machine (same CPU, chipset, RAM, but different mainboard) had this issues.

@xlz to your previous question. The old stream parser also produced those messages. The frame rate is similar between the new and old implementation.

floe, 2015-11-09:
Is this fixed with 0.1-rc1?

closed 2015-11-09 by floe