OpenKinect/libfreenect2 #165: 'OpenCL/cl.hpp' file not found

jessekirbs, 2015-02-12:
Everything goes smooth until 'make && make install'. I get this error:

```
/Users/jessekirberger/Documents/Kinect/libfreenect2/examples/protonect/src/opencl_depth_packet_processor.cpp:43:10: fatal error: 
      'OpenCL/cl.hpp' file not found
#include <OpenCL/cl.hpp>
         ^
1 warning and 1 error generated.
make[2]: *** [CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2
```

Any ideas? I don't have a folder called 'OpenCL' anywhere, but I do have the OpenCL.frameworks folder.

larshg, 2015-02-12:
What GPU are you using and driver/Opencl sdk?

jessekirbs, 2015-02-12:
@larshg NVIDIA GeForce GT 650M 512 MB. Not really sure what OpenCL SDK I'm using.. how do I check that? How would I disable the OpenCL and use OpenGL? I'm fairly new to all this. Thanks for the reply.

larshg, 2015-02-13:
If you use CMake GUI you can go under ENABLE and deselect ENABLE_OPENCL.
If you use CMake in a command prompt I'm not entirely sure how it should be passed - maybe like this: cmake -DENABLE_OPENCL=OFF - or something like it :)

Otherwise you could try add the following:

#define CL_USE_DEPRECATED_OPENCL_1_1_APIS
+#include <CL/cl.h>
+#undef CL_VERSION_1_2

in opencl_depth_packet_processor.cpp - instead of the #include <OpenCL/cl.hpp>

As far as I know - Nvidia doesn't support the CL version 1.2 yet and on my machine with nVidia, the cl.hpp file wasn't there.

I haven't tested the part about OpenCL includes, so I might be totally off track here :)

jessekirbs, 2015-02-13:
@larshg You are legend, my friend. 'cmake -DENABLE_OPENCL=OFF' did the trick and it installed correctly. Thank you so much. Now it's having trouble finding the Kinect. Does it ever end? 

```
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 13 usb devices connected
[Freenect2Impl] found 0 devices
[Freenect2Impl] requested device 0 is not connected!
no device connected or failure opening the default one!
```

jessekirbs, 2015-02-13:
I'm getting a black screen on the ir and depth windows. Could this be from turning OpenCL off?

larshg, 2015-02-13:
Nice to hear :-)  - It should be possible to use the opencl as well, if you make the change I wrote earlier, but it's being worked on in #167 - so you can also wait for that to be merged!

jessekirbs, 2015-02-13:
@larshg Any idea why ir and depth windows are black? How do I access these?

larshg, 2015-02-13:
@jessekirbs  Unfortunately I have no idea :( - its being worked on in  #31.

I'm trying to look into getting OpenCL working on my nvida machine also - I'll post my result when I'm done :)

Meanwhile you can add the -DENABLE_OPENGL=OFF also - which the makes it use the CPU.. It is slow and make the processer do a lot, but hopefully you should get images and can work with that further ?

jessekirbs, 2015-02-18:
Tried running 'cmake -DENABLE_OPENGL=OFF' and I am still getting black screens for IR and Depth.

floe, 2015-05-21:
Is this fixed with current master?

xlz, 2015-06-07:
@floe For the bug in the title, yes. I have no problem using OpenCL on a certain Mac. `OpenCL/cl.hpp` is downloaded in `install_mac.sh`. The OpenGL black output issue is in other ones.

floe, 2015-06-07:
Thanks for the update, closing.

closed 2015-06-07 by floe