OpenKinect/libfreenect2 #172: Depth image is mirrored?

xionglingfeng, 2015-02-24:
Hi all,
It seems the depth image read from libfreenect2 is a mirrored one... Is this a normal behavior?
The formula which used to calculate the coordinate in world coordination system:
```C++
Point3d(
	(target.x - ir_params.cx) * depth / ir_params.fx,
	(target.y - ir_params.cy) * depth / ir_params.fy,
	depth
	)
```
is applied to the mirrored image, or a normal image?

DjLysy, 2015-02-26:
I have the same issue. For me all images are mirrored, not only the depth image

wiedemeyer, 2015-02-27:
Yes, all images are mirrored. You can flip them using OpenCV: `cv::flip(image, mirrored, 1);`

@christiankerl: Is it the intended behavior, does the Windows SDK do the same?

christiankerl, 2015-02-27:
afaik yes

zardchim, 2015-05-26:
In Kinect 2 SDK we can get the world coordinate by input the image pixel coordinate and depth.

Is there any functions in libfreenect2 achieve similar function? Or we need to do the calculation as mentioned above by ourself? Sorry for my newbie question -,-"

ahundt, 2015-12-12:
Presumably the point cloud is mirrored as well? Any advice to un-mirror that?

ahundt, 2016-01-31:
FYI my version of freenect2pcl grabber does un-mirroring for those that need an example now:
https://github.com/ahundt/libfreenect2pclgrabber/blob/master/k2g.h#L127

xlz, 2016-01-31:
The intention is to introduce some change in the pixel indexing in the depth processors then you get un-mirrored image for free.

ahundt, 2016-01-31:
Sounds ideal, I was just mentioning that in case someone didn't know how to do it, sorry if that just ended up being unnecessary noise.

martinruenz, 2018-01-05:
@xlz: Could you elaborate on:

> pixel indexing in the depth processors

Is this work in progress?

Also, the default behaviour of libfreenect2 (mirrored depth images) is rather unintuitive and it should definitely be documented (with high visibility).

xlz, 2018-01-05:
`mirrored_pixel[y,x] = pixel[y,512-x]` is free in GPU kernel. Probably not in progress because I don't have a device to test with recently.

The default behavior is to match the official SDK. It's hard to say which is more unintuitive.

hygxy, 2018-07-19:
@xlz  Can you please explain what is "free in GPU kernel" and "not in progress"? thanks in advance

hygxy, 2018-07-19:
@xlz  Can you please explain what is "free in GPU kernel"?  thanks in advance