OpenKinect/libfreenect2 #174: libusb acting up

tysonhammen, 2015-02-26:
This is a repost of something I originally posted over on iai_kinect2. I am posting here by suggestion of Thiemo. Hopefully you guys can help me out.

I had everything running fine then I had to shut down the machine to move it and when it came back up I now keep getting "[DepthPacketStreamParser::handleNewData] image data too short!" and "[DepthPacketStreamParser::handleNewData] working buffer full, resetting it!" Though I do still get RGB data just fine. Not sure if some libraries or other updates processed as a part of the reboot or not.

I thought that maybe connecting my v1 kinect had messed up the usb somehow so I completely rebuilt, as in reloaded the OS and updated everything, and am still getting the same issue. Have there been recent changes to this library that could be causing this or does anyone have any other ideas as to what is happening? 

For reference I have included my system setup and the following messages in my dmesg log.

System:
AMD A10 7850K APU on an MSI board with the A78 Chipset. I set the USB buffer to be ~4GB in the grub config file as I have 16GB available but this did not help either.
Kernel:3.16.0-31-generic #41~14.04.1-Ubuntu 

[ 2709.860400] xhci_hcd 0000:00:10.0: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[ 2800.750936] kinect2_bridge[29085]: segfault at 1c0 ip 00007f9c565cf4f8 sp 00007fff90c0fde0 error 4 in libglfw.so.3.0[7f9c565c8000+12000]
[ 2850.897495] usb 7-2.1: reset SuperSpeed USB device number 10 using xhci_hcd
[ 2850.913309] xhci_hcd 0000:00:10.0: xHCI xhci_drop_endpoint called with disabled ep ffff880415ef2c00

each time I launch protonect I see these:
[ 3923.204511] usb 7-2.1: reset SuperSpeed USB device number 10 using xhci_hcd
[ 3923.220227] xhci_hcd 0000:00:10.0: xHCI xhci_drop_endpoint called with disabled ep ffff880415ef2c00
[ 3923.220235] xhci_hcd 0000:00:10.0: xHCI xhci_drop_endpoint called with disabled ep ffff880415ef2c48
[ 3923.220239] xhci_hcd 0000:00:10.0: xHCI xhci_drop_endpoint called with disabled ep ffff880415ef2cd8
[ 3923.220243] xhci_hcd 0000:00:10.0: xHCI xhci_drop_endpoint called with disabled ep ffff880415ef2c90
[ 3923.246165] xhci_hcd 0000:00:10.0: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[ 3923.261067] xhci_hcd 0000:00:10.0: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[ 3923.265315] xhci_hcd 0000:00:10.0: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[ 3923.265508] xhci_hcd 0000:00:10.0: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[ 3923.270280] xhci_hcd 0000:00:10.0: WARN Event TRB for slot 3 ep 2 with no TDs queued?

xlz, 2015-02-26:
I've also seen these warnings while running fine.

Have you checked your USB autosuspend settings? What is your output of
```
grep . /sys/bus/usb/devices/*/power/autosuspend
```
and `lsusb` when running protonect? Kinect2 devices should have autosuspend disabled.

Also, have you checked USB connectors here? [image]

tysonhammen, 2015-02-26:
I checked all the connections and plugged the kinect into my windows box to verify. When using the Windows SDK everything works fine.

Here is the output of  "grep . /sys/bus/usb/devices/*/power/autosuspend":
The Kinect is 7-2.1 according to the dmesg output 
/sys/bus/usb/devices/3-5/power/autosuspend:2
/sys/bus/usb/devices/6-2/power/autosuspend:0
/sys/bus/usb/devices/7-2.1/power/autosuspend:2
/sys/bus/usb/devices/7-2/power/autosuspend:0
/sys/bus/usb/devices/usb1/power/autosuspend:0
/sys/bus/usb/devices/usb2/power/autosuspend:0
/sys/bus/usb/devices/usb3/power/autosuspend:0
/sys/bus/usb/devices/usb4/power/autosuspend:0
/sys/bus/usb/devices/usb5/power/autosuspend:0
/sys/bus/usb/devices/usb6/power/autosuspend:0
/sys/bus/usb/devices/usb7/power/autosuspend:0
/sys/bus/usb/devices/usb8/power/autosuspend:0
/sys/bus/usb/devices/usb9/power/autosuspend:0

and here is the lsusb output:
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 009 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 003: ID 045e:02c4 Microsoft Corp. 
Bus 007 Device 002: ID 045e:02d9 Microsoft Corp. 
Bus 007 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 006 Device 002: ID 045e:02d9 Microsoft Corp. 
Bus 006 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

tysonhammen, 2015-02-26:
Also I tried to use this command "echo -1 > /sys/bus/usb/devices/7-2.1/power/autosuspend" that did not seem to make a difference and was overwritten on reboot of the machine.

xlz, 2015-02-26:
Try `sudo sh -c 'for i in /sys/bus/usb/devices/*/power/autosuspend; do echo -1 >$i; done'` after connecting before running protonect?

tysonhammen, 2015-02-26:
Unfortunately I get the same issue.

tysonhammen, 2015-02-26:
could this be related to #164 ?

xlz, 2015-02-26:
I noticed your system is also AMD. So revert 7936c4171284e6856cb502f1c8434284832369fb, clear depends directory, rerun ./install_deps.sh, and rebuild?

tysonhammen, 2015-02-26:
sorry, I am not well versed in how to pull specific revisions from git.. any way you could give me a bit more detail on how to do the above? I can do the install part but ensuring I have the right code is the part I would need help on.

xlz, 2015-02-26:
```
git reset --hard c38a321e96df9d9c0143739e2f26821cd262908b
```
to revert.

tysonhammen, 2015-02-26:
Here is what I get for output.. ran that command from the libfreenect2 directory that was cloned from git. Does this look right?

git reset --hard c38a321e96df9d9c0143739e2f26821cd262908b
HEAD is now at c38a321 Merge pull request #158 from floe/registration

xlz, 2015-02-26:
Yes, everything in your local repo is one commit before 7936c41.

tysonhammen, 2015-02-26:
Thanks! so what do I need to clear out of the depends directory? obviously I can't clear everything or the install scripts will be gone as well.

xlz, 2015-02-26:
All directories under ./depends/ are generated. `rm -rf` just them would be enough.

tysonhammen, 2015-02-26:
that fixed that portion! I still do not see depth data but I suspect that is an OpenGL issue.  Thanks for your help! and if you have any hints on the OpenGL stuff with AMD APUs I would greatly appreciate that as well! Thanks again!

xlz, 2015-02-26:
So this is another confirmation of the regression in #164.

You can definitely use OpenCL acceleration with AMD much better than the OpenGL depth processor.

tysonhammen, 2015-02-26:
Is that specified as a part of the build or as an argument at runtime?

xlz, 2015-02-26:
If you have OpenCL SDK installed you should see something about OpenCL during `cmake ..`. http://wiki.tiker.net/OpenCLHowTo looks like a good guide for configuring OpenCL.

tysonhammen, 2015-02-26:
it appears to be using CPU at the moment but still isn't publishing any depth data or IR

tysonhammen, 2015-02-26:
Installed the SDK and then ran the cmake but got this about OpenCL.. 

-- Could NOT find OpenCL (missing:  OPENCL_INCLUDE_DIRS)

wiedemeyer, 2015-02-27:
Hi, for OpenCL you have to install `opencl-headers`. And if you are using the latest catalyst, 14.12, then you should check `/usr/lib` or `usr/lib/x86_64...`. There should be a `libOpenCL.so` (or similar), for me the symlink from `libOpenCL.so.1` to `libOpenCL.so` was missing.

tysonhammen, 2015-02-27:
@xlz don't suppose you would have any ideas as to why Protonect would run but only TurboJpeg would be outputting? If I run @wiedemeyer's kinect2_bridge I get RGB and IR data but no depth. It appears when running Protonect that the depth data is either not being received or sent. When launching kinect2_bridge it identifies OpenCL as the depth processor. I don't see anything when launching protonect that identifies whether or not it can see an OpenCL or OpenGL device.

tysonhammen, 2015-02-27:
@xlz I may have found another issue with this.... I have been trying some different things building libfreenect2 and kinect2_bridge and if I leave the "linux_usbfs_increase_max_iso_buffer_length.patch" file in the depends directory I get the same error I did before. If I remove it then everything builds and protonect runs but I get no depth data. Is this file from before or after the c38a321 build?

tysonhammen, 2015-02-27:
when I do the "git reset --hard c38a321e96df9d9c0143739e2f26821cd262908b" this file gets pulled to the depends directory.

xlz, 2015-03-02:
@tysonhammen 
Okay, I have been busy last two days for the IROS deadline.

If you are not getting depth output with Protonect, please attach the console output if there is any, and also related messages in dmesg, and check sysfs USB autosuspend settings. If the issue is with iai_kinect2, I have yet to familiarize myself with that.

If you have trouble with versions, you can always delete everything, re-clone, do the hard reset, and rebuild.

tysonhammen, 2015-03-02:
I think this is an AMD chip set issue. I got tired of bothering with it and
got an Intel board and processor so that I could move forward with my
project.

floe, 2015-04-01:
So can this be closed?

tysonhammen, 2015-04-02:
For me, yes. But that is just because I switched to Intel in order to avoid
the bug. If you want the bug fixed then further testing would be required.

floe, 2015-04-02:
I will close it for now, as these issues are very hard to debug - they only appear with some specific combination of kernel, USB controller, libusb(x) and perhaps even UEFI firmware. We will try to track this down in #164 (apparently it's the same issue).

closed 2015-04-02 by floe