OpenKinect/libfreenect2 #175: Accelerometer data?

iamnotnader, 2015-03-01:
Was wondering if accelerometer data is/will be made available and, if so, when?

Looking at libfreenect, it looks like it's available through the audio interface, which isn't supported in libfreenect2 currently.

See update_tilt_state_alt from libfreenect(1) below:
https://github.com/OpenKinect/libfreenect/blob/da2d8ba28d6f1eb39c4841781355b48ca1b25259/src/tilt.c

Love this lib, by the way.

iamnotnader, 2015-03-01:
This is where they open the device:
https://github.com/OpenKinect/libfreenect/blob/da2d8ba28d6f1eb39c4841781355b48ca1b25259/src/usb_libusb10.c#L417

This looks like the endpoint for the motor. But maybe it's different for v2?
https://github.com/OpenKinect/libfreenect/blob/da2d8ba28d6f1eb39c4841781355b48ca1b25259/src/freenect_internal.h#L140

iamnotnader, 2015-03-01:
Uh-oh.. I see the "protocol" section of the libfreenect documentation is empty-- not a good sign I suppose:

http://openkinect.org/wiki/Kinect_2

gaborpapp, 2015-03-01:
The Kinect v2 has no tilt motor. It is not really needed because of the bigger field of view.

iamnotnader, 2015-03-01:
Ah, I see.

Do you know if anyone's documented all of the endpoints that ARE available for the v2, similar to the way they did for the v1 below?

http://openkinect.org/wiki/USB_Devices

xlz, 2015-11-16:
It is possible there is indeed accelerometer data. KinectService.exe has a dedicated function that does something and uses a string literal "SensorAccelerometerData". If there is, the most likely place is in DepthSubPacketFooter.

daxav, 2017-01-19:
Hello guys.
I was just wondering if there is anything new on this? I realize the thread is over a year old. Nonetheless, I haven't found any solutions on how to read the accelerometer data. I have read somewhere that best chance is to stay within the MS SDK... that is, of course, not a good solution at this point. So, any news?

floe, 2017-01-27:
This shouldn't be too difficult to figure out. As @xlz suggested, maybe just add a bit of code to hexdump the unknown fields in DepthSubPacketFooter (https://github.com/OpenKinect/libfreenect2/blob/83f88b4c09f0b00724ae65785abcd4f3eeb79f93/include/internal/libfreenect2/depth_packet_stream_parser.h#L45), record that data while holding the Kinect in different orientations, and compare the results. That should give us an idea where the data is actually hidden.

remmel, 2021-12-07:
I've spend 15min trying to understand the data, tilting the kinect by 90° without success :
```
footer: m0:0 m1:11 s:57 ss:3 fields: 275249 1634861113 4093624308 49140 2147516416 0 467 0 1270480916 167248331 1013778692 211 511 0 0 0 
footer: m0:0 m1:11 s:58 ss:3 fields: 275249 2274951226 438747162 49179 2147516416 0 464 0 1270480916 167313867 1013778691 211 511 0 0 0 
footer: m0:0 m1:11 s:59 ss:3 fields: 275249 2914975803 1078771776 49217 2147516416 0 467 0 1270480916 167313867 1013778691 211 511 0 0 0 
**I tilt the kinect ~90°**
footer: m0:0 m1:11 s:388 ss:3 fields: 275249 3041264004 1205072199 61768 2147516416 0 473 0 1270480916 166986187 1013778691 211 511 0 0 0 
footer: m0:0 m1:11 s:389 ss:3 fields: 275249 3681354117 1845162349 61806 2147516416 0 468 0 1270480916 167051723 1013778691 211 511 0 0 0 
footer: m0:0 m1:11 s:390 ss:3 fields: 275249 26411398 2485186964 61844 2147516416 0 472 0 1270480916 167051723 1013778691 211 511 0 0 0 
```

Add in https://github.com/OpenKinect/libfreenect2/blob/fd64c5d9b214df6f6a55b4419357e51083f15d93/src/depth_packet_stream_parser.cpp#L102 :
```
        if(footer->subsequence == 3)
        std::cout << "footer: m0:" << footer->magic0 << " m1:" << footer->magic1 << " s:" << footer-> sequence << " ss:" << footer->subsequence
        << " fields: " << footer->fields[0] << " " << footer->fields[1] << " "<< footer->fields[2] << " "<< footer->fields[3] << " "
                << footer->fields[4] << " "<< footer->fields[5] << " "<< footer->fields[6] << " "<< footer->fields[7] << " "
                << footer->fields[8] << " "<< footer->fields[9] << " "<< footer->fields[10] << " "<< footer->fields[11] << " "
                << footer->fields[12] << " "<< footer->fields[13] << " "<< footer->fields[14] << " "<< footer->fields[15] << " "
                << std::endl;
```