OpenKinect/libfreenect2 #176: "no device connected or failure opening the default one!"

MathiasGallardo, 2015-03-03:
Hi All,

I tried to install libfreenect2 for Ubuntu. I followed all your instructions and I got this error when I ran the program: "./Protonect"
************
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found 0 devices
[Freenect2Impl] requested device 0 is not connected!
no device connected or failure opening the default one!
************

Do you have some advices to solve this problem?
Thank you in advance,

Here, some informations:
Doing "lsusb"
************
Protocol spec without prior Class and Subclass spec at line 17406
Bus 001 Device 002: ID 8087:8000 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 413c:5534 Dell Computer Corp. 
Bus 003 Device 008: ID 045e:02d9 Microsoft Corp. 
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 007: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 002 Device 006: ID 413c:2107 Dell Computer Corp. 
Bus 002 Device 005: ID 413c:2134 Dell Computer Corp. 
Bus 002 Device 004: ID 0c45:64d2 Microdia 
Bus 002 Device 012: ID 045e:02d9 Microsoft Corp. 
Bus 002 Device 002: ID 413c:2513 Dell Computer Corp. internal USB Hub of E-Port Replicator
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
************
When I plug the Kinect, these two devices appear:
************
Bus 003 Device 008: ID 045e:02d9 Microsoft Corp. 
Bus 002 Device 012: ID 045e:02d9 Microsoft Corp. 
************
For the first device, using "lsusb -vv -s 008", I get:
bcdUSB               3.00
And, for the second, "lsusb -vv -s 012":
bcdUSB               2.10

About my laptop:
************
Description:	Ubuntu 14.04.2 LTS
************
Linux mathiasgallardo-Latitude-E7440 3.16.0-031600-generic #201408031935 SMP Sun Aug 3 23:36:11 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux
************
        *-display
             description: VGA compatible controller
             produit: Haswell-ULT Integrated Graphics Controller
             fabriquant: Intel Corporation
             identifiant matériel: 2
             information bus: pci@0000:00:02.0
             version: 0b
             bits: 64 bits
             horloge: 33MHz
             fonctionnalités: msi pm vga_controller bus_master cap_list rom
             configuration: driver=i915 latency=0
             ressources: irq:63 mémoire:f7800000-f7bfffff mémoire:e0000000-efffffff portE/S:f000(taille=64)
************
If you need more, do not hesitate!

gaborpapp, 2015-03-03:
Have you tried the workaround in #78?

MathiasGallardo, 2015-03-03:
Thanks for your answer :)

I've already tried #78 "Try unplugging Kinect from power source, while keeping the USB cable connected. ", then checked the devices information. It still remains what I noted in the additional informations:
"When I plug the Kinect, these two devices appear:

Bus 003 Device 008: ID 045e:02d9 Microsoft Corp.
Bus 002 Device 012: ID 045e:02d9 Microsoft Corp.

For the first device, using "lsusb -vv -s 008", I get:
bcdUSB           3.00
And, for the second, "lsusb -vv -s 012":
bcdUSB           2.10"

floe, 2015-03-24:
It's rather weird that the Kinect should show up on both the USB3 and the USB2 bus at the same time. There are some issues regarding the host controller, could you try a different one (e.g. by adding a PCIe card or testing on a different machine)?

MathiasGallardo, 2015-03-25:
We have tested on a different machine and we observed the same situation. Sometimes, we managed to get only the RGB frames and black windows for the IR and Depth. As we didn't have too much time to work on the Kinect (for our general works), we decide to switch to W10TP to use the Microsoft framework. Maybe, we will come back to libfreenect when some deadlines will be over ;)
Thanks for helping me :)

closed 2015-03-25 by floe

eschneider1992, 2015-03-28:
+1 Three devices appear for us, one is 3.00, the other two are 2.10

eschneider1992, 2015-03-28:
idProduct 0x2d9 shows up twice! one is the 3.00 one, the other the 2.10. Then 2d8 is also 2.10.