OpenKinect/libfreenect2 #178: PCL grabber

frederico00, 2015-03-07:
HI there,

I installed the driver with sucess, i i think that it is working properly (at leas i can visualize the clouds from the protonect example). 

I was trying to use a grabber from: https://github.com/giacomodabisias/libfreenect2pclgrabber

but in the Cmakelist-txt i have to make

 include_directories( ${GLEW_INCLUDE_DIRS})

that are in the following directory /depends/glew/lib64/libGLEW.so

but the glew directory doesn't exist in my driver directory.

Any idea?

giacomodabisias, 2015-03-23:
Hi, I did that grabber.
Did you run the correct script in /depends? You have then to pass to the cmake the directory of the includes (ex: libfreenect2/depends/glew/include/) doing cmake -DGLEW_INCLUDE_DIRS=path

frederico00, 2015-03-23:
Hi there again.

We have been talking about that, it is already working. I change the path to my root folder where i have the glew library.

thanks again

closed 2015-03-24 by floe