OpenKinect/libfreenect2 #181: Ideal FPS?

lolz0r, 2015-03-19:
Hi,
I have the driver up and working great! Awesome work! 

That being said quick question:
What is the ideal FPS for the xbox one kinect hardware when pulling both depth and RGB frames? Using code based off the protonect example I am clocking exactly 15FPS a second while not pegging out my CPU. I thought this hardware was 30FPS? Please advise. 

- Thanks Again!

wiedemeyer, 2015-03-19:
Is it like in this issue: https://github.com/OpenKinect/libfreenect2/issues/141?
Do you get a higher frame rate if you turn on the lights?

lolz0r, 2015-03-19:
Hi Wiedemeyer,
Thank you for the insight! After reading the thread you linked I did a few expirements with my kinect. Ultimently I shinned a flashlight directly into the camera and at that point it switched to 30FPS mode. Problem solved ;)

Thanks again and keep up the awesome work! :)

closed 2015-03-19 by lolz0r