OpenKinect/libfreenect2 #182: "Failed to submit transfer" in custom application

svope, 2015-03-23:
Hi guys,

I have a problem using libfreenect in my application.

The code:
bool protonect_shutdown = false;

void sigint_handler(int s)
{
  protonect_shutdown = true;
}
int main(int argc, char* argv[])
{
    cv::namedWindow( "rgb", cv::WINDOW_AUTOSIZE );// Create a window for display.

    libfreenect2::Freenect2 freenect2;
    libfreenect2::Freenect2Device *dev = freenect2.openDefaultDevice();

    if(dev == 0)
    {
      std::cout << "no device connected or failure opening the default one!" << std::endl;
      return -1;
    }

    signal(SIGINT,sigint_handler);
    protonect_shutdown = false;

    libfreenect2::SyncMultiFrameListener listener(libfreenect2::Frame::Color | libfreenect2::Frame::Ir | libfreenect2::Frame::Depth);
    libfreenect2::FrameMap frames;

    dev->setColorFrameListener(&listener);
    dev->setIrAndDepthFrameListener(&listener);
    dev->start();

    std::cout << "device serial: " << dev->getSerialNumber() << std::endl;
    std::cout << "device firmware: " << dev->getFirmwareVersion() << std::endl;

    while(!protonect_shutdown)
    {
      std::cout << "in while (!protonect_shutdown) loop " << std::endl;
      listener.waitForNewFrame(frames);
      libfreenect2::Frame *rgb = frames[libfreenect2::Frame::Color];
      libfreenect2::Frame *ir = frames[libfreenect2::Frame::Ir];
      libfreenect2::Frame *depth = frames[libfreenect2::Frame::Depth];

      cv::imshow("rgb", cv::Mat(rgb->height, rgb->width, CV_8UC3, rgb->data));
      cv::imshow("ir", cv::Mat(ir->height, ir->width, CV_32FC1, ir->data) / 20000.0f);
      cv::imshow("depth", cv::Mat(depth->height, depth->width, CV_32FC1, depth->data) / 4500.0f);

      int key = cv::waitKey(1);
      protonect_shutdown = protonect_shutdown || (key > 0 && ((key & 0xFF) == 27)); // shutdown on escape

      listener.release(frames);
    }

    dev->stop();
    dev->close();
    return 0;
}

Which is basically copy & paste from Protonect.cpp example. When I run this application the output is:

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 9 usb devices connected
[Freenect2Impl] found valid Kinect v2 @3:6 with serial 000393642047
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    ............C..A
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    .!3U... ........
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    ................
0x0040:  31 33 00 00 00 08 13 0b 47 4b 53 36 34 35 2e 31    13......GKS645.1
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                X...........

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  7f 26 00 00                                        &..

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  7f 26 00 00                                        &..

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 000393642047
device firmware: 4.1.3911.0.7
in while (!protonect_shutdown) loop 
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TurboJpegRgbPacketProcessor] avg. time: 24.3891ms -> ~41.002Hz
[TurboJpegRgbPacketProcessor] avg. time: 25.3013ms -> ~39.5237Hz

etc..

Where could be a problem? I run this app as root and the Protonect example runs without any problem. What am I missing?
Thank you very much!

eschneider1992, 2015-04-07:
I've got a similar problem, would be curious if anyone has a fix :)

floe, 2015-04-07:
Just had an idea: is your application linking to the patched libusb, or the default one from the system? Maybe you just need to adjust your LD_LIBRARY_PATH and/or linker flags?

eschneider1992, 2015-04-07:
@floe - that was my problem! I checked the libusb for the Protonect example and for my project.

ldd ./kratos_robot | grep usb
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f46678c9000)

ldd ./Protonect | grep usb
	libusb-1.0.so.0 => /home/kratos/Downloads/libfreenect2/examples/protonect/../../depends/libusb/lib/libusb-1.0.so.0 (0x00007f6519c0b000)

It may be a bit hacky (I'm actually not sure) but on a team member's advice I copied the libusb-1.0.so* files from libfreenect2/depends/libusb/lib to my project's lib folder and that fixed the issue.

wiedemeyer, 2015-04-20:
https://github.com/code-iai/iai_kinect2#modifications-to-upstream-libfreenect2
Add those modifications and it should work in external applications. It forces it to use the libusb from libfreenect2.

floe, 2015-04-22:
@wiedemeyer good idea, can you send a quick PR?

wiedemeyer, 2015-04-22:
We discussed it in #101, but I could create a pull request.

BenjaminRothschild, 2015-04-23:
have anyone solved it? I met the same problem, my OS is win8,
C:\Users\benjamin>E:\pro_VS\libfreenect2-master\examples\protonect\bin\Debug\Protonect.exe
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 9 usb devices connected
[Freenect2Impl] found valid Kinect v2 @3:5 with serial 500344540242
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    ............C..A
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    .!3U... ........
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    ................
0x0040:  31 33 00 00 00 00 08 10 47 4b 53 39 33 30 2e 31    13......GKS930.1
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                X...........

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  7f 2e 00 00                                        ...

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  7f 2e 00 00                                        ...

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
......
[TransferPool::submit] failed to submit transfer
[Freenect2DeviceImpl] started
device serial: 500344540242
d[eTvuircbeo JfpiergmRwgabrPea:c k4e.t1P.r3o9c1e1s.s0o.r7:
:doProcess] Failed to decompress rgb image! TurboJPEG error: 'Not a JPEG file: starts with 0xe9 0x06'
[TurboJpegRgbPacketProcessor] avg. time: 15.4169ms -> ~64.864Hz
[TurboJpegRgbPacketProcessor] avg. time: 15.5135ms -> ~64.4599Hz
[TurboJpegRgbPacketProcessor] avg. time: 15.5345ms -> ~64.3727Hz
[TurboJpegRgbPacketProcessor] avg. time: 15.5005ms -> ~64.5142Hz
......
anyone knows about it?
I want to know whether I build is wrong???

floe, 2015-04-23:
@wiedemeyer I think official packages are still some time away, so to deal with all those nasty runtime linking issues, I would suggest that we integrate this change?

Edwinzero, 2015-04-25:
I have exactly problem with you when I build libfreenect2 in WIN32. @BenjaminRothschild

zhangtemplar, 2015-05-13:
Hi, it still doesn't work though ldd seems to create the correct output:

ldd bin/Protonect | grep usb
	libusb-1.0.so.0 => /home/qiang/Documents/libfreenect2/examples/protonect/../../depends/libusb/lib/libusb-1.0.so.0 (0x00007feb6ea39000)

floe, 2015-06-07:
This should be fixed by #268, closing here. If it's still an issue, please reopen.

closed 2015-06-07 by floe

zhangtemplar, 2015-06-10:
To those still interested, please make sure your USB controller is compatible with Kinect 3. In fact, Kinect 3 is very picky, which only works with NEC/Renasas and Intel controller. Please make sure you are using one of them. After switching to one compatible controller, the code works nicely.

Please read the following for more details:
http://support.xbox.com/en-US/xbox-on-other-devices/kinect-for-windows/kinect-for-windows-v2-known-issues#026a8a6b31bb44a8ad832a9d9accd04c
http://social.msdn.microsoft.com/Forums/en-US/bb379e8b-4258-40d6-92e4-56dd95d7b0bb/confirmed-list-of-usb-30-pcie-cardslaptopsconfigurations-which-work-for-kinect-v2-during?forum=kinectv2sdk