OpenKinect/libfreenect2 #183: OpenCL gives white noise in depth buffer

julienaubert, 2015-03-23:
The first screen is using OpenGL (same result with CPU). The second screen is from when using OpenCL. 

[image]
[image]

wiedemeyer, 2015-03-24:
Hi,
I looked at it, and for me, the OpenCL version creates a bit more white noise than the OpenGL version, but not as much as in your images. Anyway, I took a closer look at the code and I found the possible cause for this problem in the bilateral filter.
I compared the [OpenGL code](https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/src/shader/filter1.fs#L95-L97) with the [OpenCL code](https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/src/opencl_depth_packet_processor.cl#L165) and saw that I mixed up the order of the arguments of the `select` statement.
Could you replace [line 165](https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/src/opencl_depth_packet_processor.cl#L165) by `dist_acc += select(dist, (float3)(0.0f), c1);` and try if it solves the issue for you? I will then prepare a pull request for it.

@christiankerl: While looking through the code I found some parts that could be improved.
For the bilateral filter:
- Shoudn't it only run when the full gaussian kernel fits in the image, similar to the edge aware filter `if(uv.x < 1 || uv.y < 1 || uv.x > 510 || uv.y > 422)`

For the edge aware filter:
- `FilterDepth = !max_edge_test_ok ? 0.0f : v.x;` this statement is inside the else branch of `if(max_edge_test_ok)`, so the inline if is redundant
- `FilterDepth = cond0 ? 0.0f : v.x;` could be replaced by a else branch for `if(!cond0)`, where `FilteredDepth` is set to zero

boulabiar, 2015-03-24:
I have the same noise issue and after replacing the line 165 as you mentioned, the noise disappeared.

julienaubert, 2015-03-24:
Thanks! Good catch. Now the output is more equal.

wiedemeyer, 2015-03-25:
added a pull-request #189 to fix it. So this issue might be closed.

floe, 2015-03-25:
Merged #189 - but as you mentioned some other issues, I'll leave this open for now.

floe, 2015-09-28:
Issues mentioned in https://github.com/OpenKinect/libfreenect2/issues/183#issuecomment-85427941 are somewhat related to #404 & #405, needs another look before 0.1.

floe, 2015-11-06:
I'll close this now, I think it has been addressed by some of the refactoring steps taken by @alberth.

closed 2015-11-06 by floe