OpenKinect/libfreenect2 #184: rgb window, pink:ish output

julienaubert, 2015-03-23:
This is regardless if I use CPU, OpenCL or OpenGL. On a macbook (Intel HD 4000) OS X 10.10.2.

[image]

julienaubert, 2015-03-24:
The issue seems to be with TurboJPEG. 

In TurboJpegRgbPacketProcessor::TurboJpegRgbPacketProcessor() I load this JPEG into a buffer

[image]

In void TurboJpegRgbPacketProcessor::process(const RgbPacket &packet), I use that buffer instead of packet.jpeg_buffer. 

    int r = tjDecompress2(impl_->decompressor, _compressedImage, _jpegSize, impl_->frame->data, 1920, 0, 1080, TJPF_BGR, 0);

I then, to rule out it is any other issue, also modifiy the written frame data on y < 500:

        for (int y = 500; y < 1080; ++y) {
            for (int x = 0; x < 1920; x++) {
                impl_->frame->data[0+x*3+y*1920*3] = x < 600 ? x*y : 0;
                impl_->frame->data[1+x*3+y*1920*3] = x > 600 && x < 1200 ? x*y : 0;
                impl_->frame->data[2+x*3+y*1920*3] = x > 1200 ? x*y : 0;
            }
        }

And the result is:

[image]

I had issues when I ran ./install_mac.sh, it would try to downoad turbojpeg but got redirected and so it downloaded a html file, I therefore manually downloaded it to depends and after that it worked out fine. I downloaded the same version as install_mac tried to download: 1.3.1. (I have also checked md5 sum).

probably next step is to make a minimal example and file a ticket for turbojpeg?

julienaubert, 2015-03-24:
same result with libjpeg-turbo-1.4.0

pjryan93, 2015-03-25:
I'm experiencing the same thing on my macbook 10.10.2 mid 2011 Intel  HD3000.

julienaubert, 2015-03-25:
this fixes it for me: (having first run install_mac.sh)
brew install jpeg-turbo
rm -r depends/libjpeg_turbo 
cp -r /usr/local/opt/jpeg-turbo depends/libjpeg_turbo 

Could we remove libjpeg-turbo from the install_mac.sh completely and instead rely on user installing it via brew?

(brew installs it as keg only, so would need to add: 
   LDFLAGS:  -L/usr/local/opt/jpeg-turbo/lib
   CPPFLAGS: -I/usr/local/opt/jpeg-turbo/include)

floe, 2015-05-22:
Does this problem still occur?

PyrApple, 2015-05-26:
++ julienaubert, this solved the issue for me (OSX 10.9)

xlz, 2015-06-07:
homebrew jpeg-turbo 1.4.0 build from source also reproduces this error. `./install_mac.sh` with clang and gcc-5 both reproduce this error.

homebrew pre-built binary outputs correct results. I suspect nasm problems.

closed 2015-06-07 by floe

nkeynes, 2015-09-22:
For what it's worth, I had this problem (OSX 10.10), and was able to fix it by building with yasm (1.30) instead of nasm. Would seem to confirm suspicions that nasm is somehow at fault here.

PR #265 (2015-06-07): Quick documentation fix
This fixes #184 with some documentation changes.