OpenKinect/libfreenect2 #185: Segfault 11 (on mac)

julienaubert, 2015-03-23:
Regardless if I use OpenCL, CPU, or OpenGL.

[OpenCLDepthPacketProcessor] avg. time: 19.8993ms -> ~50.2531Hz
q[Freenect2DeviceImpl] stopping...
[Freenect2DeviceImpl] disabling usb transfer submission...
[Freenect2DeviceImpl] canceling usb transfers...
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
libusb: error [darwin_cancel_transfer] unknown endpoint type 49
libusb: error [libusb_cancel_transfer] cancel transfer failed error -2
Segmentation fault: 11

floe, 2015-04-01:
Does this happen when you quit, or randomly at some point during runtime?

julienaubert, 2015-04-01:
Each time when I quit.

floe, 2015-04-02:
OK, in this case, can you try to run Protonect from gdb and post a stacktrace from the segfault?

julienaubert, 2015-04-05:
Actually, it only occurs if I ctrl-c to exit. Running with gdb (using signal cmd in gdb to pass it to program):

Program received signal EXC_BAD_ACCESS, Could not access memory.
Reason: KERN_INVALID_ADDRESS at address: 0x0000000000000000
libfreenect2::usb::TransferPool::cancel (this=0x10130e268) at /Users/julien/Documents/projects/libfreenect2/examples/protonect/src/transfer_pool.cpp:114
114	    int r = libusb_cancel_transfer(*it);

xlz, 2015-06-08:
Fixed by #275.

closed 2015-06-09 by floe