OpenKinect/libfreenect2 #186: USB not receiving frames from the Kinect2 / No depth getting published

bg-rad, 2015-03-24:
Hello,

basically I do not seem to be able to get depth pubishing running through either OpenCL or CPU. There exists a likelyhood that the issue is related to the USB / the USB controller.

Here are some of the USB outputs,

lsusb finds the three seperate MS components using the retail connector (only one when using the prototype connector, does not work with either):

Bus 004 Device 003: ID 045e:02c4 Microsoft Corp. 
Bus 004 Device 002: ID 045e:02d9 Microsoft Corp. 

Bus 003 Device 003: ID 045e:02d9 Microsoft Corp. 

The lsusb -t command displays the following (these outputs relating to the kinect2 device I believe)

/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 3: Dev 2, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=usbfs, 5000M
        |__ Port 1: Dev 3, If 1, Class=Vendor Specific Class, Driver=usbfs, 5000M
        |__ Port 1: Dev 3, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 3, If 3, Class=Audio, Driver=snd-usb-audio, 5000M

/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 3, If 0, Class=Hub, Driver=hub/1p, 480M

I can find the following information relating to the device using dmesg, it seems that the part of the device on port 3 is running on high-speed instead on SuperSpeed, could that be a potential issue? Although I do not know why it finds two seperate adaptors.

[    2.891656] usb 3-1.2: new high-speed USB device number 3 using xhci_hcd
[    2.981887] usb 3-1.2: New USB device found, idVendor=045e, idProduct=02d9
[    2.981894] usb 3-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.981898] usb 3-1.2: Product: NuiSensor Adaptor      
[    2.981901] usb 3-1.2: Manufacturer: Microsoft Corporation  
[    2.982745] hub 3-1.2:1.0: USB hub found
[    2.983002] hub 3-1.2:1.0: 1 port detected

[    2.108049] usb 4-2.1: new SuperSpeed USB device number 3 using xhci_hcd
[    2.123837] usb 4-2.1: New USB device found, idVendor=045e, idProduct=02c4
[    2.123840] usb 4-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[    2.123841] usb 4-2.1: Product: Xbox NUI Sensor
[    2.123843] usb 4-2.1: Manufacturer: Microsoft
[    2.123844] usb 4-2.1: SerialNumber: 093508334347

[    1.170596] usb 4-2: new SuperSpeed USB device number 2 using xhci_hcd
[    1.188197] usb 4-2: New USB device found, idVendor=045e, idProduct=02d9
[    1.188204] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    1.188208] usb 4-2: Product: NuiSensor Adaptor      
[    1.188212] usb 4-2: Manufacturer: Microsoft Corporation  
[    1.189163] hub 4-2:1.0: USB hub found
[    1.189452] hub 4-2:1.0: 1 port detected

--------------------------------------------------------------------------------------------------------------------------------

Running the ./Protonect gives the following:

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 14 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 093508334347
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    ............C..A
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    .!3U... ........
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    ................
0x0040:  31 33 00 00 00 0b 0a 14 47 4b 47 31 37 32 2e 31    13......GKG172.1
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                X...........

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        .&..

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        .&..

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 093508334347
device firmware: 4.1.3911.0.7
[TurboJpegRgbPacketProcessor] avg. time: 24.3404ms -> ~41.0839Hz
[TurboJpegRgbPacketProcessor] avg. time: 23.8971ms -> ~41.8461Hz
[TurboJpegRgbPacketProcessor] avg. time: 24.0691ms -> ~41.5471Hz
[TurboJpegRgbPacketProcessor] avg. time: 24.2626ms -> ~41.2157Hz
[TurboJpegRgbPacketProcessor] avg. time: 23.8146ms -> ~41.991Hz
[TurboJpegRgbPacketProcessor] avg. time: 24.226ms -> ~41.278Hz
[TurboJpegRgbPacketProcessor] avg. time: 23.9806ms -> ~41.7005Hz
[TurboJpegRgbPacketProcessor] avg. time: 23.9706ms -> ~41.7177Hz
[TurboJpegRgbPacketProcessor] avg. time: 23.2376ms -> ~43.0337Hz
[TurboJpegRgbPacketProcessor] avg. time: 24.1555ms -> ~41.3984Hz
[TurboJpegRgbPacketProcessor] avg. time: 24.1524ms -> ~41.4037Hz

floe, 2015-03-24:
What type of USB controller and what kernel version are you using?

bg-rad, 2015-03-25:
I'm using the 3.16.0-33-generic kernel, and the PCIe usb 3.0 card from STLab, or to be more specific the 24:00.0 USB controller: VIA Technologies, Inc. Device 3483 (rev 01).
 

Here are the full specs:

24:00.0 USB controller: VIA Technologies, Inc. Device 3483 (rev 01) (prog-if 30 [XHCI])
        Subsystem: VIA Technologies, Inc. Device 3483
        Physical Slot: 1-2
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 45
        Region 0: Memory at f3200000 (64-bit, non-prefetchable) [size=4K]
        Capabilities: [80] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0+,D1+,D2+,D3hot+,D3cold+)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [90] MSI: Enable+ Count=1/4 Maskable- 64bit+
                Address: 00000000fee0300c  Data: 41d1
        Capabilities: [c4] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 256 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset-
                DevCtl: Report errors: Correctable- Non-Fatal- Fatal+ Unsupported-
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr+ TransPend-
                LnkCap: Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <16us
                        ClockPM+ Surprise- LLActRep- BwNot-
                LnkCtl: ASPM Disabled; RCB 64 bytes Disabled- CommClk+
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Range B, TimeoutDis+, LTR-, OBFF Not Supported
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
                LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis+
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
                         EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
        Capabilities: [100 v1] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
                AERCap: First Error Pointer: 00, GenCap- CGenEn- ChkCap- ChkEn-
        Kernel driver in use: xhci_hcd

floe, 2015-03-25:
Hm, most people so far had success with controllers from Intel or NEC. No idea about VIA... could you add a different USB3 controller card for testing?

bg-rad, 2015-03-30:
I managed to get it to publish depth on a different computer which uses intel USB 3.0 controllers, however that also came with a heap of issues (for example the opencl reg method does not seem to work [yet] despite all the same libraries being installed on both and the imageviewer segfaulting instantly but those are different issues). 
So at the moment I would assume that indeed it seems that it's a hardware issue with the VIA controller.

anshulvj, 2015-05-20:
Hi floe,

I am facing a similar issue (I'm running Ubuntu 15.04 32-bit, 3.19.0-15-generic). I cout'ed the depth->data output and I'm getting all 0's. Compiling and running the protonect example was no issue at all. My lsusb output is as follows:

Bus 004 Device 002: ID 8087:8000 Intel Corp. 
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 8087:8008 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 007: ID 045e:02c4 Microsoft Corp. 
Bus 002 Device 006: ID 045e:02d9 Microsoft Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 413c:2111 Dell Computer Corp. 
Bus 001 Device 004: ID 046d:c077 Logitech, Inc. 
Bus 001 Device 007: ID 0cf3:0036 Atheros Communications, Inc. 
Bus 001 Device 009: ID 045e:02d9 Microsoft Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lshw output for usb is as follows:
*-usb:0
             description: USB controller
             product: 8 Series/C220 Series Chipset Family USB xHCI
             vendor: Intel Corporation
             physical id: 14
             bus info: pci@0000:00:14.0
             version: 05
             width: 64 bits
             clock: 33MHz
             capabilities: xhci bus_master cap_list
             configuration: driver=xhci_hcd latency=0
             resources: irq:27 memory:f7e00000-f7e0ffff
 
        *-usb:1
             description: USB controller
             product: 8 Series/C220 Series Chipset Family USB EHCI #2
             vendor: Intel Corporation
             physical id: 1a
             bus info: pci@0000:00:1a.0
             version: 05
             width: 32 bits
             clock: 33MHz
             capabilities: ehci bus_master cap_list
             configuration: driver=ehci-pci latency=0
             resources: irq:16 memory:f7e1c000-f7e1c3ff
*-usb:2
             description: USB controller
             product: 8 Series/C220 Series Chipset Family USB EHCI #1
             vendor: Intel Corporation
             physical id: 1d
             bus info: pci@0000:00:1d.0
             version: 05
             width: 32 bits
             clock: 33MHz
             capabilities: ehci bus_master cap_list
             configuration: driver=ehci-pci latency=0
             resources: irq:23 memory:f7e1b000-f7e1b3ff

Any help or pointers will be appreciated. Thanks in advance

closed 2015-05-21 by floe