OpenKinect/libfreenect2 #188: problem in building project with visualstudio 2012

dinprojector, 2015-03-25:
I'm trying to build the project on windows 8 with visual studio 2012.
I've followed all the istructions:
	1. Downloaded libusbK  driver with zadig
	2. I've copied libusbx folder in depends folder
	3. I've tried to build the project but it is not possible because all references to opencv files and libfreenect2 are missing
(additional information: microsoft sdk are not installed)	
In the visual studio i've included opencv 2.9 downloaded separately but i'm not able to find out lib files of libfreenect2.

Second solution:
i've  tried to build the software with cmake, with CMakeList in the folder \libfreenect2\examples\protonect\ but it gives me the following error:

CMake Error at CMakeLists.txt:45 (FIND_PACKAGE):
  By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenCV", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

  
  
  What I want is to try libfreenect2 in the fastest way, with VS 2012 on W8.

Edwinzero, 2015-04-23:
I have met the same issue with you, @dinprojector 
If you find OpenCV_DIR_NOT_FOUND,
Please just configure the path in CMAKE GUI to "opencv/build/" (This is where you build your build binaries of opencv). You will find OpenCVConfig.cmake and opencv-config.cmake there.
I have already created a tutorial for building libfreenect2 on windows8  #105 
I think we may have one findOpenCV.cmake file that will make it easier for building on Windows @floe.

AliShug, 2015-07-20:
I had a similar problem when I tried to use OpenCV 3.0; switching to 2.4 seemed to fix it.

xlz, 2015-08-21:
OpenCV has been removed. It should be no longer an issue.

closed 2015-09-02 by floe