OpenKinect/libfreenect2 #193: device not found

panjia1983, 2015-03-30:
Hi,
I ran into some problem while using the Kinect V2. The installation of the src code was smooth, but when I tried ```./bin/Protonect``` there is error:
```
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 15 usb devices connected
[Freenect2Impl] failed to open Kinect v2 @4:17!
[Freenect2Impl] found 0 devices
[Freenect2Impl] requested device 0 is not connected!
no device connected or failure opening the default one!
```

I checked using lsusb, and seems this should be the Kinect device:
```
Bus 004 Device 017: ID 045e:02d8 Microsoft Corp. 
Bus 004 Device 016: ID 045e:02d9 Microsoft Corp. 
```

floe, 2015-03-31:
You probably don't have access rights to the raw USB device, try running Protonect as root.

matthappens, 2015-04-12:
I'm having a similar issue, though my Protonect output doesn't have the "failed to open Kinect..."
```
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 10 usb devices connected
[Freenect2Impl] found 0 devices
[Freenect2Impl] requested device 0 is not connected!
no device connected or failure opening the default one!
```

My lsusb output is 
```
Bus 020 Device 024: ID 045e:02d9 Microsoft Corporation NuiSensor Adaptor 
Bus 020 Device 031: ID 045e:02d8 Microsoft Corporation Xbox NUI Sensor  Serial: 511707742542
```

I tried running Protonect with sudo, without success.  Any further suggestions?

ipekbocegi, 2015-04-16:
Did you give permission to access Kinect? Please check "/etc/udev/rules.d/" there should be a file named "90-kinect2.rules" which should include 
# ATTR{product}=="Kinect2"
SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02c4", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02d8", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02d9", MODE="0666"

matthappens, 2015-04-20:
Amazing! that did the trick.

I had to manually create the above file however.  How would it normally come to exist?

ipekbocegi, 2015-04-20:
I think only way is to create it manually as stated here:
https://github.com/code-iai/iai_kinect2
Check the Readme file > Permission to access Kinect One.
I also created it manually

floe, 2015-04-20:
OK, so can we close this? Perhaps a note in the README would be a good idea, though.

floe, 2015-09-28:
This is documented in the README and the rules file is actually available in the repo by now, closing.

closed 2015-09-28 by floe

OmegaNalphA, 2017-01-31:
I have the same problem on Windows 10, can't seem to find the file to add. Can anybody recommend a solution?
`
64 windows
Init Kinect2 
[Freenect2Impl] enumerating devices...
No Device Connected!
[Freenect2Impl] 8 usb devices connected
[Freenect2Impl] failed to open Kinect v2 @1:18!
[Freenect2Impl] failed to open Kinect v2 @1:18!
[Freenect2Impl] found 0 devices
Cannot Find Devices
`

wiseshrimp, 2018-01-30:
`64 windows
Init Kinect2 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 8 usb devices connected
[Freenect2Impl] failed to open Kinect v2 @2:10!
[Freenect2Impl] found valid Kinect v2 @2:10 with serial 008114765147
[Freenect2Impl] found 1 devices
1 Device Connected!
[OpenCLDepthPacketProcessor::listDevice]  devices:
  0: GeForce GTX 1070 (GPU)[NVIDIA Corporation]
  1: Intel(R) HD Graphics 530 (GPU)[Intel(R) Corporation]
  2: Intel(R) Core(TM) i7-6700HQ CPU @ 2.60GHz (CPU)[Intel(R) Corporation]
[OpenCLDepthPacketProcessor::init]  selected device: GeForce GTX 1070 (GPU)[NVIDIA Corporation]
Devce: 0
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 8 usb devices connected
[Freenect2Impl] failed to open Kinect v2 @2:10!
[Freenect2Impl] failed to open Kinect v2 @2:10!
[Freenect2Impl] found 0 devices
[Freenect2Impl] requested device 0 is not connected!
no device connected or failure opening the default one!
OpenCV for Processing 0.5.4 by Greg Borenstein http://gregborenstein.com
Using Java OpenCV 2.4.5.0
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
java.lang.NullPointerException
java.lang.NullPointerException
java.lang.NullPointerException
java.lang.NullPointerException
java.lang.NullPointerException
java.lang.NullPointerException
java.lang.NullPointerException
EXIT
closing Kinect v2 :
`

Also getting the error when trying to run with Windows 10. Any suggestions?