OpenKinect/libfreenect2 #195: Unresolved External Symbols When Building

zweicoder, 2015-03-30:
New to libfreenect2/C++, am trying to build libfreenect2 following instructions [here](https://github.com/OpenKinect/libfreenect2/issues/105).

Currently stuck as I'm getting these errors when building freenect2. Is there anything I'm missing?
```
Error	6	error LNK2019: unresolved external symbol _tjDecompress2 referenced in function "protected: virtual void __thiscall libfreenect2::TurboJpegRgbPacketProcessor::process(struct libfreenect2::RgbPacket const &)" (?process@TurboJpegRgbPacketProcessor@libfreenect2@@MAEXABURgbPacket@2@@Z)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\turbo_jpeg_rgb_packet_processor.obj	freenect2
Error	25	error LNK2019: unresolved external symbol _libusb_unref_device@4 referenced in function "public: void __thiscall libfreenect2::Freenect2Impl::clearDeviceEnumeration(void)" (?clearDeviceEnumeration@Freenect2Impl@libfreenect2@@QAEXXZ)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\libfreenect2.obj	freenect2
Error	2	error LNK2019: unresolved external symbol _libusb_submit_transfer@4 referenced in function "public: void __thiscall libfreenect2::usb::TransferPool::submit(unsigned int)" (?submit@TransferPool@usb@libfreenect2@@QAEXI@Z)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\transfer_pool.obj	freenect2
Error	18	error LNK2019: unresolved external symbol _libusb_set_interface_alt_setting@12 referenced in function "public: enum libfreenect2::protocol::UsbControl::ResultCode __thiscall libfreenect2::protocol::UsbControl::setIrInterfaceState(enum libfreenect2::protocol::UsbControl::State)" (?setIrInterfaceState@UsbControl@protocol@libfreenect2@@QAE?AW4ResultCode@123@W4State@123@@Z)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\usb_control.obj	freenect2
Error	15	error LNK2019: unresolved external symbol _libusb_set_configuration@8 referenced in function "public: enum libfreenect2::protocol::UsbControl::ResultCode __thiscall libfreenect2::protocol::UsbControl::setConfiguration(void)" (?setConfiguration@UsbControl@protocol@libfreenect2@@QAE?AW4ResultCode@123@XZ)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\usb_control.obj	freenect2
Error	31	error LNK2019: unresolved external symbol _libusb_reset_device@4 referenced in function "protected: class libfreenect2::Freenect2Device * __thiscall libfreenect2::Freenect2::openDevice(int,class libfreenect2::PacketPipeline const *,bool)" (?openDevice@Freenect2@libfreenect2@@IAEPAVFreenect2Device@2@HPBVPacketPipeline@2@_N@Z)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\libfreenect2.obj	freenect2
Error	17	error LNK2019: unresolved external symbol _libusb_release_interface@8 referenced in function "public: enum libfreenect2::protocol::UsbControl::ResultCode __thiscall libfreenect2::protocol::UsbControl::releaseInterfaces(void)" (?releaseInterfaces@UsbControl@protocol@libfreenect2@@QAE?AW4ResultCode@123@XZ)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\usb_control.obj	freenect2
Error	29	error LNK2019: unresolved external symbol _libusb_open@8 referenced in function "protected: class libfreenect2::Freenect2Device * __thiscall libfreenect2::Freenect2::openDevice(int,class libfreenect2::PacketPipeline const *,bool)" (?openDevice@Freenect2@libfreenect2@@IAEPAVFreenect2Device@2@HPBVPacketPipeline@2@_N@Z)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\libfreenect2.obj	freenect2
Error	21	error LNK2019: unresolved external symbol _libusb_init@4 referenced in function "public: __thiscall libfreenect2::Freenect2Impl::Freenect2Impl(void *)" (??0Freenect2Impl@libfreenect2@@QAE@PAX@Z)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\libfreenect2.obj	freenect2
Error	5	error LNK2019: unresolved external symbol _libusb_handle_events_timeout_completed@12 referenced in function "private: void __thiscall libfreenect2::usb::EventLoop::execute(void)" (?execute@EventLoop@usb@libfreenect2@@AAEXXZ)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\event_loop.obj	freenect2
Error	32	error LNK2019: unresolved external symbol _libusb_get_string_descriptor_ascii@16 referenced in function "public: void __thiscall libfreenect2::Freenect2Impl::enumerateDevices(void)" (?enumerateDevices@Freenect2Impl@libfreenect2@@QAEXXZ)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\libfreenect2.obj	freenect2
Error	12	error LNK2019: unresolved external symbol _libusb_get_ss_endpoint_companion_descriptor@12 referenced in function "int __cdecl libfreenect2::protocol::libusb_ext::get_max_iso_packet_size(struct libusb_device *,int,int,int)" (?get_max_iso_packet_size@libusb_ext@protocol@libfreenect2@@YAHPAUlibusb_device@@HHH@Z)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\usb_control.obj	freenect2
Error	23	error LNK2019: unresolved external symbol _libusb_get_device_list@8 referenced in function "public: void __thiscall libfreenect2::Freenect2Impl::enumerateDevices(void)" (?enumerateDevices@Freenect2Impl@libfreenect2@@QAEXXZ)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\libfreenect2.obj	freenect2
Error	26	error LNK2019: unresolved external symbol _libusb_get_device_descriptor@8 referenced in function "public: void __thiscall libfreenect2::Freenect2Impl::enumerateDevices(void)" (?enumerateDevices@Freenect2Impl@libfreenect2@@QAEXXZ)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\libfreenect2.obj	freenect2
Error	28	error LNK2019: unresolved external symbol _libusb_get_device_address@4 referenced in function "public: bool __thiscall libfreenect2::Freenect2DeviceImpl::isSameUsbDevice(struct libusb_device *)" (?isSameUsbDevice@Freenect2DeviceImpl@libfreenect2@@QAE_NPAUlibusb_device@@@Z)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\libfreenect2.obj	freenect2
Error	14	error LNK2019: unresolved external symbol _libusb_get_device@4 referenced in function "public: enum libfreenect2::protocol::UsbControl::ResultCode __thiscall libfreenect2::protocol::UsbControl::getIrMaxIsoPacketSize(int &)" (?getIrMaxIsoPacketSize@UsbControl@protocol@libfreenect2@@QAE?AW4ResultCode@123@AAH@Z)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\usb_control.obj	freenect2
Error	9	error LNK2019: unresolved external symbol _libusb_get_configuration@8 referenced in function "public: enum libfreenect2::protocol::UsbControl::ResultCode __thiscall libfreenect2::protocol::UsbControl::setConfiguration(void)" (?setConfiguration@UsbControl@protocol@libfreenect2@@QAE?AW4ResultCode@123@XZ)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\usb_control.obj	freenect2
Error	10	error LNK2019: unresolved external symbol _libusb_get_config_descriptor_by_value@12 referenced in function "int __cdecl libfreenect2::protocol::libusb_ext::get_max_iso_packet_size(struct libusb_device *,int,int,int)" (?get_max_iso_packet_size@libusb_ext@protocol@libfreenect2@@YAHPAUlibusb_device@@HHH@Z)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\usb_control.obj	freenect2
Error	27	error LNK2019: unresolved external symbol _libusb_get_bus_number@4 referenced in function "public: bool __thiscall libfreenect2::Freenect2DeviceImpl::isSameUsbDevice(struct libusb_device *)" (?isSameUsbDevice@Freenect2DeviceImpl@libfreenect2@@QAE_NPAUlibusb_device@@@Z)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\libfreenect2.obj	freenect2
Error	4	error LNK2019: unresolved external symbol _libusb_free_transfer@4 referenced in function "public: void __thiscall libfreenect2::usb::TransferPool::deallocate(void)" (?deallocate@TransferPool@usb@libfreenect2@@QAEXXZ)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\transfer_pool.obj	freenect2
Error	13	error LNK2019: unresolved external symbol _libusb_free_ss_endpoint_companion_descriptor@4 referenced in function "int __cdecl libfreenect2::protocol::libusb_ext::get_max_iso_packet_size(struct libusb_device *,int,int,int)" (?get_max_iso_packet_size@libusb_ext@protocol@libfreenect2@@YAHPAUlibusb_device@@HHH@Z)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\usb_control.obj	freenect2
[... cut ...] ferPool@usb@libfreenect2@@MAEPAUlibusb_transfer@@XZ)	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\transfer_pool.obj	freenect2
Error	8	error LNK2001: unresolved external symbol _libusb_error_name@4	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\build\command_transaction.obj	freenect2
Error	33	error LNK1120: 31 unresolved externals	D:\time2code\kinect stuff\libfreenect2-master\libfreenect2-master\examples\protonect\lib\Debug\freenect2.dll	freenect2
```

BenjaminRothschild, 2015-04-02:
clearly,you did not import libusb completely,you can make it like this:https://github.com/OpenKinect/libfreenect2/issues/105,take notice of the cmake procedure,make sure no warings ,no errors,so that you can build successfully.good luck.

zweicoder, 2015-04-04:
Hi, following #105 I added the necessary lines to the Cmake file and put the downloaded libusbx binaries at the specified directories. Just in case I built libusb from source and added the Win32/lib folder to my Library Directories, but this still happens. Is there an additional step to import libusb completely?

zweicoder, 2015-04-04:
I replaced the downloaded libusb-1.0 with the one built from source and it builds properly now.

closed 2015-04-04 by zweicoder

BenjaminRothschild, 2015-04-05:
yes,you must rebuilt the libusb source code  locally,and import the lib and dll.