OpenKinect/libfreenect2 #196: libusb.h: No such file or directory

navid33, 2015-04-01:
Hi,
I am using Ubuntu 14.04.2, ROS Indigo. I am getting this after    sudo apt-get install

libopencv-dev is already the newest version.
libopencv-dev set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.
 mesa-common-dev-lts-utopic : Conflicts: mesa-common-dev
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

 and  after running make && sudo make install,I am getting this:

In file included from /home/navidd33/libfreenect2/examples/protonect/src/transfer_pool.cpp:27:0:
/home/navidd33/libfreenect2/examples/protonect/include/libfreenect2/usb/transfer_pool.h:31:20: fatal error: libusb.h: No such file or directory
 #include <libusb.h>
                    ^
compilation terminated.
make[2]: *** [CMakeFiles/freenect2.dir/src/transfer_pool.cpp.o] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2

In my libfreenect2/depends folder, libusb does not exist.

any Ideas?
Thanks,

floe, 2015-04-01:
Can you try `sudo apt-get install -f` to fix broken packages first?

navid33, 2015-04-01:
this is the result:
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.

I checked var/log/dist-upgrade. nothing is there too.

Edwinzero, 2015-04-23:
@navid33 , If you check CMakeLists.txt, you will find the ADD_DEPENDENCIES shows that the library is named usb-1.0 instead of libusb-1.0. Just change it to libusb-1.0 then you will get the good result.

navid33, 2015-04-24:
@Edwinzero , Thank you for your help but I have already sort that out and I successfully installed libfreenect2.
However when I run protonect, I have RGB colour image but IR and Depth are black.I am not getting any errors and my outputs is:
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 11 usb devices connected
[Freenect2Impl] found valid Kinect v2 @3:3 with serial 010812241047
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 04 0f 14 47 4b 56 33 34 31 2e 31    31332e2e2e2e2e2e474b563334312e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  23 26 00 00                                        23262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  23 26 00 00                                        23262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 010812241047
device firmware: 4.1.3912.0.7
[TurboJpegRgbPacketProcessor::doProcess] Failed to decompress rgb image! TurboJPEG error: 'Not a JPEG file: starts with 0x68 0x62'
[DepthPacketStreamParser::handleNewData] not all subsequences received 0
[DepthPacketStreamParser::handleNewData] not all subsequences received 512
[DepthPacketStreamParser::handleNewData] not all subsequences received 992
[OpenGLDepthPacketProcessor] avg. time: 5.54912ms -> ~180.209Hz
[OpenGLDepthPacketProcessor] avg. time: 5.72282ms -> ~174.739Hz
[TurboJpegRgbPacketProcessor] avg. time: 20.4993ms -> ~48.7822Hz
[OpenGLDepthPacketProcessor] avg. time: 5.60259ms -> ~178.489Hz
[OpenGLDepthPacketProcessor] avg. time: 4.41606ms -> ~226.446Hz
[TurboJpegRgbPacketProcessor] avg. time: 20.6756ms -> ~48.3662Hz
[OpenGLDepthPacketProcessor] avg. time: 4.32521ms -> ~231.202Hz
[OpenGLDepthPacketProcessor] avg. time: 4.23414ms -> ~236.175Hz
[TurboJpegRgbPacketProcessor] avg. time: 20.6813ms -> ~48.3528Hz
[OpenGLDepthPacketProcessor] avg. time: 4.28717ms -> ~233.254Hz
[OpenGLDepthPacketProcessor] avg. time: 4.28554ms -> ~233.343Hz
[TurboJpegRgbPacketProcessor] avg. time: 20.4074ms -> ~49.0018Hz
[OpenGLDepthPacketProcessor] avg. time: 4.24873ms -> ~235.365Hz
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 1021
[OpenGLDepthPacketProcessor] avg. time: 4.27235ms -> ~234.063Hz

any Ideas?
Thanks

Edwinzero, 2015-04-24:
@navid33  That is the part I confused. In CMakeLists.txt there is said only use turbojpeg on MAC but this print out shows that you have something wrong with turbojpeg. Prehaps you need to install turbojpeg on your computer.
Is this right? @larshg

larshg, 2015-04-24:
Hmm @Edwinzero, since he gets RGB images there is nothing wrong with the turbojpeg install, even though he gets one Jpeg error, which seems the parsing can be wrong at the current implementation. But the black depth/ir images might be some of the reasons talked about in #31 ?
Whats your hardware @navid33 ?

If you are using the OpenGL packetprocesser this is probably the case - then you have to try out the OpenCL implementation or worst case the CPU implementation, but this is really slow.

You can change it during cmake or by passing in specific packetprocessor like in https://github.com/OpenKinect/libfreenect2/issues/31#issuecomment-72007631

navid33, 2015-04-24:
@larshg  My hardware is:
HP ENVY TS 15 Notebook PC
CPU:  Intel(R) Core(TM) i7-4700MQ CPU @ 2.40GHz
Graphics Device 1:Intel(R) HD Graphics 4600
Graphics Device 2:NVIDIA GeForce GT 740M 

I will try OpenCl and let you know.
Thanks guys,
I appreciate your help

navidd33, 2015-04-29:
yes.It's working great.
Thanks a lot

navid33, 2015-04-29:
Yes,it's working Great 

Thanks a lot

closed 2015-04-29 by navid33

Edwinzero, 2015-04-29:
@navid33  By the way, short question. You build libfreenect2 in win32 or x64?

navid33, 2015-05-01:
in x64 ubuntu

rajnikanthsabnekar, 2016-05-12:
im getting this error 
fatal error: libusb.h: No such file or directory
 #include <libusb.h>
                    ^
compilation terminated.
make[3]: *** [CMakeFiles/interface_cc2531.dir/interface_cc2531.c.o] Error 1
make[3]: Leaving directory `/home/rajnikant/foren6/capture/build'
make[2]: *** [CMakeFiles/interface_cc2531.dir/all] Error 2
make[2]: Leaving directory `/home/rajnikant/foren6/capture/build'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/home/rajnikant/foren6/capture/build'
make: *** [build] Error 2
i have tried the above given solutions , im enable to find( If you check CMakeLists.txt, you will find the ADD_DEPENDENCIES shows that the library is named usb-1.0 instead of libusb-1.0. Just change it to libusb-1.0 then you will get the good result.) i didnt find either usb-1.0 or libusb-1.0 
please help me

prasanthsubramaniam, 2016-09-24:
tgg

netspaces, 2019-05-29:
fatal error: libusb.h: File not found
`sudo apt-get install libusb-1.0-0-dev`
[https://github.com/cetic/foren6/issues/10](url)

virtuary, 2019-10-22:
(Applicable to #Ubuntu, #Debian, #Raspbian in my case) 
*1st, I followed and apply the steps above of awesome advice and for the most part it got me to the part that I just figured out below.  So hopefully the advice aboe + mine equals finally 1 place where all the solutions live and not scattered all over the interwebs.
*Then, made sure I started the steps in the correct directories. So don't start at /home if you're supposed to have a specific path per some instructions you already have.
*Test starting at /root, and another test at /yourusername 
*if you are making a dir "/src" try "~/src" or "~/build" (using the Tilde symbol).  Sounds stupid but fixed one of the errors.
Call me crazy, and I'm not a coder, or linux buff, just basic, so I used the help of the GUI.  
In my case, getting this to Build and Install and process would fail at multiple stages like  Errors above, and at some tar.bz2 extract stages.  This is what I did via Terminal and then GUI as I encountered errors:

1 - a few errors at first were simple syntax errors (facepalm complete), such as spelling "libusb" as "lisusb".  Again, :facepalm: initiated multiple times, but CHECK THIS FIRST.
2 - these are relevant to the above comments, MOST of the issues were DIR issues.
(ie The ones related to the "make" process, a lot of times were because the DIR those files  -(ie configure, makefile, the 'tar.bz2 or tar.gz' files, etc.,)  - were in the wrong directories.
*To fix this, I had to first figure out where they extracted, why they extracted in wrong directories in the first places, then get them into the correct DIRs (/src,  /var, /build,  /libusb /etc) I did this part **via the GUI.** as I mentioned.
It was weird cause I was running all the WGET and TAR steps inside the directories some of these online instructions were telling me, like within 'SRC' or 'BUILD' and so on.

3 - There was a point in the Steps, where you have to 'cd' into the DIR 'libusb vxX.X.X' and perform commands within it like the MAKE and CONFIGURE process, HOWEVER, you get the errors either:
'No such Directory' or another one (can't and don't want to remember) with 4 or 5 lines related to Error: Child Directory Something Or Rather and with some numbers, along with the error above in those steps.
*To fix that one, basically it was that the 'libusb' file was just that, a FILE.  How this happened or why? IDK.  The point is I fixed it by old-fashion common sense, 'MKDIR'.  Now that I made a directory, of course it was empty.  ##**oh, make sure the syntax and version of 'libusb' is the version you are installing+downloaded (or 'wget'ing) when you create the directory manually**##
*So basically create the folder you are getting stuck on and errors, then place all the files that are supposed to be in there- I did this using BOTH terminal and GUI.
** Files like "libusb-vX.X" & "libfreekenect" or the distro files.  Jus dump them in this new directory you manually made.
**Another error was extracting, "No Such File or DIR" -- just use the GUI and double check everything.

4- Missing files or DIRs - Copying and Pasting files from the extracted folder using the GUI into the directories that were either empty, didn't exist, or already had files in it but were related to where I was getting stuck, is how I fixed a lot of these frustrating errors sometimes.
*Then in each DIR I dropped files in or did some of these things I mentioned, for good measure:
* I would again do perform a WGET 
+
*sudo apt-get- update (and upgrade) always 
*double check everything as you go along 'ls -a' and see if things are getting more logical and files are there that are supposed to be (or again, use the GUI)

#Basically I loaded, re-extracted, and packed the sh*t out of everything; didn't care at this point, just Copy/Pasted Everything starting at the '/src' all the way through '/libusb%/build', even if certain things weren't supposed to be in there, or had no purpose (as long as it would not trigger adverse effects.)#
The majority of the files I was duplicating and loading every directory with were GENERIC files anyway that came with the distros and releases I was trying to get installed.  So I didn't care, just use common sense (ie You cannot put a "build" directory within itself)
- If you are copy/pasting using the GUI, Select All the files from the parent DIRs and starts pasting them in the next directory, and so on.
#I know this might be messy and not the best method, but hey guess what, this is the only thing that worked out off all the advice and sites I found that did nothing ( or a little good but not the ultimate goal).#

Finally started seeing the files that were supposed to be these and solved the 'no such file' and 'makelist' or no make file errors.

One last thing, since I was Copy/Pasting contents of the DIRS that did have a MAKEFILE and CONFIGURE files, into the DIRs that did not or were acting wonky,  launching the config file (GUI) or command (terminal) needs to be done carefully.
For example, make sure if you are taking,say,  the CONFIGURE file from a parent directory, that it is still related to the distro or process versions you are trying to install.
GOOD - taking the CONFIGURE file from DIRs in 'libusb' and pasting to sub DIRs of linusb.
BAD - taking a CONFIGURE file from DIRs in 'libusb' and pasting them into DIRS of 'freekenct'.

[... cut ...]
- and when you get passed your errors and are at steps where you are actually trying to start communicating via usb or testing the cam/device

If somebody can consolidate this into one gold nugget of an article or solution, please do so. 
Sorry for long post.  I'm thorough, and also come across of posts that people offer solutions but are vague or say things like "try this" out of context and without saying at exactly what steps of before and after what command.  Lol.  
Hope this helps at least 1 person.

carlos-garcia, 2020-04-13:
for those who stumbled upon here, found that latest ubuntu installs libusb 2, i followed the install of netspaces to unstal 1.0 version but still got issues..

No rule to make target '/usr/lib/x86_64-linux-gnu/libusb.so', needed by 'lib/libfreenect.so.0.6.1'.  Stop.