OpenKinect/libfreenect2 #197: Request for DLL and LIB file

dinprojector, 2015-04-01:
I'm trying to build libfreenect2 following istruction <a href="https://github.com/OpenKinect/libfreenect2/issues/105"> here</a>.
After that i faild to build with the proceudre described at beginning. 
I need to test this driver fory mutlikinect support fast as possible.
Has anyone the possibility to uplodad binaries of libfreenect2?

I need to make it work on windows 8, with VS 2012.

I will really appreciate it.
Thank you.

floe, 2015-04-02:
I don't think that will make much sense, as libfreenect2 is still very much a work in progress; DLL/LIB files only will be useful once we've reached an initial feature freeze & release. If you post some more detailed error descriptions, maybe somebody will be able to help you here.

dinprojector, 2015-04-03:
You are right, i will describe my building procedure.

DemonEater, 2015-04-04:
Will detailed instructions on installation be provided when you've reached freeze and release? I was following the same instructions as dinprojector, but had very little success.

Edwinzero, 2015-04-23:
Hey guys,
 
He is using the instruction I write...
I try to build libfreenect2 in Win32 and I get the result like below
[image]
[image]

A

Edwinzero, 2015-04-23:
Hello @larshg ,

Sorry for bother you again, could you give me some suggestion about my problem?

larshg, 2015-04-24:
Hi @Edwinzero 

Yes, you also need the libjpeg-turbo on windows for handling the RGB data.

Since you get errors with the USB - you are using USB 3.0 ? Can you try another USB controller?

Else it seems fine.

Edwinzero, 2015-04-24:
Hi @larshg ,

Yes I am using USB 3.0. So Kinect v2 is not support USB 3.0? I think I am using the SATA usb 3.0 interface... I will change and try~

Thank you very much man!

larshg, 2015-04-24:
@Edwinzero the kinect v2 requires USB 3.0 - it was just to verify :) I don't know how the SATA usb 3.0 and the belonging chipset works - so if you can try another USB port with a proper chipset, this could probably make it work.

Perhaps updating the drivers for the USB/SATA controller could also solve it.

Edwinzero, 2015-04-24:
Hi @larshg ,

I change it back to 64bit since I create a sensor network interface in 64bit. But I compile all the libraries without error but when I run Protonect.ext there is an error.
[image]

Do you have any ideas?

larshg, 2015-04-24:
Hmm as I remember this is normally due to mixing 32 and 64 bit libs, or using unsupported instruction set.
try to use dependency walker and see if it unintentionallyder links to a wrong library.

Edwinzero, 2015-04-24:
Hi @larshg, I think I am using all the dependencies in x64 bit. 
Here is my dependencies.
[image]
And the libusb I am using is from libusbx-winso.zip 
And the turbo-jpeg I am using is 64 bit, too... 
[image]

Is there any possibility? I didn't use CUDA on my computer and I am using AMD opencl for now.

Edwinzero, 2015-04-24:
I figure it out, I set x86 of opencv in my system variable.... Thanks a lot!

elliotwoods, 2015-05-26:
Agree there should be binary releases (even with warnings/expectations of it not working)
especially 'bury' the binary releases within the readme

i've linked to my build from here:
https://github.com/OpenKinect/libfreenect2/issues/244

closed 2015-06-07 by floe