OpenKinect/libfreenect2 #198: USB 2 Bandwidth Allocation

mitchellwills, 2015-04-01:
I have a situation where i'm trying to use a USB 2.0 webcam and a Kinectv2 on the same computer. When I start up only the libfreenect2 driver or the webcam separately they both work fine. However, if I try to run both of them at the same time I get the following error when trying to open the camera
```
VIDIOC_STREAMON error 28, No space left on device
```
I normally get this error when I am trying to use two usb 2.0 webcams and the USB 2.0 bus cannot handle the bandwidth, but I am only using one camera.

Does the Kinectv2 use USB2.0 in addition to USB3.0 or is libfreenect2 incorrectly allocating bandwidth on the usb2.0 bus?

Linux walrus 3.13.0-48-generic #80-Ubuntu SMP Thu Mar 12 11:16:15 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

floe, 2015-04-02:
Two generic suggestions for most USB problems include:
* try a more recent kernel, e.g. 3.16 from package linux-image-generic-lts-utopic
* if possible, try a different USB3 controller, I think this kind of bandwidth allocation issue often happens with Intel controllers

mitchellwills, 2015-04-05:
Upgrading to the 3.16 kernel (clean install of 14.04.2) appears to have solved this problem

closed 2015-04-05 by mitchellwills