OpenKinect/libfreenect2 #199: access violation exception at : glfwGetWindowAttrib() ;

dinprojector, 2015-04-03:
I've been succed in building my project with Visual studio 12 on Windows 8.1. 
But i get the following error when i run Protonect.exe example:
    Access violation exception:
         window.c -> line 560: return window->glMajor; at glfwGetWindowAttrib(GLFWwindow* handle, int attrib);

Following i described my building procedure; 
Where i'm wrong?

_______________________________________________________________________________
<b>Requirements:</b>
Git Shell (https://msysgit.github.io/)
CMake (from http://www.cmake.org/download/ the windows version x86)

<b>From Git Shell:</b>
go to the folder where you want to export your library:

git clone "https://github.com/OpenKinect/libfreenect2.git"
cd libfreenect2/depends/
sh install_deps.sh

into depends folder now you will find the directories
glfw_src and libusb_src

<b>Other dependences</b>
download opencv (http://sourceforge.net/projects/opencvlibrary/files/opencv-win/2.4.10/opencv-2.4.10.exe/download)
run the installer;
(in my case the opencv directory is C:\opencv)

download turbojpeg (http://sourceforge.net/projects/libjpeg-turbo/files/1.4.0/libjpeg-turbo-1.4.0-vc.exe/download)
run the installer (in my case the turbojpeg directory is C:\libjpeg-turbo);

create a new folder named as libjpeg and copy inside it all the content of:
.\libjpeg-turbo\bin\
.\libjpeg-turbo\lib\
.\libjpeg-turbo\include\

In my case the position of libjpeg folder is :
C:\libjpeg

<b>Generate with cmake</b>

Download from http://code.opencv.org/attachments/196/FindOpenCV.cmake, FindOpenCV.cmake and copy it into the:
path\libfreenect2\examples\proctonet\cmake_modules

Open that file and change as following (to edit this file i suggest to use Notepad++:
	download at: http://notepad-plus-plusorg/download/v6.7.5.html
	open FindOpenCV.cmake with notepad++
	go to setting->preferences->editing->display line number 
):
	go to line 49, before  find_path(OpenCV_DIR "OpenCVConfig.cmake" DOC "Root directory of OpenCV")
;
	write SET ( OpenCV_DIR "opencv folder path/build/" ) in my case opencv folder path is C:/opencv;
		(pair attenction to the slash, the right one should be / and not \. And remember to close the slash (C:/opencv/build/ and not C:/opencv/build);
	save and close;

(for path i mean folder where you have copied your libfreenect2 files);
go to path/libfreenect2
and add a new empty directory named as build	
run cmake;
select browse source:
	path/libfreenect2/examples/protonect
select build directory:
	path/libfreenect2/build

click on configure
click on generate (ignore warnings)

go to the path/libfreenect2/depends/libusb_src/msvc/
run libusb_2012.sln 
and compile libusb-1.0(dll) the project;
go to the folder: 
	path\libfreenect2\depends\libusb_src\Win32\Debug\dll
	copy the content to the new folder defined 
	path\libfreenect2\depends\ as libusb
go to the folder:
	path\libfreenect2\depends\libusb_src\libusb
	and copy libusb.h to the folder path\libfreenect2\depends\libusb
at the end you must have the following files to the path\libfreenect2\depends\libusb\ folder:
	libusb-1.0 (folder)
	libusb.h 
	libusb-1.0.dll
	libusb-1.0.exp
	libusb-1.0.ilk
	libusb-1.0.lib
	libusb-1.0.pdb

<b> building</b>

run:
	path\libfreenect2\build\libreenect2.sln 
	from projec properties go to:
		Linker->Input->Additional dependencies
			substitute usb-1.0.lib with libusb-1.0.lib
		add add opencv lib files:
			opencv_calib3d249d.lib;
			opencv_contrib249d.lib;
			opencv_core249d.lib;
			opencv_features2d249d.lib;
			opencv_flann249d.lib;
			opencv_gpu249d.lib;
			opencv_highgui249d.lib;
			opencv_imgproc249d.lib;
			opencv_legacy249d.lib;
			opencv_ml249d.lib;
			opencv_nonfree249d.lib;
			opencv_objdetect249d.lib;
			opencv_photo249d.lib;
			opencv_stitching249d.lib;
			opencv_superres249d.lib;	
			opencv_ts249d.lib;
			opencv_video249d.lib;
			opencv_videostab249d.lib;
	compiling project freenect2 different times solve the different errors, including lib and dll files:
		for libusb needs the folder is: path\libfreenect2\depends\libusb\
		for opencv needs the folder is: opencvpath\opencv\build\include (for include files);
						opencvpath\opencv\build\x86\vc11\lib
						opencvdll are in \opencvpath\opencv\build\x86\vc11\bin
		for turbojpeg needs the folder is turopjpegpath\libjpeg\

Once that you have compiled with success freenect project, it is time to compile with success Protonec project;
You have to copy to protonect project the same include and lib files of freenect;

Compile of Protonect has been made with success but you have to copy the dll files (it is  almost done):

<b>Running the example:</b>
	copy the following dll:
		all opencv dll contained in opencvpath\opencv\build\x86\vc11\bin;
		turbojpeg.dll in turbojpegpath\libjpeg\
		libusb-1.0.dll path\libfreenect2\depends\libusb
		freenect2.dll and glfw3.dll from path\libfreenect2\examples\protonect\lib\Debug
	 
	path\libfreenect2\examples\protonect\bin\Debug

now set as starting project Protonect and run it:

if you are lucky like me you will get the access violation exception at :
	window.c -> line 560: return window->glMajor; at glfwGetWindowAttrib;

floe, 2015-04-03:
Can you post the full stacktrace? This looks like OpenGL has not been initialized properly.

dinprojector, 2015-04-03:
following sta call stack:

 	glfw3.dll!5f196035()	Unknown
 	[Frames below may be incorrect and/or missing, no symbols loaded for glfw3.dll]	
 	freenect2.dll!5e0e9504()	Unknown
 	freenect2.dll!5e0ef89b()	Unknown
 	freenect2.dll!5e0eb568()	Unknown
 	freenect2.dll!5e0f9f63()	Unknown
 	freenect2.dll!5e0c5bed()	Unknown
 	freenect2.dll!5e0f7c59()	Unknown
 	freenect2.dll!5e0c56e6()	Unknown
 	freenect2.dll!5e0f7bc7()	Unknown
 	freenect2.dll!5e0f8f44()	Unknown
 	freenect2.dll!5e0d77f3()	Unknown
 	freenect2.dll!5e0d7928()	Unknown
	Protonect.exe!main(int argc, char * * argv) Line 58	C++
 	Protonect.exe!__tmainCRTStartup() Line 536	C
 	Protonect.exe!mainCRTStartup() Line 377	C
 	kernel32.dll!74e3919f()	Unknown
 	ntdll.dll!772fa8cb()	Unknown
 	ntdll.dll!772fa8a1()	Unknown

floe, 2015-04-06:
This doesn't help much - I'm no VS expert, but can you use a Debug build instead?

dinprojector, 2015-04-07:
it is already the debug version. I think that there is an error in building procedure. Can you check if it is missing something? 

That building procedure has been made by me, trying to find out the missing libraries.

Edwinzero, 2015-04-23:
You haven't build your dependencies. Run install_deps.sh first.

larshg, 2015-04-24:
It might be a driver issue with wrong opengl reported back - like in #204 - even though this is linux, it seems to make the error the same place.

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe