OpenKinect/libfreenect2 #204: Segmentation fault (core dumped)  Ubuntu 14.04

pammirato, 2015-04-09:
I had this all working, then messed around with some drivers and ROS things and it no longer works.

When I run ./bin/Protonect

I get the following:

  Segmentation fault (core dumped)

running with gdb i get:

Starting program: /playpen/ammirato/libfreenect2/examples/protonect/bin/Protonect 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe301f700 (LWP 14110)]
[New Thread 0x7fffe281e700 (LWP 14111)]

Program received signal SIGSEGV, Segmentation fault.
glfwGetWindowAttrib (handle=handle@entry=0x0, attrib=attrib@entry=139266)
    at /home/ammirato/libfreenect2/depends/glfw_src/src/window.c:560
560	            return window->glMajor;

I am guessing my OpenGL version is not new enough. I get this:

:~$ glxinfo | grep "OpenGL version"
OpenGL version string: 2.1.2 NVIDIA 304.125

Though I believe my graphics card should support OpenGL 4.+
I get: 

$ lspci | grep VGA
02:00.0 VGA compatible controller: NVIDIA Corporation GF119 [NVS 315] (rev a1)

I do not know how to avoid using OpenGL and use OpenCL instead. Is it an option on command line?

pammirato, 2015-04-09:
solved it with:

sudo apt-get install nvidia-opencl-dev

don't ask me why

closed 2015-04-09 by floe

ipekbocegi, 2015-04-13:
Hi,
I got the same error "Segmentation fault (core dumped)"
I have these:
$ glxinfo | grep "OpenGL version"
OpenGL version string: 3.0 Mesa 10.3.2
$ lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Broadwell-U Integrated Graphics (rev 09)

When I
$ cd libfreenect2/examples/protonect/bin
$ gdb -ex r --args ./Protonect

I got the followings below. What should I do now? I have 340,346,349 Nvidia drivers and Xorg in additional drivers. Xorg is chosen. I afraid to chose any Nvidia yet, because I got black screen on startup when I chose any of them before.. Please help..

Kind regards

GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./Protonect...done.
Starting program: /home/ipek/libfreenect2/examples/protonect/bin/Protonect 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe5709700 (LWP 3057)]
[New Thread 0x7fffe4f08700 (LWP 3058)]
[New Thread 0x7fffdb92d700 (LWP 3059)]
[New Thread 0x7fffdb12c700 (LWP 3060)]
[New Thread 0x7fffda92b700 (LWP 3061)]
[New Thread 0x7fffda12a700 (LWP 3062)]

Program received signal SIGSEGV, Segmentation fault.
glfwGetWindowAttrib (handle=handle@entry=0x0, attrib=attrib@entry=139266)
    at /home/milvus-ipek/libfreenect2/depends/glfw_src/src/window.c:560
560	            return window->glMajor;
(gdb) bt

#0  glfwGetWindowAttrib (handle=handle@entry=0x0, attrib=attrib@entry=139266)
    at /home/ipek/libfreenect2/depends/glfw_src/src/window.c:560
#1  0x00007ffff7a0ba54 in flextInit (window=0x0, bindings=0x685410)
    at /home/pek/libfreenect2/examples/protonect/src/flextGL.c:20
#2  0x00007ffff7a0f0b0 in libfreenect2::OpenGLDepthPacketProcessorImpl::initialize (this=this@entry=0x685e50)
    at /home/ipek/libfreenect2/examples/protonect/src/opengl_depth_packet_processor.cpp:472
#3  0x00007ffff7a0d119 in libfreenect2::OpenGLDepthPacketProcessor::OpenGLDepthPacketProcessor (this=0x664740, 
    parent_opengl_context_ptr=<optimized out>, debug=<optimized out>)
    at /home/ipek/libfreenect2/examples/protonect/src/opengl_depth_packet_processor.cpp:786
#4  0x00007ffff79fb526 in libfreenect2::OpenGLPacketPipeline::createDepthPacketProcessor (this=0x672f70)
    at /home/ipek/libfreenect2/examples/protonect/src/packet_pipeline.cpp:109
#5  0x00007ffff79fb645 in libfreenect2::BasePacketPipeline::initialize (this=this@entry=0x672f70)
    at /home/ipek/libfreenect2/examples/protonect/src/packet_pipeline.cpp:43
#6  0x00007ffff79fbab3 in libfreenect2::OpenGLPacketPipeline::OpenGLPacketPipeline (this=0x672f70, 
    parent_opengl_context=<optimized out>, debug=<optimized out>)
    at /home/ipek/libfreenect2/examples/protonect/src/packet_pipeline.cpp:102
#7  0x00007ffff7a05c0f in libfreenect2::createDefaultPacketPipeline ()
    at /home/ipek/libfreenect2/examples/protonect/src/libfreenect2.cpp:631
#8  0x00007ffff7a07eb0 in libfreenect2::Freenect2::openDevice (this=this@entry=0x7fffffffd550, idx=idx@entry=0)
    at /home/ipek/libfreenect2/examples/protonect/src/libfreenect2.cpp:669
#9  0x00007ffff7a07ed7 in libfreenect2::Freenect2::openDefaultDevice (this=this@entry=0x7fffffffd550)
    at /home/ipek/libfreenect2/examples/protonect/src/libfreenect2.cpp:782
#10 0x0000000000401ac1 in main (argc=<optimized out>, argv=<optimized out>)
    at /home/ipek/libfreenect2/examples/protonect/Protonect.cpp:58
(gdb)