OpenKinect/libfreenect2 #206: Restarting Device fails

linknum23, 2015-04-13:
While working on reducing the processor load of the iai_kinect2 interface (The plan is to acquire images at a rate of .25 FPS) I noticed that the kinect device fails if it is restarted. If restarted, a near endless stream of messages like this appear:

*[DepthPacketStreamParser::handleNewData] remaining data should be 8 bytes but is 11231*
*[DepthPacketStreamParser::handleNewData] working buffer full, resetting it!*

The maintainer suggested that I post here as it was problem with this library. To reproduce this bug simply insert a start(),stop() pair in the shut down code (Protonect.cpp,main()) as shown below.

```C
dev->stop();
dev->start();
libfreenect2::this_thread::sleep_for(libfreenect2::chrono::milliseconds(200));
dev->stop();
dev->close();
```
 
How hard would this be to fix? Is there a simpler way to reduce the load on the processor other than stopping and starting?

larshg, 2015-04-14:
Hi @linknum23 

It is properly similar to what I experienced in #82, that the magic1 marker is the last commandindex send to the kinect2. Hence the first time you call start it ends up to be 9, but if you stop and then start it will be 18 and hence the depthpacketparser won't work, as it looks for the magic 9 marker.

What could be done, is to set the command_seq_ number to 0, in the stop method - so when you call start again, it will become 9.

Alternatively, the depthpacketparser has to get the last command_seq_ number, used in matching or a new implementation has to be made, that doesn't use the "magic1 marker".

Lars

linknum23, 2015-04-14:
I'll give that a try today. Thanks :)

mcevoyandy, 2015-04-23:
I'm having trouble with this issue as well with iai_kinect2.

@linknum23 we're you able to resolve this? For me the error happens a few minutes after running the kinect2_bridge launch file.

I have yet to see the error when I run Protonect.

linknum23, 2015-04-23:
@mcevoyandy I was unable to fix this with Lars suggestions, even after digging in deeper. It looked like it was a problem in the different depth_packet_processors, specifically the opencl_depth_packet_processor, but it felt like I was in over my head. Someone who understands this more should look into it.
Realize also that I only found this problem after modifying the iai_kinect2 code to add restart capabilities. I did not experience this in out of the box operation with the iai_kinect2.

I hope this helps.

wiedemeyer, 2015-05-20:
I just tested it, if you add `command_seq_ = 0;` to the end of the stop method, restarting the device works fine. I will create a small pull request for it.

larshg, 2015-05-20:
if #221 gets merged it won't be a problem as it doesn't use it as far as I can see :)

wiedemeyer, 2015-05-20:
Ok, but did anyone tried if restarting the device works with that pull request?

wiedemeyer, 2015-05-20:
tested it, and it works fine.

linknum23, 2015-05-20:
Fantastic. Thanks for figuring it out:)

closed 2015-11-09 by floe