OpenKinect/libfreenect2 #209: unable to locate package libturbojpeg

brant203, 2015-04-16:
When trying to install dependencies by using "sudo apt-get install -y build-essential libturbojpeg libtool autoconf libudev-dev cmake mesa-common-dev freeglut3-dev libxrandr-dev doxygen libxi-dev libopencv-dev automake" I get the error "unable to locate packge libturbojpeg." Is there any way to fix this?

xlz, 2015-04-16:
apt-get install libjpeg-turbo8-dev

brant203, 2015-04-16:
I'm getting the same error. It is important for me to note that I am trying to do this on a raspberry pi running rasbian.

xlz, 2015-04-16:
Try with apt-file:

`sudo apt-get install apt-file && sudo apt-file update && apt-file search libturbojpeg.so`

brant203, 2015-04-16:
apt-file search libturbojpeg.so did not return any results. Does this mean that there are no packages with the libturbojpeg.so file in it? Do I need to add a repo or something?

wiedemeyer, 2015-04-16:
@brant203: As far as I know the raspberry pi has only USB 2.0. And from our experience the Kinect2 requires a USB 3.0 port to work.

brant203, 2015-04-16:
I thought usb 3.0 is backwards compatible.

wiedemeyer, 2015-04-16:
You can plug it in, it will be detected, but you probably won't receive any data.

brant203, 2015-04-16:
hmmm. Is there any way to make it work that you know of?

wiedemeyer, 2015-04-16:
Normally I would suggest to get a PCIe USB 3.0 controller card. But that won't work with the raspberry pi.

brant203, 2015-04-16:
Is there a way to use the GPIO pins?

wiedemeyer, 2015-04-16:
probably no.

brant203, 2015-04-16:
hmmm. ok so it is for sure that I will get no data if i try to use this on my pi?

wiedemeyer, 2015-04-16:
You could try, but the chances for it to not work are pretty high.

floe, 2015-04-16:
It's 100% sure that the Kinect v2/One will not work with any Raspberry Pi. You need an embedded board with native USB3 support, such as the Arndale boards or the Odroid-XU. Even then, nobody has tested these yet AFAIK.

floe, 2015-04-23:
BTW, to address the original bug regarding libturbojpeg.so, this is very likely a duplicate of #36 - closing.

closed 2015-04-23 by floe

baomin, 2017-11-18:
sudo apt-get install apt-file && sudo apt-file update && apt-file search libturbojpeg.so