OpenKinect/libfreenect2 #211: Any ideas about how to use VRPN + Libfreenect2

Edwinzero, 2015-04-22:
Hello Guys,

I want to integrate multiple Kinect v2 (total number is16) into a sensor network and each computer connects two Kinect v2.
I didn't find any interface on vrpn about libfreenect2 but I find libusb interface. So do you have any ideas that creating an interface for libfreenect on vrpn?

Edwinzero, 2015-04-24:
#Installation on Windows8 detailed tutorial
This tutorial is for x64 target machine. x86 has error with libusb

##1. About libusbK driver

If you have the Kinect for Windows v2 SDK, install it first. You don't need to uninstall the SDK or the driver before doing this procedure.

Install the libusbK backend driver for libusbx:

Download Zadig from http://zadig.akeo.ie/.
Run Zadig and in options, check List All Devices and uncheck Ignore Hubs or Composite Parents
Select the Xbox NUI Sensor (composite parent) from the drop-down box. (Ignore the Interface 0 and Interface 2 varieties.) The current driver will list usbccgp. USB ID is VID 045E, PID 02C4.
Select libusbK (v3.0.6.0) from the replacement driver list.
Click the Replace Driver button. Click yes on the warning about replacing a system driver. (This is because it is a composite parent.)
Done.
To uninstall the libusbK driver (and get back the official SDK driver, if installed):

Open Device Manager
Under libusbK USB Devices, right click the "Xbox NUI Sensor (Composite Parent)" device and select uninstall.
Important: Check the "Delete the driver software for this device." checkbox, then click OK.
If you already had the official SDK driver installed and you want to use it:

In Device Manager, in the Action menu, click "Scan for hardware changes."
This will enumerate the Kinect sensor again and it will pick up the K4W2 SDK driver, and you should be ready to run KinectService.exe again immediately.

You can go back and forth between the SDK driver and the libusbK driver very quickly and easily with these steps.

##2. Preparation

###Extra Dependencies:
ALL x64 please. (Assume you have CMake)
OpenCV: http://opencv.org/downloads.html (test version 2.4.11)

  Set opencv/build as system variable: OPENCV_BUILD. This will be convinence for future using.
  
  Set opencv/build/vc"put version here"/x86 or x64/bin into Path variable.
  
OpenCL: http://developer.amd.com/tools-and-sdks/opencl-zone/amd-accelerated-parallel-processing-app-sdk/

  Automatically set.
  
libjpeg-turbo: http://sourceforge.net/projects/libjpeg-turbo/files/

  After installation, put it to ../depends/libjpeg_turbo.

  File organization like libjpeg_turbo/include; libjpeg_turbo/lib; libjpeg_turbo/bin
  
###Dependencies

Run install_deps.sh. You will get libusb_src and glfw_src.

Download libusbx-winso.zip from https://www.dropbox.com/s/madoye1ayaoajet/libusbx-winiso.zip.

Change the file organization like: 

libusb/include/libusb-1.0/libusb.h; libusb/lib/usb-1.0.dll and libusb/lib/usb-1.0.lib (Both .dll and .lib are generated by running .DLL build configuration)
(IN the later runtime of Protonect, you still need to change the library name back to libusb.dll/.lib)

##3. Congifuring in CMake.

Open CMake GUI. 

Set source path:  libfreenect2/examples/protonect/; Set build path: libfreenect2/build. 
Check grouped, advanced.

Press Configure. It should be all right. Then Generate.

###Trouble Shooting:
1. If you occurs something like "opencvconfig.cmake" not found. Please manually set the path to opencv/build(You can find these cmake files there)

2. Please double check your opencv and opencl setting (Especially the x86/x64 part!)

##4. Build in Visual Studio
Just run ALL_BUILD, it will run without any error. You can ignore the warnning.

For fast testing, you can put all the files in ../protonect/lib/Debug or Release to ../protonect/bin/Debug or Release. And put ../depends/libusb to ../protonect/bin/Debug or Release 

Just run Protonect.exe and enjoy!

##5. Install

Run INSTALL in visual studio solution. The installation path can be changed in CMake configuration. In CMAKE-> CMAKE_INSTALL_PREFIX

BenjaminRothschild, 2015-04-28:
Hello,I just tried two ways: Multi-process or Multithreading to interview the Kinect V2 based on the libfreenect2 , and made them successfully.Only two devices in my hands.So I just only tried two Kinect V2. Also I don't know VRPN,  but I think you can have a try  in your work. Whether you can add the Distributed structure and Network to achieve it.
Easy multithreading demo modefied the protonect.c file, Multi-process is easier:

#include <iostream>
#include <signal.h>
#include <opencv2/opencv.hpp>
#include <libfreenect2/libfreenect2.hpp>
#include <libfreenect2/frame_listener_impl.h>
#include <libfreenect2/threading.h>
#include <windows.h>
#include <process.h>

bool protonect_shutdown = false;

void sigint_handler(int s)
{
	protonect_shutdown = true;
}

libfreenect2::Freenect2 freenect2_1;
libfreenect2::Freenect2 freenect2_2;

libfreenect2::Freenect2Device *dev_1 = freenect2_1.openDevice(0);
libfreenect2::Freenect2Device *dev_2 = freenect2_2.openDevice(1);

HANDLE hMutex;
unsigned int __stdcall libfreenect2_1(PVOID pM);
unsigned int __stdcall libfreenect2_2(PVOID pM);

int main(int argc, char *argv[])
{
	std::string program_path(argv[0]);
	size_t executable_name_idx = program_path.rfind("Protonect");

	std::string binpath = "/";

	if(executable_name_idx != std::string::npos)
	{
		binpath = program_path.substr(0, executable_name_idx);
	}

	signal(SIGINT,sigint_handler);
	protonect_shutdown = false;

	HANDLE hThread[2];
	
	hThread[0]=(HANDLE)_beginthreadex(NULL,0,libfreenect2_1,NULL,0,NULL);
	hThread[1]=(HANDLE)_beginthreadex(NULL,0,libfreenect2_2,NULL,0,NULL);

	std::cout << "beginthreadex" << std::endl;

	WaitForMultipleObjects(2, hThread, TRUE, INFINITE);  

	CloseHandle(hThread[0]);
	CloseHandle(hThread[1]);
	return 0;
}

unsigned int __stdcall libfreenect2_1(PVOID pM)
{
	libfreenect2::SyncMultiFrameListener listener1(libfreenect2::Frame::Color | libfreenect2::Frame::Ir | libfreenect2::Frame::Depth);
	libfreenect2::FrameMap frames_1;

	if(dev_1 == 0)
	{
		std::cout << "no device connected or failure opening the default one!" << std::endl;
		return -1;
	}
	
	dev_1->setColorFrameListener(&listener1);
	dev_1->setIrAndDepthFrameListener(&listener1);
	dev_1->start();

	std::cout << "device serial: " << dev_1->getSerialNumber() << std::endl;
	std::cout << "device firmware: " << dev_1->getFirmwareVersion() << std::endl;

	while(!protonect_shutdown)
	{
		listener1.waitForNewFrame(frames_1);

		libfreenect2::Frame *rgb1 = frames_1[libfreenect2::Frame::Color];
		libfreenect2::Frame *ir1 = frames_1[libfreenect2::Frame::Ir];
		libfreenect2::Frame *depth1 = frames_1[libfreenect2::Frame::Depth];

		cv::imshow("rgb1", cv::Mat(rgb1->height, rgb1->width, CV_8UC3, rgb1->data));
		cv::imshow("ir1", cv::Mat(ir1->height, ir1->width, CV_32FC1, ir1->data) / 20000.0f);
		cv::imshow("depth1", cv::Mat(depth1->height, depth1->width, CV_32FC1, depth1->data) / 4500.0f);

		int key1 = cv::waitKey(1);

		protonect_shutdown = protonect_shutdown || (key1 > 0 && ((key1 & 0xFF) == 27)); // shutdown on escape

		listener1.release(frames_1);
	}

	dev_1->stop();
	dev_1->close();

	return 0;
}

unsigned int __stdcall libfreenect2_2(PVOID pM)
{
	libfreenect2::SyncMultiFrameListener listener2(libfreenect2::Frame::Color | libfreenect2::Frame::Ir | libfreenect2::Frame::Depth);

	libfreenect2::FrameMap frames2;

	if(dev_2 == 0)
	{
		std::cout << "no device connected or failure opening the dev1 one!" << std::endl;
		return -1;
	}

	dev_2->setColorFrameListener(&listener2);
	dev_2->setIrAndDepthFrameListener(&listener2);
	dev_2->start();

	std::cout << "device serial: " << dev_2->getSerialNumber() << std::endl;
	std::cout << "device firmware: " << dev_2->getFirmwareVersion() << std::endl;

	while(!protonect_shutdown)
	{
		listener2.waitForNewFrame(frames2);

		libfreenect2::Frame *rgb2 = frames2[libfreenect2::Frame::Color];
		libfreenect2::Frame *ir2 = frames2[libfreenect2::Frame::Ir];
		libfreenect2::Frame *depth2 = frames2[libfreenect2::Frame::Depth];

		cv::imshow("rgb2", cv::Mat(rgb2->height, rgb2->width, CV_8UC3, rgb2->data));
		cv::imshow("ir2", cv::Mat(ir2->height, ir2->width, CV_32FC1, ir2->data) / 20000.0f);
		cv::imshow("depth2", cv::Mat(depth2->height, depth2->width, CV_32FC1, depth2->data) / 4500.0f);

		int key2 = cv::waitKey(1);

		protonect_shutdown = protonect_shutdown || (key2 > 0 && ((key2 & 0xFF) == 27)); // shutdown on escape
		listener2.release(frames2);
	}
	dev_2->stop();
	dev_2->close();

	return 0;
}

Edwinzero, 2015-04-28:
@BenjaminRothschild  Man you are so cool!!!! Thanks a lot! I will try it tomorrow!

BenjaminRothschild, 2015-04-28:
except the original head files , add the <windows.h> < process.h> ,I also don't know why the head files disappeared  ~~(>_<)~~

Edwinzero, 2015-04-28:
@BenjaminRothschild Awesome man, I tried it and it works! Maybe we can do more discussion about this part! Could I have your email? Mine is zhaoxiaoshang@gmail.com

Edwinzero, 2015-05-01:
@BenjaminRothschild  I think synchronize two Kinect v2 is very very hard to do. I am considering use boost to do some synchronization work. Do you have any ideas about synchronization two Kinects?

zardchim, 2015-05-05:
@Edwinzero 

Awesome work!
My PC is running Windows 8.1, i7-4790 and GTX960.
Could I use the opencl from amd as you suggested? {OpenCL: http://developer.amd.com/tools-and-sdks/opencl-zone/amd-accelerated-parallel-processing-app-sdk/}

Edwinzero, 2015-05-05:
@zardchim 
I am using AMD opencl now. CUDA also work. Your PC is awesome....!

zardchim, 2015-05-06:
@Edwinzero 

I am following your steps but face two error when running "ALL_BUILD"

<1> error C1083: Cannot open include file: 'turbojpeg.h': No such file or directory	C:\libfreenect2-master\examples\protonect\src\turbo_jpeg_rgb_packet_processor.cpp	30

<2> Error	60	error LNK1104: cannot open file 'C:\libfreenect2-master\examples\protonect\lib\Debug\freenect2.lib'	C:\libfreenect2-master\build\LINK	Protonect

any idea >.< ?

Edwinzero, 2015-05-06:
@zardchim 
Error1 means you didn't include turbojpeg in your solution. You need to download turbo-jpegx64
When you solve error 1 there won't be error for error2

BenjaminRothschild, 2015-05-11:
hello,I send an e-mail to you in Chinese, have you read it? @Edwinzero

Edwinzero, 2015-05-12:
@BenjaminRothschild  Sorry man, I just find it out... Lots of account linked to this email...

Edwinzero, 2015-05-12:
@BenjaminRothschild  I try to store the depth image from CV_32FC1 to CV_16UC1 but I get nothing in the depth image and will get error. Do you have any ideas?

vedanshu95, 2015-05-14:
hi @Edwinzero 
i have no cmake erros now. i only have one error in vs 12 2013
[image]

Edwinzero, 2015-05-15:
@vedanshu95  you didn't put turbojpeg.lib in your lib direction. And the library's name in your library folder must match the name you enter in the property->linker->input

vedanshu95, 2015-05-15:
@Edwinzero yeah i actually installed the gcc file not the vc one. this created the problem. Now i am all done. freenect2 is working all gereat.

thanks!

vedanshu95, 2015-05-26:
@Edwinzero I am trying to use 2 or more kinects simultaneously. I changed the protonect.cpp file as mentionede above but didnt see any difference. Can you help me out with this?

Edwinzero, 2015-05-28:
@vedanshu95  I tried 5 time to work almost in real-time (29fps) without synchronize each other. And I think multi-thread is the best solution for real-time capturing. You can try pthread if you want to improve the performance. Loop means you change them from simultaneous working to working in order. That obvious will reduce the performance.

zardchim, 2015-05-28:
@edwinzero regarding multithread, have u tried the performance between using std::thread/winapi thread/ pthread? which one provide better real time performance?

Edwinzero, 2015-05-28:
@zardchim I think maybe pthread will be better since it is more widely used in different kinds of operating system. I haven't done a pthread version and it is still in progress. I will post it later.

zardchim, 2015-05-30:
@Edwinzero 
I just have the time to try the code -,-"
When two kinects is running, with opengl enabled. My CPU consumption is about 80%, and GPU 7%, do you have similar performance?

Mine CPU: i7-4790, GPU: gtx980, 16GB Ram.

While for the Windows SDK, the CPU is about 25%, and GPU 13%

Edwinzero, 2015-05-30:
@zardchim  Have you tried to allocate more memory for each thread? I change the buffer size for each thread and now my computer can handle 6 Kinects at the same time now.(30fps)

zardchim, 2015-05-31:
@Edwinzero 

Oh I see. Are you using pthread? I am using beginthreadex and it seems no such option

jasoncatchpole, 2015-07-14:
@Edwinzero Hey, just curious what your PC specs are given you can run 6 Kinects? Do you have additional USB3 cards or are you just using the ports built into the motherboard?

Edwinzero, 2015-07-17:
@jasoncatchpole 
I use two additional parallel USB3 cards, each of them connect 3 Kinects. You can find the type on Amazon.com.
The PC we are using is DELL workstation.

jasoncatchpole, 2015-07-26:
@Edwinzero 
Thanks for the info. Yeah there are lists of specific USB3 chipsets that the Kinect works with as it is a little particular. There are heaps of USB3 cards on Amazon.com. Is there a specific one you use? Presumably it has multiple USB3 controllers on a single card so you can connect 3 Kinects from it?

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe

Zantinga, 2016-03-29:
Does anyone know of a way similar to @BenjaminRothschild 's code that works in linux?