OpenKinect/libfreenect2 #213: Can't build libfreenect2 in Win32

Edwinzero, 2015-04-24:
Hello guys,

I think there are something wrong with transfer_pool.cpp when it is compiled in 32bit.
The build procedure is good but the result is like below.
[image]
Can't open Protonect.exe since they shows fail to submit transfer. And this leads not a jpeg file.
Is there any solutions for this?

I also have a small suggestion that modify the CMakeLists.txt  library part.
```
SET(LIBRARIES
  usb-1.0
  ${OpenCV_LIBS}
  turbojpeg
  ${LIBFREENECT2_THREADING_LIBRARIES}
)
```
to
```
SET(LIBRARIES
  libusb-1.0
  ${OpenCV_LIBS}
  turbojpeg
  ${LIBFREENECT2_THREADING_LIBRARIES}
)
```
This will work for windows building.
@larshg @floe @christiankerl

larshg, 2015-04-25:
@Edwinzero What libusb version do you use? Recompiled for 32 bit?

I can't see why it shouldn't work in 32 bit. Maybe I'll try it later.

I think the RGB usb transfers is working, but not the transfers for depth packages.

You can try to only register for RGB image and comment out the depth/ir image handling, to see if you can get RGB images.

Edwinzero, 2015-04-25:
Yes, @larshg. I am using libusb_src to build Win32 Release and I use the dll libraries:
libjpeg_turbo 32bit too. AMD opencl and Opencv are 32 bits too. I will try only RGB first.

xlz, 2015-04-29:
I am able to build it and run Protonect.exe correctly under 32-bit.

I used [this guide](https://github.com/xlz/libfreenect2/blob/cuda/depends/README.depends.txt) to build. (Replace x64 with x86 or Win32, and remove Win64 in cmake parameters). Compare it with your steps.

xlz, 2015-04-29:
In addition, TurboJPEG is slower (43Hz) in 32-bit than 64-bit (50Hz).

Edwinzero, 2015-04-29:
@xlz  Thank you for the tutorial, I will try it now! But based on the performance I think I 'd better stick on x64 platform.

closed 2015-04-29 by Edwinzero