OpenKinect/libfreenect2 #214: Crash on OSX Yosemite

andrewadare, 2015-04-27:
To add a bit of info to the issues filed by others, I ran Protonect under lldb:

```
[bin]$ lldb Protonect
(lldb) target create "Protonect"
Current executable set to 'Protonect' (x86_64).
(lldb) run
Process 50675 launched: '/Users/adare/cv/libfreenect2/examples/protonect/bin/Protonect' (x86_64)
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 8 usb devices connected
[Freenect2Impl] found valid Kinect v2 @21:20 with serial 013274450547
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 15 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 ef 00 17 40 00 08 00 00 10 00 00 00    2e2133552e2e2e402e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 09 13 14 47 4d 46 31 35 32 2e 31    31332e2e2e2e2e2e474d463135322e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 013274450547
device firmware: 2.3.3913.0.7
[DepthPacketStreamParser::handleNewData] not all subsequences received 3
[TurboJpegRgbPacketProcessor] avg. time: 15.4878ms -> ~64.5671Hz
[TurboJpegRgbPacketProcessor] avg. time: 15.3508ms -> ~65.143Hz
[TurboJpegRgbPacketProcessor] avg. time: 14.2768ms -> ~70.0435Hz
[TurboJpegRgbPacketProcessor] avg. time: 14.2935ms -> ~69.9618Hz
[TurboJpegRgbPacketProcessor] avg. time: 14.271ms -> ~70.072Hz
Process 50675 stopped
* thread #9: tid = 0x1113afd, 0x00007fff8de34286 libsystem_kernel.dylib`__pthread_kill + 10, stop reason = signal SIGABRT
    frame #0: 0x00007fff8de34286 libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff8de34286 <+10>: jae    0x7fff8de34290            ; <+20>
    0x7fff8de34288 <+12>: movq   %rax, %rdi
    0x7fff8de3428b <+15>: jmp    0x7fff8de2fc53            ; cerror_nocancel
    0x7fff8de34290 <+20>: retq
(lldb)
```
Then I stepped once:
```
(lldb) step
Process 50675 exited with status = 0 (0x00000000) Terminated due to signal 6
```

Thanks for any help you can provide.

xlz, 2015-04-29:
Can you provide a backtrace with the command "bt"?

andrewadare, 2015-04-29:
Sure:
```
[bin]$ lldb Protonect
(lldb) target create "Protonect"
Current executable set to 'Protonect' (x86_64).
(lldb) run
Process 24167 launched: '/Users/adare/cv/libfreenect2/examples/protonect/bin/Protonect' (x86_64)
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 6 usb devices connected
[Freenect2Impl] found valid Kinect v2 @21:20 with serial 018132634047
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 07 0a 01 47 48 56 32 32 37 2e 31    31332e2e2e2e2e2e4748563232372e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 018132634047
device firmware: 4.1.3912.0.7
[TurboJpegRgbPacketProcessor] avg. time: 15.5538ms -> ~64.2932Hz
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 1015
[TurboJpegRgbPacketProcessor] avg. time: 15.4545ms -> ~64.7061Hz
[TurboJpegRgbPacketProcessor] avg. time: 15.356ms -> ~65.1213Hz
[TurboJpegRgbPacketProcessor] avg. time: 15.264ms -> ~65.5135Hz
[TurboJpegRgbPacketProcessor] avg. time: 14.2918ms -> ~69.9702Hz
Process 24167 stopped
* thread #9: tid = 0xa085f, 0x00007fff9b564286 libsystem_kernel.dylib`__pthread_kill + 10, stop reason = signal SIGABRT
    frame #0: 0x00007fff9b564286 libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff9b564286 <+10>: jae    0x7fff9b564290            ; <+20>
    0x7fff9b564288 <+12>: movq   %rax, %rdi
    0x7fff9b56428b <+15>: jmp    0x7fff9b55fc53            ; cerror_nocancel
    0x7fff9b564290 <+20>: retq
(lldb) bt
* thread #9: tid = 0xa085f, 0x00007fff9b564286 libsystem_kernel.dylib`__pthread_kill + 10, stop reason = signal SIGABRT
  * frame #0: 0x00007fff9b564286 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff8e8cc42f libsystem_pthread.dylib`pthread_kill + 90
    frame #2: 0x00007fff9a63eb53 libsystem_c.dylib`abort + 129
    frame #3: 0x00007fff8ceeab81 libGPUSupportMercury.dylib`gpusGenerateCrashLog + 173
    frame #4: 0x000012344022728a GeForceGLDriver`___lldb_unnamed_function6431$$GeForceGLDriver + 9
    frame #5: 0x00007fff8ceec068 libGPUSupportMercury.dylib`gpusSubmitDataBuffers + 500
    frame #6: 0x000012344031a49b GeForceGLDriver`___lldb_unnamed_function11146$$GeForceGLDriver + 349
    frame #7: 0x000012344031a301 GeForceGLDriver`___lldb_unnamed_function11145$$GeForceGLDriver + 232
    frame #8: 0x000012344032fbe4 GeForceGLDriver`___lldb_unnamed_function11281$$GeForceGLDriver + 4774
    frame #9: 0x000000010c312bf8 GLEngine`glReadPixels_Exec + 1390
    frame #10: 0x00007fff9b53cbd5 libGL.dylib`glReadPixels + 57
    frame #11: 0x00000001000316af libfreenect2.dylib`libfreenect2::Texture<libfreenect2::ImageFormat<4ul, 33326u, 6403u, 5126u> >::downloadToNewFrame() [inlined] libfreenect2::Texture<libfreenect2::ImageFormat<4ul, 33326u, 6403u, 5126u> >::downloadToBuffer(data=<unavailable>) + 111 at opengl_depth_packet_processor.cpp:300
    frame #12: 0x0000000100031690 libfreenect2.dylib`libfreenect2::Texture<libfreenect2::ImageFormat<4ul, 33326u, 6403u, 5126u> >::downloadToNewFrame(this=0x0000000106826f18) + 80 at opengl_depth_packet_processor.cpp:329
    frame #13: 0x000000010002fc0c libfreenect2.dylib`libfreenect2::OpenGLDepthPacketProcessorImpl::run(this=0x0000000106826c00, ir=0x0000000110580e78, depth=0x0000000110580e70) + 972 at opengl_depth_packet_processor.cpp:653
    frame #14: 0x000000010002e3a8 libfreenect2.dylib`libfreenect2::OpenGLDepthPacketProcessor::process(this=0x00000001029012d0, packet=<unavailable>) + 232 at opengl_depth_packet_processor.cpp:935
    frame #15: 0x000000010001aa71 libfreenect2.dylib`libfreenect2::AsyncPacketProcessor<libfreenect2::DepthPacket>::execute(this=0x0000000102b3ccf0) + 81 at async_packet_processor.h:105
    frame #16: 0x000000010002acd3 libfreenect2.dylib`tthread::thread::wrapper_function(aArg=0x0000000102b3c490) + 19 at tinythread.cpp:169
    frame #17: 0x00007fff8e8ca268 libsystem_pthread.dylib`_pthread_body + 131
    frame #18: 0x00007fff8e8ca1e5 libsystem_pthread.dylib`_pthread_start + 176
    frame #19: 0x00007fff8e8c841d libsystem_pthread.dylib`thread_start + 13
(lldb)
```

xlz, 2015-04-29:
This is a duplicate of issue #205 .

closed 2015-05-21 by floe