OpenKinect/libfreenect2 #216: Ubuntu 14.04 libusb error LIBUSB_ERROR_NO_DEVICE (-4)

rastaxe, 2015-04-28:
I am trying to run Kinect v2 on Ubuntu 14.04 (kernel 3.16)
When I run Protonect I got the -4 libusb error (LIBUSB_ERROR_NO_DEVICE).
The output of the program is
```
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 15 usb devices connected
[Freenect2Impl] found valid Kinect v2 @9:3 with serial 013375243747
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 05 0b 14 47 4d 46 36 30 33 2e 31    31332e2e2e2e2e2e474d463630332e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 013375243747
device firmware: 4.3.3913.0.7
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] R is -4  ## I added this print on the code!
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] R is -4
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] R is -4
...
```

When I plug the kinect, I got from dmesg:
```
usb 8-2: new high-speed USB device number 3 using xhci_hcd
[  324.411131] usb 8-2: New USB device found, idVendor=045e, idProduct=02d9
[  324.411140] usb 8-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  324.411146] usb 8-2: Product: NuiSensor Adaptor      
[  324.411150] usb 8-2: Manufacturer: Microsoft Corporation  
[  324.412541] hub 8-2:1.0: USB hub found
[  324.412702] hub 8-2:1.0: 1 port detected
[  324.857621] usb 9-2: new SuperSpeed USB device number 4 using xhci_hcd
[  327.837733] usb 9-2: New USB device found, idVendor=045e, idProduct=02d9
[  327.837742] usb 9-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  327.837747] usb 9-2: Product: NuiSensor Adaptor      
[  327.837751] usb 9-2: Manufacturer: Microsoft Corporation  
[  327.839117] hub 9-2:1.0: USB hub found
[  327.839362] hub 9-2:1.0: 1 port detected
[  328.310234] usb 9-2.1: new SuperSpeed USB device number 5 using xhci_hcd
[  328.326682] usb 9-2.1: New USB device found, idVendor=045e, idProduct=02c4
[  328.326690] usb 9-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[  328.326696] usb 9-2.1: Product: Xbox NUI Sensor
[  328.326700] usb 9-2.1: Manufacturer: Microsoft
[  328.326704] usb 9-2.1: SerialNumber: 013375243747
```
As soon as I start Protonect I got:
```
[  502.613637] usb 9-2.1: reset SuperSpeed USB device number 5 using xhci_hcd
[  502.629167] xhci_hcd 0000:03:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff880036aa8e00
[  502.629175] xhci_hcd 0000:03:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff880036aa8e48
[  502.629180] xhci_hcd 0000:03:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff880036aa8ed8
[  502.629185] xhci_hcd 0000:03:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff880036aa8e90
[  502.689949] xhci_hcd 0000:03:00.0: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[  502.976943] xhci_hcd 0000:03:00.0: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[  502.985108] xhci_hcd 0000:03:00.0: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[  502.985625] xhci_hcd 0000:03:00.0: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[  502.991287] xhci_hcd 0000:03:00.0: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[  503.234346] xhci_hcd 0000:03:00.0: ERROR Transfer event TRB DMA ptr not part of current TD
[  503.234589] xhci_hcd 0000:03:00.0: ERROR Transfer event TRB DMA ptr not part of current TD
[  503.234842] xhci_hcd 0000:03:00.0: ERROR Transfer event TRB DMA ptr not part of current TD
...a lot of these lines...
[  509.238571] xhci_hcd 0000:03:00.0: xHCI host not responding to stop endpoint command.
[  509.238580] xhci_hcd 0000:03:00.0: Assuming host is dying, halting host.
[  509.238799] usb 9-2.1: usbfs: usb_submit_urb returned -22
[  509.238886] usb 9-2.1: usbfs: usb_submit_urb returned -22
[  509.238941] usb 9-2.1: usbfs: usb_submit_urb returned -22
[  509.238995] usb 9-2.1: usbfs: usb_submit_urb returned -22
[  509.239049] xhci_hcd 0000:03:00.0: HC died; cleaning up
[  509.239051] usb 9-2.1: usbfs: usb_submit_urb returned -22
[  509.239094] usb 8-2: USB disconnect, device number 3
```
Protonect is using the patched version of libusb:
```
 ldd Protonect | grep usb
	libusb-1.0.so.0 => /home/ale/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0x00007f418b0c0000)
```
Listing the USB Controller gives:
```
lspci | grep USB
00:12.0 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB OHCI0 Controller
00:12.1 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0 USB OHCI1 Controller
00:12.2 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB EHCI Controller
00:13.0 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB OHCI0 Controller
00:13.1 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0 USB OHCI1 Controller
00:13.2 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB EHCI Controller
00:14.5 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB OHCI2 Controller
03:00.0 USB controller: ASMedia Technology Inc. ASM1042 SuperSpeed USB Host Controller
```

Any ideas to solve this problem?

wiedemeyer, 2015-04-28:
Take a look to this issue #164. Maybe that helps for your problem as well.

rastaxe, 2015-04-28:
I reduced the MAX_ISO_BUFFER_LENGTH to 49152, but it did not help. I tried also the solution of #174, but still does not work. It is frustrating because any time I use the Protonect program all the usb ports stop to work and I have to restart the PC. 
     In the meanwhile, I am trying the same thing on a Intel NUC and at least the RGB works, but the depth and ir windows are black. I noted that this is a common problem. As far as I understood, I have several options to solve this problem: upgrade OpenGL (the NUC has OpenGL 3.0), or compile with the CpuPacketPipeline (how can I do that?) or use OpenCL. What is the best option?

xlz, 2015-04-29:
Since you have not mentioned it, I have to ask: have you checked the USB autosuspend settings? Ubuntu 14.04 has power saving config by default.

The OpenGL processor is basically using GPU but with overhead of abstraction several layers removed from the hardware. You should be able to do just fine with `cmake .. -DENABLE_OPENGL:BOOL=OFF`.

I have a JPEG decoding acceleration patch PR #210 for Intel. You might also want to try it out.

rastaxe, 2015-04-29:
I tried the autosuspend as you suggest in #174, but it did not work.

On the NUC I tried with OpenCL and disabling the OpenGL and it works. I will try the JPEG decoding patch you suggest. Thank you!

xlz, 2015-04-29:
I think this is a kernel level issue. As you can see
```
[  509.238571] xhci_hcd 0000:03:00.0: xHCI host not responding to stop endpoint command.
[  509.238580] xhci_hcd 0000:03:00.0: Assuming host is dying, halting host.
```
LIBUSB_ERROR_NO_DEVICE came from here.

> It is frustrating because any time I use the Protonect program all the usb ports stop to work and I have to restart the PC. 

This is something between the ASMedia USB3 controller and its kernel driver. I don't know what more can be done within libusb or here. You can still try different versions of kernel, from 3.12 to 4.0. Further than that, I guess you have to assume this ASMedia controller is not supported.

zhangtemplar, 2015-05-14:
The same issue is me. I tried many combinations, e.g., disabling opencl, disabling opengl and none of them works.

@xlz to reset the usb port, you can use the following command (usb3.0 port only)
for i in $(ls /sys/bus/pci/drivers/xhci_hcd/|grep :)
 do echo $i >/sys/bus/pci/drivers/xhci_hcd/unbind
 echo $i >/sys/bus/pci/drivers/xhci_hcd/bind
done

My output is 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 13 usb devices connected
[Freenect2Impl] found valid Kinect v2 @6:3 with serial 028336645047
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 ef 00 17 40 00 08 00 00 10 00 00 00    2e2133552e2e2e402e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 09 04 0c 47 4d 48 38 35 31 2e 31    31332e2e2e2e2e2e474d483835312e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  5d 26 00 00                                        5d262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  5f 26 00 00                                        5f262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 028336645047
device firmware: 2.3.3913.0.7
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
[TurboJpegRgbPacketProcessor] avg. time: 15.7738ms -> ~63.3962Hz
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[... 25 lines cut ...]
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4
[TransferPool::submit] failed to submit transfer -4

xlz, 2015-05-14:
@zhangtemplar 
`lsusb` and `dmesg`?

bellekci, 2015-06-28:
I am getting the same LIBUSB_ERROR_NO_DEVICE error on Ubuntu 15.04 with kernel 3.19.0-21-generic.

dmesg starts to following errors right after protonect is started. After a while, device gets disconnected and program stops.

dmesg:

[  441.150310] xhci_hcd 0000:00:14.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 8 comp_code 4
[  441.150323] xhci_hcd 0000:00:14.0: Looking for event-dma 0000000035dd02c0 trb-start 0000000035dd02d0 trb-end 0000000035dd02d0 seg-start 0000000035dd0000 seg-end 0000000035dd03f0

lsusb returns:

Bus 003 Device 002: ID 8087:8001 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 045e:02d8 Microsoft Corp. 
Bus 002 Device 002: ID 045e:02d9 Microsoft Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 8087:07dc Intel Corp. 
Bus 001 Device 007: ID 17e9:03cc DisplayLink 
Bus 001 Device 008: ID 0eef:0001 D-WAV Scientific Co., Ltd eGalax TouchScreen
Bus 001 Device 004: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 003: ID 045e:02d9 Microsoft Corp. 
Bus 001 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lspci returns:
00:00.0 Host bridge: Intel Corporation Broadwell-U Host Bridge -OPI (rev 09)
00:02.0 VGA compatible controller: Intel Corporation Broadwell-U Integrated Graphics (rev 09)
00:03.0 Audio device: Intel Corporation Broadwell-U Audio Controller (rev 09)
00:14.0 USB controller: Intel Corporation Wildcat Point-LP USB xHCI Controller (rev 03)
00:16.0 Communication controller: Intel Corporation Wildcat Point-LP MEI Controller #1 (rev 03)
00:19.0 Ethernet controller: Intel Corporation Ethernet Connection (3) I218-V (rev 03)
00:1b.0 Audio device: Intel Corporation Wildcat Point-LP High Definition Audio Controller (rev 03)
00:1c.0 PCI bridge: Intel Corporation Wildcat Point-LP PCI Express Root Port #1 (rev e3)
00:1c.3 PCI bridge: Intel Corporation Wildcat Point-LP PCI Express Root Port #4 (rev e3)
00:1d.0 USB controller: Intel Corporation Wildcat Point-LP USB EHCI Controller (rev 03)
00:1f.0 ISA bridge: Intel Corporation Wildcat Point-LP LPC Controller (rev 03)
00:1f.2 SATA controller: Intel Corporation Wildcat Point-LP SATA Controller [AHCI Mode] (rev 03)
00:1f.3 SMBus: Intel Corporation Wildcat Point-LP SMBus Controller (rev 03)
02:00.0 Network controller: Intel Corporation Wireless 7260 (rev bb)

ldd Protonect | grep usb returns:

	libusb-1.0.so.0 => /home/bellekci/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0x00007f1937fb4000)

floe, 2015-06-29:
00:14.0 USB controller: Intel Corporation Wildcat Point-LP USB xHCI Controller (rev 03)

AFAICT this controller hasn't yet been tested, this seems to be a pretty new one. The only ones we can confirm as working are those listed in the README; for any other controller, it may or may not work. Trying a different USB controller (either on a different machine or on an add-on card) will help to narrow it down.

zhangtemplar, 2015-07-02:
@xlz I have found the problem that my usb 3.0 controller in the mother board is Asmedia, which is obviously in non-compatible list of the Kinect 2.0. After switch to one suggested in the Kinect 2.0 support site, it works with no problem.

xlz, 2015-10-24:
@rastaxe @zhangtemplar @bellekci 

"ERROR Transfer event TRB DMA ptr not part of current TD" is the place the bug happens. What immediately follows is "HC is busted, give up!" and the rest is the fallout.

This may have something to do with the XHCI_SPURIOUS_SUCCESS quirk. What is the hci version from your `dmesg | grep 'hci version'`? If it's 0x96 or less, the XHCI_SPURIOUS_SUCCESS quirk is not added. If so you can try to boot the kernel (at least 3.14) with parameter `xhci_hcd.quirks=0x10` to see if it is this issue. The result may be interesting.

bellekci, 2015-12-02:
@xlz sorry for late return, somehow slipped from my attention.

`dmesg | grep 'hci version'` returns unfortunately nothing. I am running 3.19.0.22-generic.
I have tested many kernels up to 4.2 but usb-freeze problem was still there. I will try to boot with `xhci_hcd.quirks=0x10` now.

bellekci, 2015-12-02:
I booted with kernel 4.3 and no additional quirk parameter and this is what get for `dmesg | grep 'hci version'`

`[    0.625109] xhci_hcd 0000:00:14.0: hcc params 0x200077c1 hci version 0x100 quirks 0x00009810`

Thus, XHCI_SPURIOUS_SUCCESS quirk is added and I do not know if I still have the no-device problem as it takes some time until device gets disconnected. I will let you know.

bellekci, 2015-12-02:
Yes, the problem is still there even with the XHCI_SPURIOUS_SUCCESS quirk.

Before disconnection, dmesg issues following warnings:

`[  378.038245] perf interrupt took too long (2524 > 2500), lowering kernel.perf_event_max_sample_rate to 50000
[  479.772372] perf interrupt took too long (5070 > 5000), lowering kernel.perf_event_max_sample_rate to 25000
[  996.839996] perf interrupt took too long (10123 > 10000), lowering kernel.perf_event_max_sample_rate to 12500`

I also got this error:

`[   14.513302] [drm:gen8_irq_handler [i915]] *ERROR* The master control interrupt lied (SDE)!`

xlz, 2015-12-02:
@bellekci Thanks for testing. I guess kernel's USB 3.0 maintenance has really gone down hill since Sarah Sharp.

closed 2015-12-23 by floe