OpenKinect/libfreenect2 #217: freenect2Config.cmake

Algomorph, 2015-04-29:
This file has the lines:
```CMake
IF("1")
    SET(freenect2_INCLUDE_DIRS ${freenect2_INCLUDE_DIR} ${freenect2_INCLUDE_DIR}/tinythread)
ENDIF("1")
```
On Ubuntu, with CMake 3.2.1, in CMake applications using freenect2,
```CMake 
message(STATUS "${freenect2_INCLUDE_DIRS}")
``` 
produces ```/usr/local/include;/usr/local/include/tinythread```.
It seems like the correct last entry should be ```/usr/local/include/libfeenect2/tinythread```.
This would be produced by changing the above code to ```SET(freenect2_INCLUDE_DIRS ${freenect2_INCLUDE_DIR} ${freenect2_INCLUDE_DIR}/freenect2/tinythread)``` in the freenectConfig.cmake file, and the bug should be fixed.

However, I'm not sure if that is the intended use. I.e. perhaps it is more reasonable to force user code to include the whole path, i.e. ```#include 'freenect2/tinythread/tinythread.h'``` instead. Either way, this prompts a change.

johannesmik, 2015-05-15:
I experienced the same error. (Ubuntu 14.04, cmake 2.8)

More specifically, I got a `tinythread.h: No such file or directory` error when linking something that included `<threading.h>`.

For me, replacing /examples/protonect/freenect2.cmake.in

```cmake
SET(freenect2_INCLUDE_DIRS ${freenect2_INCLUDE_DIR} ${freenect2_INCLUDE_DIR}/tinythread)
```
with 
```cmake
SET(freenect2_INCLUDE_DIRS ${freenect2_INCLUDE_DIR} ${freenect2_INCLUDE_DIR}/@PROJECT_NAME@/tinythread)
```

solved that. That fix looks reasonable to me because `libfreenect2/examples/protonect/CMakeLists.txt` (line 171-173) puts tinythread.h into the `include/libfreenect2/tinythread` directory:

```cmake
IF(LIBFREENECT2_THREADING_TINYTHREAD)
  INSTALL(FILES  "${MY_DIR}/src/tinythread/tinythread.h" DESTINATION include/${PROJECT_NAME}/tinythread/)
ENDIF(LIBFREENECT2_THREADING_TINYTHREAD)
```

Algomorph, 2015-05-15:
Strange, I was under the impression that this was resolved. Not sure when I will have a chance to take a look.

Algomorph, 2015-05-15:
Yeah, I thought  #131 fixed it. @johannesmik, are you on the latest commit?

johannesmik, 2015-05-15:
Yep, I'm on 2e72fa9 right now.

xlz, 2015-09-03:
`tinythread.h` is to be removed from the exported header of libfreenect2. libfreenect2 is not a threading library. Users can choose their own threading libraries.

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe