OpenKinect/libfreenect2 #218: Failed to synchronize two Kinectv2 RGB and Depth images

Edwinzero, 2015-05-01:
Hello guys,

I have already implemented two Kinect v2 on one computer without realtime refresh rate( fps: 18-22)
But there is huge distortion on time sequence between these two devices.
Is there any ideas for synchronizing two Kinect v2 so that we can make sure each pair of frames from same time stamp will be aligned?

Edwinzero, 2015-05-01:
Hey guys,  @larshg  @JoshBlake  @BenjaminRothschild

I found that libfreenect use pthread for multiple threading. If I use pthread instead of using Windows default thread libraries, will the performance be better. Maybe reach 30fps for both?

floe, 2015-05-08:
USB is notoriously bad at synchronizing different devices, even on the same bus, so I'm not sure if there is much you can do about it. Performance may of course be a secondary issue here.

jonnycowboy, 2015-05-08:
Probably two different paths to fix the issue completely - 1) as @floe says, performance is an issue, a faster computer would help.  2) use an external identifier (flashing LED, clock, something similar) that the Kinect reads through its camera and synchronizes itself every minute or two...

Edwinzero, 2015-05-08:
Hey guys, I am trying to use parallel USB 3.0 board to test two Kinect. I will give you guys a reply after I done that.

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe