OpenKinect/libfreenect2 #219: Kinect2 Serial Number

dvonhausen, 2015-05-02:
I have libfreenect2 running fine on my box and now I want to run with two Kinects. I read the other threads on this and my question is how do I run two instances of Protonect and specify the serial numbers of each Kinect? It appears that Protonect just uses the first Kinect that it finds.

Edwinzero, 2015-05-03:
Please check this issue#211

dvonhausen, 2015-05-05:
Thanks - it looks like what I need but its for Windows. Is there code for this that is ported to linux/ubuntu?

Edwinzero, 2015-05-06:
@dvonhausen 
I think you can use pthread on Linux, which is I am trying to do on windows right now. If you have solution to make two Kinect2 both in real-time(30fps), I hope you could let me know.

floe, 2015-05-22:
This will be addressed by PR #82 (or its successor). Closing here.

closed 2015-05-22 by floe