OpenKinect/libfreenect2 #220: Bulk Transfer fails with error -7 (LIBUSB_ERROR_TIMEOUT)

atomala, 2015-05-02:
I am working with a few others on interfacing a Kinect v2 with ROS via libfreenect2. We are using Ubuntu 14.04.2 on a Parallels VM. Currently, we are getting a weird timeout error that I can't resolve. 

We are running `roslaunch` on the following file:
```
<launch>
  <node name="kinect2_bridge" pkg="kinect2_bridge" type="kinect2_bridge" output="screen">
    <param name="depth_method" type="string" value="cpu"/>
    <param name="reg_method" type="string" value="cpu"/>
  </node>
</launch>
```
And we get the following output:
```
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 6 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:2 with serial 030921644947
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
 0: 030921644947 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    ............C..A
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
0x0020:  0a 21 33 55 ef 00 17 40 00 08 00 00 10 00 00 00    .!3U...@........
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    ................
0x0040:  31 33 00 00 00 00 0d 18 47 4b 53 38 39 35 2e 31    13......GKS895.1
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                X...........

[CommandTransaction::receive] bulk transfer failed! libusb error -7: LIBUSB_ERROR_TIMEOUT
[CommandTransaction::execute] missing response complete!
[kinect2_bridge-1] process has died [pid 2645, exit code -11, cmd /home/ros/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge __name:=kinect2_bridge __log:=/home/ros/.ros/log/53a5fe68-f06e-11e4-a1f1-001c42dfcfd3/kinect2_bridge-1.log].
log file: /home/ros/.ros/log/53a5fe68-f06e-11e4-a1f1-001c42dfcfd3/kinect2_bridge-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

I noticed that others had a similar issue before (#8), but the issue was solved by using the 3.16 Linux Kernel, which is what we are running. We are confused about what the issue can be.

### Possibly useful info

`lsusb` returns

```
Bus 001 Device 002: ID 203a:fff9  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 045e:02c4 Microsoft Corp. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

`grep . /sys/bus/usb/devices/*/power/autosuspend ` returns:

```
/sys/bus/usb/devices/1-1/power/autosuspend:-1
/sys/bus/usb/devices/4-1/power/autosuspend:-1
/sys/bus/usb/devices/usb1/power/autosuspend:0
/sys/bus/usb/devices/usb2/power/autosuspend:0
/sys/bus/usb/devices/usb3/power/autosuspend:0
/sys/bus/usb/devices/usb4/power/autosuspend:0
```

`lspci | grep USB` returns:

```
00:1d.0 USB controller: Intel Corporation 82801FB/FBM/FR/FW/FRW (ICH6 Family) USB UHCI #1 (rev 02)
00:1d.6 USB controller: NEC Corporation uPD720200 USB 3.0 Host Controller (rev 03)
00:1d.7 USB controller: Intel Corporation 82801FB/FBM/FR/FW/FRW (ICH6 Family) USB2 EHCI Controller (rev 02)
```

`uname -r` returns:

```
3.16.0-031600-generic
```

xlz, 2015-05-02:
Virtual machines usually have bad USB 3 support. There are also multiple reports about USB errors running natively on Mac OS. I guess it is most likely an issue with the VM.

Edwinzero, 2015-05-03:
@xlz  I want to ask that if I use parallel USB3.0 board, is it possible to increase the frame rate when I use two Kinectv2 on one computer?

atomala, 2015-05-03:
I'll take a look into using a native Ubuntu installation. It may take a while unfortunately.

guanchar, 2015-06-03:
This is working native on a native Ubuntu 14.04 installation on a Lenovo Y40-80.

atomala, 2015-06-03:
Can confirm that using a native installation fixes the problem.

closed 2015-06-03 by atomala

bobperet, 2015-06-04:
I seem to be having the same issue on a native ubuntu (trusty) system. running "Protonect" works just fine so I think the USB3 port is configured properly.

------snippet from "roslaunch kinect2_bridge kinect2_bridge.launch"-------
 
SUMMARY
========

PARAMETERS
 * /kinect2_points_xyz/queue_size: 2
 * /rosdistro: indigo
 * /rosversion: 1.11.13

NODES
  /
    kinect2_bridge (kinect2_bridge/kinect2_bridge)
    kinect2_points_xyz (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[kinect2_bridge-1]: started with pid [12749]
process[kinect2_points_xyz-2]: started with pid [12750]
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 14 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:7 with serial 500264443042
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 500264443042 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 07 01 05 47 4b 53 39 31 36 2e 31    31332e2e2e2e2e2e474b533931362e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  7f 26 00 00                                        7f262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  7f 26 00 00                                        7f262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started

device serial: 500264443042
device firmware: 4.3.3912.0.7
[kinect2_bridge-1] process has died [pid 12749, exit code -11, cmd /opt/ros/indigo/lib/kinect2_bridge/kinect2_bridge -fps -1 -scale raw -calib /opt/ros/indigo/share/kinect2_bridge/data/ -depthProc cpu -clRegSource /opt/ros/indigo/share/kinect2_depth_registration/cl/depth_registration.cl __name:=kinect2_bridge __log:=/home/bperet/.ros/log/392f36f0-0ae3-11e5-8ffe-6c881481bfec/kinect2_bridge-1.log].
log file: /home/bperet/.ros/log/392f36f0-0ae3-11e5-8ffe-6c881481bfec/kinect2_bridge-1*.log
[kinect2_bridge-1] restarting process
--------- end snippet ----------

This repeats itself until I kill it

whatisor, 2016-02-03:
I got this issue many times in my Ubuntu 14.04 x64. :(