OpenKinect/libfreenect2 #223: Registration

giacomodabisias, 2015-05-05:
Hi,
I am trying to apply the new registration you did with the new parameters taken directly from the kinect2.

I created the Registration object and took the parameters from the device with the 
getIrCameraParams()
getColorCameraParams()

functions.
Now I have to use the apply function of the class, but I am wondering what size the two images should have. Should I resize the rgb to the depth size or can I use the function without resizing the images?

Thanks

xlz, 2015-05-05:
#190 has demos showing how to use the registration class. Code is here https://github.com/larshg/libfreenect2/tree/registration .

floe, 2015-05-05:
Thanks @xlz - I will hopefully get around to integrating these examples into the master branch soon (a bit swamped with work at the moment :-).

giacomodabisias, 2015-05-06:
I tried the registration, but I assume there must be something wrong in the parameters probably (?)
Here is the cloud created with the registration:
[image]

while the one made with my own calibration looks like this:
[image]

I could create a small video to see the difference if needed.

floe, 2015-05-06:
Can you describe in a bit more detail how they differ? They look quite similar to me... but in general, a proper chessboard calibration will usually have higher quality than the factory calibration, which is IMHO optimized for speed.

giacomodabisias, 2015-05-06:
I will prepare a short video, so you can see better. If you enlarge the two
images and look at the hand you will see a lot of noise

floe, 2015-05-08:
PR #226 contains a minimal implementation of the registration code; could you give that a try and see if it still looks wrong? Note that this is not yet using a depth buffer, so some pixel duplication is inevitable.

giacomodabisias, 2015-05-11:
Hi,
I just checked it and it doesn't look too bad. I will recheck everything
and let you know.

Thanks

floe, 2015-05-12:
I've merged PR #223 with the basic example code, I'll use this issue to track the remaining features to be implemented:
- [ ] add depth buffer to avoid color pixel duplication
- [ ] always use 4-channel image as output (optionally using the 4th channel for downsampled depth data)
- [ ] add helper functions for point clouds (i.e. xyzRGB output)?

xlz, 2015-05-13:
Is undistorted depth image required for generating point clouds?

JuantAldea, 2015-05-13:
I'd say so, just to keep straight lines as straight as possible.

floe, 2015-07-06:
This has been sitting here for a while; I've just remembered that a few bits and pieces regarding registration are still open. In particular, there's one thing I hope someone can help with (/cc @xlz @wiedemeyer): currently, we map color onto depth simply because that's what the reverse-engineered code from the original SDK does. 

That's fine, but what would be the best way to also provide a reversed mapping (depth onto color, in order to have a full HD image with depth information)? Right now, the best solution I can think of would be to extend the depth filtering code from #253 and basically overlay a 5x3 upsampled depth image over the color image. Thoughts?

xlz, 2015-07-06:
I think @weidemeyer's approach in iai Kinect2 is  most reasonable: downsample and/or crop rgb to a size similar to depth resolution and map depth to color with minimal upsampling. A 1920x1080 depth image is quite large to work with for segmentation, voxel filtering etc. It is still easy to locate object positions given visual detection from either original resolution or downsampled rgb images.

wiedemeyer, 2015-07-07:
@floe with the current registration we can compute for each depth pixel the corresponding color pixel. This would enable different ways to create HD or quarter HD depth images.
We could create a color image sized depth image, which just consists of sparse depth points by using the corresponding coordinates. To fill it up, one could either set the neighborhood around each sparse depth pixel to the same value, or interpolate between neighboring sparse depth pixels, or something else.
Another solution might be to up-scale and interpolate the depth image by factor 3 to 4 and then compute the corresponding color pixel coordinates, and fill a color sized depth image, which would be less sparse or even not sparse at all.

What I did in iai_kinect2 is to convert the depth image to a 1920x1080 image with the same intrinsic parameters as the color image. The up-scaling combines the undistortion of the depth image and some kind of linear edge aware interpolation. The registration is then a simple rotation, translation and projection. This could be pretty accurate but you need to know the exact rotation and translation between the color and ir sensor. You can freely use my implementation and integrate it into libfreenect2, but it is nothing that would run out of the box.

To create a depth image which has quarter (or less) of the resolution of the color image, has the advantage, that you can simple map between the high resolution and low resolution images by doubling or halving the xy values. You could use the high resolution image for your 2d vision algorithms and the low resolution point cloud for the 3d vision while still be able to simply combine the results.

HenningJ, 2015-07-08:
I'd love to have the option to get upscaled and registered-to-color depth images in addition to the current downscaled registered-to-depth color image. 

> This could be pretty accurate but you need to know the exact rotation and translation between the color and ir sensor

While this is not available in libfreenect2 right now, it has to be stored on the sensor somewhere, right? The Microsoft SDK does device dependent mapping between depth and color images and to me it seems to be pretty good (and different between the two devices I got here). So, if anybody can figure out how to get those values from the device, @wiedemeyer 's solution from iai_kinect2 seems to be the "most correct" one to me. What's known and what values are available, but are still unknown is scattered around different issues (like #41 and #144) and USB transactions the code is not well enough documented to be helpful for a noob like me. ;-)

If not, then the other proposed solutions seem fine.

wiedemeyer, 2015-07-08:
Can't we just compute a rotation and translation from the mapping that is currently implemented in libfreenect2? A 3d point can be computed by the following code:
```
x_3d = (x - cx) / fx * depth_xy
y_3d = (y - cy) / fy * depth_xy
z_3d = depth_xy
```
If we now compute a 3d point from the depth image perspective and then find the corresponding x,y coordinates in the color image and also create a 3d point from the color image perspective, we will end of with multiple pairs of corresponding 3d points, from which we could calculate the corresponding rotation and translation matrices.

floe, 2015-07-09:
But how could you compute a 3D point from the color image perspective alone? I don't think that's possible without resorting to some external knowledge, such as the geometry of a calibration chessboard?

wiedemeyer, 2015-07-09:
For each depth pixel we can compute the corresponding color image coordinates. So we can use those xy coordinates together with the depth value and the color intrinsics to compute a 3d point from the color image view. We can also use artificial points for computing the  rotation and translation.
The only problem is that the depth value should be slightly different from the color image perspective. So actually we can compute for each depth pixel a 3d points from the depth image perspective and a 3d Ray from the color image perspective.

floe, 2015-07-10:
> So actually we can compute for each depth pixel a 3d points from the depth image perspective and a 3d Ray from the color image perspective.
OK, that makes sense. But I think that this small difference (ray vs. point) would actually prevent you from being able to calculate a full transformation between the two camera spaces?

In any case, I don't think that this approach would change anything regarding the fundamental limitation: the depth image is far smaller than the color image. So some kind of upsampling operation is inevitable. For the moment, I'd like to propose a rather simple solution in #317 which allows to provide an optional extra `Frame` object to be used as depth buffer, resulting in a naively upsampled and aligned depth image.

xlz, 2015-07-10:
@wiedemeyer 

Reviewing the formula from https://github.com/OpenKinect/libfreenect2/issues/41#issuecomment-72022111
```
void kinect2_depth_to_color(float mx, float my, float z, float& rx, float& ry)
{
    mx *= depth_f * depth_q;
    my *= depth_f * depth_q;

    float wx =
        (mx * mx * mx * mx_x3y0) + (my * my * my * mx_x0y3) +
        (mx * mx * my * mx_x2y1) + (my * my * mx * mx_x1y2) +
        (mx * mx * mx_x2y0) + (my * my * mx_x0y2) + (mx * my * mx_x1y1) +
        (mx * mx_x1y0) + (my * mx_x0y1) + (mx_x0y0);
    float wy =
        (mx * mx * mx * my_x3y0) + (my * my * my * my_x0y3) +
        (mx * mx * my * my_x2y1) + (my * my * mx * my_x1y2) +
        (mx * mx * my_x2y0) + (my * my * my_x0y2) + (mx * my * my_x1y1) +
        (mx * my_x1y0) + (my * my_x0y1) + (my_x0y0);

    rx = wx / (color_f * color_q);
    ry = wy / (color_f * color_q);

    rx += (shift_m / z) - (shift_m / shift_d);

    rx = rx * color_f + color_cx;
    ry = ry * color_f + color_cy;
}
```

It seems wx and wy are doing something more than extrinsic transformation which is linear. It involves distortion up to the first order of radial distortion (not using k2, k3), or some kind of polynomial approximation of the third order (to k3 r^6). Then the extrinsic matrix cannot be directly computed or estimated with distortion in place.

What I don't understand is why `shift_m` is applied to `rx` *after* distortion. This should be done before it. I wonder if the extrinsic transformation is just a horizontal shift with no rotation. A sample of RgbCameraParamsResponse would help with this if anyone has the time to print it.

> We could create a color image sized depth image, which just consists of sparse depth points by using the corresponding coordinates. To fill it up, one could either set the neighborhood around each [...]

The filter map created in apply() is actually the depth image registered to color, with the only difference of its default value being infinity (65536.0f actually) instead of 0 or NaN. This is the obviously correct solution.

> Another solution might be to up-scale and interpolate the depth image by factor 3 to 4 and then compute the corresponding color pixel coordinates, and fill a color sized depth image, which would be less sparse or even not sparse at all.

What about mapping the depth image to a suitably small resolution in the color coordinate and then upscaling?

xlz, 2015-08-24:
The above equations don't seem to derived from standard models. If the rotation and translation were not trivial, z would be more involved. According to https://github.com/OpenKinect/libfreenect2/issues/41#issuecomment-55602322, there is almost no rotation, and translation is limited to along x axis only. And in the above equations wx and wy seem to be distortion only.

But the equations can't be reproduced exactly from the standard radial and tangential distortion with k1 k2 k3 p1 p2. The two models have different forms. If the above equations are approximating the standard models somehow, many factors would be zero and are not needed for computation. Given that there is no way to exactly reproduce the parameters of a standard model in order to derive its inverse, the color to depth transformation would only be more approximation of the above equations. I'd say using numeric methods to solve `kinect2_color_to_depth()` would be sufficient.

I plotted `kinect2_depth_to_color()` with a grid uniformly distributed on the undistorted focal plane of the RGB camera. The distortion is tiny:

[image]

This is the difference that `kinect2_depth_to_color()` has on a grid uniformly distributed on the distorted pixel plane of the depth camera. The points are resulting RGB pixel coordinates after `kinect2_undistort()` and `kinect2_depth_to_color()`.

[image]

A sample of all the parameters:

```
depth_f=365.976806640625;
depth_cx=255.781997680664;
depth_cy=210.049896240234;
depth_k1=0.0887134820222855;
depth_k2=-0.270025908946991;
depth_k3=0.0959607511758804;
color_f=1081.3720703125;
color_cx=959.5;
color_cy=539.5;
shift_d=863;
shift_m=52;
mx_x3y0=0.000437062612036243;
mx_x0y3=1.92725801753113e-05;
mx_x2y1=4.15435097238515e-05;
mx_x1y2=0.000348526606103405;
mx_x2y0=0.000298988888971508;
mx_x0y2=-0.000329198112012818;
mx_x1y1=0.000275816913926974;
mx_x1y0=0.653402626514435;
mx_x0y1=-0.000751816492993385;
mx_x0y0=0.142651602625847;
my_x3y0=2.32385700655868e-05;
my_x0y3=0.00100349204149097;
my_x2y1=0.000422552810050547;
my_x1y2=4.68458310933784e-05;
my_x2y0=4.83056101074908e-05;
my_x0y2=0.000365105312084779;
my_x1y1=0.000279815605608746;
my_x1y0=0.000444509903900325;
my_x0y1=0.652027189731598;
my_x0y0=0.0216553695499897;
depth_q=0.01;
color_q=0.002199;
```

xlz, 2015-10-26:
> add helper functions for point clouds (i.e. xyzRGB output)?

This is (kind of) implemented by #397. If that qualifies, this issue can be closed for 0.1.

floe, 2015-11-06:
I think it does.

closed 2015-11-06 by floe

SiddhantNadkarni, 2020-08-25:
> I tried the registration, but I assume there must be something wrong in the parameters probably (?)
> Here is the cloud created with the registration:
> [image]
> 
> while the one made with my own calibration looks like this:
> [image]
> 
> I could create a small video to see the difference if needed.

@giacomodabisias how did you manage to solve this? I get this error when I am trying to fuse point cloud (from kinect) and an  image (from an external camera) after doing extrinsic + intrinsic calibration.