OpenKinect/libfreenect2 #229: libusb problems on Mac Yosemite

ebubekirunlu, 2015-05-11:
Hi, 
I'm newbie worker for Kinect v2 on Mac. I did try run first example but, get some problems on process. 

I running this step : 
- cd ./examples/protonect/
- cmake CMakeLists.txt
- make && make install

And I get this errors: 

Building CXX object CMakeFiles/freenect2.dir/src/usb_control.cpp.o
/Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp:73:24: error: 
      use of undeclared identifier 'LIBUSB_SET_ISOCH_DELAY'
    uint8_t bRequest = LIBUSB_SET_ISOCH_DELAY;
                       ^
/Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp:88:24: error: 
      use of undeclared identifier 'LIBUSB_REQUEST_SET_SEL'
    uint8_t bRequest = LIBUSB_REQUEST_SET_SEL;
                       ^
/Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp:160:13: error: 
      unknown type name 'libusb_ss_endpoint_companion_descriptor'; did you mean
      'libusb_endpoint_descriptor'?
            libusb_ss_endpoint_companion_descriptor *companion_desc;
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            libusb_endpoint_descriptor
/Users/macpror/libfreenect2/examples/protonect/include/libfreenect2/protocol/libusb.h:452:8: note: 
      'libusb_endpoint_descriptor' declared here
struct libusb_endpoint_descriptor {
       ^
/Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp:162:17: error: 
      use of undeclared identifier 'libusb_get_ss_endpoint_companion_descriptor'
            r = libusb_get_ss_endpoint_companion_descriptor(NULL /* ctx ...
                ^
/Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp:166:33: error: 
      no member named 'wBytesPerInterval' in 'libusb_endpoint_descriptor'
            r = companion_desc->wBytesPerInterval;
                ~~~~~~~~~~~~~~  ^
/Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp:168:13: error: 
      use of undeclared identifier
      'libusb_free_ss_endpoint_companion_descriptor'
            libusb_free_ss_endpoint_companion_descriptor(companion_desc);
            ^
6 errors generated.
make[2]: *** [CMakeFiles/freenect2.dir/src/usb_control.cpp.o] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2

gaborpapp, 2015-05-11:
Have you installed the dependencies? https://github.com/OpenKinect/libfreenect2#mac-osx

ebubekirunlu, 2015-05-11:
Yes sure... I follow step by step.

gaborpapp, 2015-05-11:
Does the `depends/libusb/include/libusb-1.0/libusb.h` file exist?

ebubekirunlu, 2015-05-11:
No, it is not exist. Just "depends/libusb_src" file exist.

gaborpapp, 2015-05-11:
This means that the libusb build has failed. Try to run the relevant parts from `install_deps.sh` and check  how it fails.

ebubekirunlu, 2015-05-11:
I did try install this section : "install_deps.sh" and when finish installation create new folder glfw. But when I continue errors is same. Not difference.

gaborpapp, 2015-05-12:
What errors are you seeing?

ebubekirunlu, 2015-05-12:
same first error: 

Building CXX object CMakeFiles/freenect2.dir/src/usb_control.cpp.o
/Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp:73:24: error: 
use of undeclared identifier 'LIBUSB_SET_ISOCH_DELAY'
uint8_t bRequest = LIBUSB_SET_ISOCH_DELAY;
^
/Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp:88:24: error: 
use of undeclared identifier 'LIBUSB_REQUEST_SET_SEL'
uint8_t bRequest = LIBUSB_REQUEST_SET_SEL;
^
/Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp:160:13: error: 
unknown type name 'libusb_ss_endpoint_companion_descriptor'; did you mean
'libusb_endpoint_descriptor'?
libusb_ss_endpoint_companion_descriptor companion_desc;
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libusb_endpoint_descriptor
/Users/macpror/libfreenect2/examples/protonect/include/libfreenect2/protocol/libusb.h:452:8: note: 
'libusb_endpoint_descriptor' declared here
struct libusb_endpoint_descriptor {
^
/Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp:162:17: error: 
use of undeclared identifier 'libusb_get_ss_endpoint_companion_descriptor'
r = libusb_get_ss_endpoint_companion_descriptor(NULL / ctx ...
^
/Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp:166:33: error: 
no member named 'wBytesPerInterval' in 'libusb_endpoint_descriptor'
r = companion_desc->wBytesPerInterval;
~~~~~~~~~~~~~~ ^
/Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp:168:13: error: 
use of undeclared identifier
'libusb_free_ss_endpoint_companion_descriptor'
libusb_free_ss_endpoint_companion_descriptor(companion_desc);
^
6 errors generated.
make[2]: *** [CMakeFiles/freenect2.dir/src/usb_control.cpp.o] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2

gaborpapp, 2015-05-12:
Since libusb is not built properly I meant you should find the errors when running install_deps.sh.

ebubekirunlu, 2015-05-12:
when running install_deps.sh no exist error: 

AHMET-MacBook-Pro:libfreenect2 macpror$ sh ./depends/install_deps.sh
Cloning into '/Users/macpror/libfreenect2/depends/libusb_src'...
remote: Counting objects: 10760, done.
remote: Total 10760 (delta 0), reused 0 (delta 0), pack-reused 10760
Receiving objects: 100% (10760/10760), 2.47 MiB | 912.00 KiB/s, done.
Resolving deltas: 100% (7938/7938), done.
Checking connectivity... done.
Note: checking out '51b10191033ca3a3819dcf46e1da2465b99497c2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 51b1019... linux_usbfs: Update MAX_ISO_BUFFER_LENGTH
glibtoolize: putting auxiliary files in '.'.
glibtoolize: copying file './ltmain.sh'
glibtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
glibtoolize: copying file 'm4/libtool.m4'
glibtoolize: copying file 'm4/ltoptions.m4'
glibtoolize: copying file 'm4/ltsugar.m4'
glibtoolize: copying file 'm4/ltversion.m4'
glibtoolize: copying file 'm4/lt~obsolete.m4'
./bootstrap.sh: line 18: aclocal: command not found
./depends/install_deps.sh: line 16: ./configure: No such file or directory
make: *** No targets specified and no makefile found.  Stop.
Cloning into '/Users/macpror/libfreenect2/depends/glfw_src'...
remote: Counting objects: 14996, done.
remote: Total 14996 (delta 0), reused 0 (delta 0), pack-reused 14996
Receiving objects: 100% (14996/14996), 7.89 MiB | 295.00 KiB/s, done.
Resolving deltas: 100% (9412/9412), done.
Checking connectivity... done.
Note: checking out '3.0.4'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at eab31f2... Added description of 3.0.4 release.
-- The C compiler identification is AppleClang 6.0.0.6000057
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Found OpenGL: /System/Library/Frameworks/OpenGL.framework  
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Using Cocoa for window creation
-- Using NSGL for context creation
-- Building GLFW only for the native architecture
CMake Warning (dev) at examples/CMakeLists.txt:27 (add_executable):
  Policy CMP0037 is not set: Target names should not be reserved and should
  match a validity pattern.  Run "cmake --help-policy CMP0037" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
[... 289 lines cut ...]
[ 68%] Built target gamma
[ 72%] Built target glfwinfo
[ 76%] Built target iconify
[ 78%] Built target joysticks
[ 82%] Built target modes
[ 85%] Built target peter
[ 87%] Built target reopen
[ 89%] Built target sharing
[ 91%] Built target tearing
[ 95%] Built target threads
[ 97%] Built target title
[100%] Built target windows
Install the project...
-- Install configuration: ""
-- Installing: /Users/macpror/libfreenect2/depends/glfw/include/GLFW
-- Installing: /Users/macpror/libfreenect2/depends/glfw/include/GLFW/glfw3.h
-- Installing: /Users/macpror/libfreenect2/depends/glfw/include/GLFW/glfw3native.h
-- Installing: /Users/macpror/libfreenect2/depends/glfw/lib/cmake/glfw/glfwConfig.cmake
-- Installing: /Users/macpror/libfreenect2/depends/glfw/lib/cmake/glfw/glfwConfigVersion.cmake
-- Installing: /Users/macpror/libfreenect2/depends/glfw/lib/cmake/glfw/glfwTargets.cmake
-- Installing: /Users/macpror/libfreenect2/depends/glfw/lib/cmake/glfw/glfwTargets-noconfig.cmake
-- Installing: /Users/macpror/libfreenect2/depends/glfw/lib/pkgconfig/glfw3.pc
-- Installing: /Users/macpror/libfreenect2/depends/glfw/lib/libglfw.3.0.dylib
-- Installing: /Users/macpror/libfreenect2/depends/glfw/lib/libglfw.3.dylib
-- Installing: /Users/macpror/libfreenect2/depends/glfw/lib/libglfw.dylib

JuantAldea, 2015-05-12:
./bootstrap.sh: line 18: aclocal: command not found

ebubekirunlu, 2015-05-12:
I have automake-1.15 but problem is still available... @JuantAldea

gaborpapp, 2015-05-12:
what do you get when you run the following commands in `depends/libusb_src`?

    ./bootstrap.sh
    ./configure --prefix=`pwd`/../libusb
    make && make install

ebubekirunlu, 2015-05-12:
I did get this lines: 

AHMET-MacBook-Pro:libusb_src macpror$ ./bootstrap.sh
glibtoolize: putting auxiliary files in '.'.
glibtoolize: copying file './ltmain.sh'
glibtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
glibtoolize: copying file 'm4/libtool.m4'
glibtoolize: copying file 'm4/ltoptions.m4'
glibtoolize: copying file 'm4/ltsugar.m4'
glibtoolize: copying file 'm4/ltversion.m4'
glibtoolize: copying file 'm4/lt~obsolete.m4'
configure.ac:4: installing './compile'
configure.ac:4: installing './config.guess'
configure.ac:4: installing './config.sub'
configure.ac:2: installing './install-sh'
configure.ac:2: installing './missing'
Makefile.am: installing './depcomp'
glibtoolize: putting auxiliary files in '.'.
glibtoolize: copying file './ltmain.sh'
glibtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
glibtoolize: copying file 'm4/libtool.m4'
glibtoolize: copying file 'm4/ltoptions.m4'
glibtoolize: copying file 'm4/ltsugar.m4'
glibtoolize: copying file 'm4/ltversion.m4'
glibtoolize: copying file 'm4/lt~obsolete.m4'
configure.ac:38: installing './compile'
configure.ac:39: installing './config.guess'
configure.ac:39: installing './config.sub'
configure.ac:29: installing './install-sh'
configure.ac:29: installing './missing'
examples/Makefile.am: installing './depcomp'
AHMET-MacBook-Pro:libusb_src macpror$ ./configure --prefix=`pwd`/../libusb
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... ./install-sh -c -d
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking build system type... x86_64-apple-darwin14.0.0
checking host system type... x86_64-apple-darwin14.0.0
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld
checking if the linker (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld) is GNU ld... no
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm
checking the name lister (/usr/bin/nm) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 196608
checking how to convert x86_64-apple-darwin14.0.0 file names to x86_64-apple-darwin14.0.0 format... func_convert_file_noop
checking how to convert x86_64-apple-darwin14.0.0 file names to toolchain format... func_convert_file_noop
[... 105 lines cut ...]
#define IO_OBJECT_NULL  ((io_object_t) 0)
                        ^~~~~~~~~~~~~~~~~
1 warning generated.
  CC       os/libusb_1_0_la-poll_posix.lo
  CCLD     libusb-1.0.la
Making all in doc
make[2]: Nothing to be done for `all'.
make[2]: Nothing to be done for `all-am'.
Making install in libusb
 .././install-sh -c -d '/Users/macpror/libfreenect2/depends/libusb_src/../libusb/lib'
 /bin/sh ../libtool   --mode=install /usr/bin/install -c   libusb-1.0.la '/Users/macpror/libfreenect2/depends/libusb_src/../libusb/lib'
libtool: install: /usr/bin/install -c .libs/libusb-1.0.0.dylib /Users/macpror/libfreenect2/depends/libusb_src/../libusb/lib/libusb-1.0.0.dylib
libtool: install: (cd /Users/macpror/libfreenect2/depends/libusb_src/../libusb/lib && { ln -s -f libusb-1.0.0.dylib libusb-1.0.dylib || { rm -f libusb-1.0.dylib && ln -s libusb-1.0.0.dylib libusb-1.0.dylib; }; })
libtool: install: /usr/bin/install -c .libs/libusb-1.0.lai /Users/macpror/libfreenect2/depends/libusb_src/../libusb/lib/libusb-1.0.la
libtool: install: /usr/bin/install -c .libs/libusb-1.0.a /Users/macpror/libfreenect2/depends/libusb_src/../libusb/lib/libusb-1.0.a
libtool: install: chmod 644 /Users/macpror/libfreenect2/depends/libusb_src/../libusb/lib/libusb-1.0.a
libtool: install: ranlib /Users/macpror/libfreenect2/depends/libusb_src/../libusb/lib/libusb-1.0.a
 .././install-sh -c -d '/Users/macpror/libfreenect2/depends/libusb_src/../libusb/include/libusb-1.0'
 /usr/bin/install -c -m 644 libusb.h '/Users/macpror/libfreenect2/depends/libusb_src/../libusb/include/libusb-1.0'
Making install in doc
make[2]: Nothing to be done for `install-exec-am'.
make[2]: Nothing to be done for `install-data-am'.
make[2]: Nothing to be done for `install-exec-am'.
 ./install-sh -c -d '/Users/macpror/libfreenect2/depends/libusb_src/../libusb/lib/pkgconfig'
 /usr/bin/install -c -m 644 libusb-1.0.pc '/Users/macpror/libfreenect2/depends/libusb_src/../libusb/lib/pkgconfig'

gaborpapp, 2015-05-12:
it seemed to be built as far as i can tell. can you try building protonect now?

ebubekirunlu, 2015-05-12:
I follow this lines again: 
- cd ./libfreenect2
- sh ./depends/install_mac.sh

And created depends/libusb/include/libusb-1.0/libusb.h after I follow : 
- cd ./examples/protonect/
- cmake CMakeLists.txt
- make && make install

but I did get same errors again... 
[image]

gaborpapp, 2015-05-12:
is it possible that you have libusb installed elsewhere and make picks it up instead of the one installed in the depends folder? what is the output if you run make like:

`make VERBOSE=1`

ebubekirunlu, 2015-05-12:
libusb is installed on /home folder with libfreenect2... 

make VERBOSE=1 : 

AHMET-MacBook-Pro:protonect macpror$ make VERBOSE=1
/usr/local/Cellar/cmake/3.1.3/bin/cmake -H/Users/macpror/libfreenect2/examples/protonect -B/Users/macpror/libfreenect2/examples/protonect --check-build-system CMakeFiles/Makefile.cmake 0
/usr/local/Cellar/cmake/3.1.3/bin/cmake -E cmake_progress_start /Users/macpror/libfreenect2/examples/protonect/CMakeFiles /Users/macpror/libfreenect2/examples/protonect/CMakeFiles/progress.marks
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/Makefile2 all
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f /Users/macpror/libfreenect2/depends/glfw/src/CMakeFiles/glfw.dir/build.make /Users/macpror/libfreenect2/depends/glfw/src/CMakeFiles/glfw.dir/depend
cd /Users/macpror/libfreenect2/examples/protonect && /usr/local/Cellar/cmake/3.1.3/bin/cmake -E cmake_depends "Unix Makefiles" /Users/macpror/libfreenect2/examples/protonect /Users/macpror/libfreenect2/depends/glfw_src/src /Users/macpror/libfreenect2/examples/protonect /Users/macpror/libfreenect2/depends/glfw/src /Users/macpror/libfreenect2/depends/glfw/src/CMakeFiles/glfw.dir/DependInfo.cmake --color=
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f /Users/macpror/libfreenect2/depends/glfw/src/CMakeFiles/glfw.dir/build.make /Users/macpror/libfreenect2/depends/glfw/src/CMakeFiles/glfw.dir/build
make[2]: Nothing to be done for `/Users/macpror/libfreenect2/depends/glfw/src/CMakeFiles/glfw.dir/build'.
/usr/local/Cellar/cmake/3.1.3/bin/cmake -E cmake_progress_report /Users/macpror/libfreenect2/examples/protonect/CMakeFiles  24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40
[ 42%] Built target glfw
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/generate_resources_tool.dir/build.make CMakeFiles/generate_resources_tool.dir/depend
cd /Users/macpror/libfreenect2/examples/protonect && /usr/local/Cellar/cmake/3.1.3/bin/cmake -E cmake_depends "Unix Makefiles" /Users/macpror/libfreenect2/examples/protonect /Users/macpror/libfreenect2/examples/protonect /Users/macpror/libfreenect2/examples/protonect /Users/macpror/libfreenect2/examples/protonect /Users/macpror/libfreenect2/examples/protonect/CMakeFiles/generate_resources_tool.dir/DependInfo.cmake --color=
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/generate_resources_tool.dir/build.make CMakeFiles/generate_resources_tool.dir/build
make[2]: Nothing to be done for `CMakeFiles/generate_resources_tool.dir/build'.
/usr/local/Cellar/cmake/3.1.3/bin/cmake -E cmake_progress_report /Users/macpror/libfreenect2/examples/protonect/CMakeFiles  23
[ 45%] Built target generate_resources_tool
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/freenect2.dir/build.make CMakeFiles/freenect2.dir/depend
cd /Users/macpror/libfreenect2/examples/protonect && /usr/local/Cellar/cmake/3.1.3/bin/cmake -E cmake_depends "Unix Makefiles" /Users/macpror/libfreenect2/examples/protonect /Users/macpror/libfreenect2/examples/protonect /Users/macpror/libfreenect2/examples/protonect /Users/macpror/libfreenect2/examples/protonect /Users/macpror/libfreenect2/examples/protonect/CMakeFiles/freenect2.dir/DependInfo.cmake --color=
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/freenect2.dir/build.make CMakeFiles/freenect2.dir/build
/usr/local/Cellar/cmake/3.1.3/bin/cmake -E cmake_progress_report /Users/macpror/libfreenect2/examples/protonect/CMakeFiles 15
[ 47%] Building CXX object CMakeFiles/freenect2.dir/src/usb_control.cpp.o
/usr/bin/c++   -DRESOURCES_INC -Dfreenect2_EXPORTS -O2 -g -DNDEBUG -fPIC -I/Users/macpror/libfreenect2/examples/protonect/src/tinythread -I/usr/local/include/opencv -I/usr/local/include -I/Users/macpror/libfreenect2/examples/protonect/../../depends/libusb/include/libusb-1.0 -I/Users/macpror/libfreenect2/examples/protonect/../../depends/glfw_src/include -I/Users/macpror/libfreenect2/examples/protonect/../../depends/libjpeg_turbo/include -I/Users/macpror/libfreenect2/examples/protonect/include    -o CMakeFiles/freenect2.dir/src/usb_control.cpp.o -c /Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp
/Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp:73:24: error: 
      use of undeclared identifier 'LIBUSB_SET_ISOCH_DELAY'
    uint8_t bRequest = LIBUSB_SET_ISOCH_DELAY;
                       ^
/Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp:88:24: error: 
      use of undeclared identifier 'LIBUSB_REQUEST_SET_SEL'
    uint8_t bRequest = LIBUSB_REQUEST_SET_SEL;
                       ^
/Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp:160:13: error: 
      unknown type name 'libusb_ss_endpoint_companion_descriptor'; did you mean
      'libusb_endpoint_descriptor'?
            libusb_ss_endpoint_companion_descriptor *companion_desc;
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            libusb_endpoint_descriptor
/Users/macpror/libfreenect2/examples/protonect/include/libfreenect2/protocol/libusb.h:452:8: note: 
      'libusb_endpoint_descriptor' declared here
struct libusb_endpoint_descriptor {
       ^
/Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp:162:17: error: 
      use of undeclared identifier 'libusb_get_ss_endpoint_companion_descriptor'
            r = libusb_get_ss_endpoint_companion_descriptor(NULL /* ctx ...
                ^
[... cut ...]
            r = companion_desc->wBytesPerInterval;
                ~~~~~~~~~~~~~~  ^
/Users/macpror/libfreenect2/examples/protonect/src/usb_control.cpp:168:13: error: 
      use of undeclared identifier
      'libusb_free_ss_endpoint_companion_descriptor'
            libusb_free_ss_endpoint_companion_descriptor(companion_desc);
            ^
6 errors generated.
make[2]: *** [CMakeFiles/freenect2.dir/src/usb_control.cpp.o] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2

gaborpapp, 2015-05-12:
if you don't have libusb includes installed in `/usr/local/include` as well i am out of ideas.

ebubekirunlu, 2015-05-12:
I have libusb in include: 

[image]

gaborpapp, 2015-05-12:
ok. then probably this is used instead of the one in the depends folder. move this directory somewhere, and also the libusb libs in /usr/local/lib and try building protonect again.

ebubekirunlu, 2015-05-12:
I moved libfreenect2 folder to different root... But same problems exists. So I'm give in!

closed 2015-05-20 by floe

adamjdorwart, 2015-08-11:
Thought I'd chime in since I ended up here. I've successfully built libfreenect2 on machines and gotten this to work in the past but on a new machine I'm working with I'm getting a lot of libusb errors and no output. The last thing I tried was downloading the latest tarball for libusb:
http://sourceforge.net/projects/libusb/files/libusb-1.0/libusb-1.0.9/libusb-1.0.9.tar.bz2
and building from source. After I did this I got the same exact errors in the original issue.